espressif/openocd-esp32 #95: OpenOCD debug error openocd v0.10.0-esp32-20191114 (IDFGH-2653)

axbusch, 2020-02-08:
- Development Kit:      [ESP32-Wrover-Kit]
- Module or chip used:  [ESP32-WROVER|]
- IDF version : v4.0-rc-12-ga61fd1e42b
--
WIN7

OpenOCD v0.10.0-esp32-20191114

## Problem Description
At the moment i cant debug. It fails with this output
`
D:\esp\remote\esp32_mp3_decoder>call D:\esp\idf\export.bat
Setting IDF_PATH: D:\esp\idf

Adding ESP-IDF tools to PATH...
    D:\esp\toolchain\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    D:\esp\toolchain\.espressif\tools\cmake\3.13.4\bin
    D:\esp\toolchain\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    D:\esp\toolchain\.espressif\tools\mconf\v4.6.0.0-idf-20190628\
    D:\esp\toolchain\.espressif\tools\ninja\1.9.0\
    D:\esp\toolchain\.espressif\tools\idf-exe\1.0.1\
    D:\esp\toolchain\.espressif\tools\ccache\3.7\
    D:\esp\toolchain\.espressif\python_env\idf4.0_py2.7_env\Scripts
    D:\esp\idf\tools

Checking if Python packages are up to date...
Python requirements from D:\esp\idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

openocd.exe -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover.cfg

Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 16000 kHz
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: type 'esp32' is missing virt2phys
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 16000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset.
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : cpu0: Detected debug stubs @ 3ffb6558
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Info : Target halted. CPU0: PC=0x4018A72E (active)
Info : Target halted. CPU1: PC=0x4009571C
Info : cpu0: Target halted, PC=0x400916EA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 335 KB
Info : Flash mapping 1: 0x70018 -> 0x400d0018, 751 KB
Info : cpu0: Target halted, PC=0x400916EA, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.flash' size 4096 KB
Info : Using flash bank 'esp32.flash' size 4096 KB
Info : cpu0: Target halted, PC=0x400916EA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 335 KB
Info : Flash mapping 1: 0x70018 -> 0x400d0018, 751 KB
Info : cpu0: Target halted, PC=0x400916EA, debug_reason=00000001
Info : Auto-detected flash bank 'irom' size 4096 KB
Info : Using flash bank 'irom' size 4096 KB
Info : cpu0: Target halted, PC=0x400916EA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 335 KB
Info : Flash mapping 1: 0x70018 -> 0x400d0018, 751 KB
Info : cpu0: Target halted, PC=0x400916EA, debug_reason=00000001
Info : Auto-detected flash bank 'drom' size 4096 KB
Info : Using flash bank 'drom' size 4096 KB
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400
** Programming Started **
auto erase enabled
Info : cpu0: Target halted, PC=0x400916EA, debug_reason=00000001
Info : cpu0: Target halted, PC=0x400916EA, debug_reason=00000001
wrote 1241088 bytes from file D:/esp/remote/esp32_mp3_decoder/build/webradio.bin in 8.140012s (148.894 KiB/s)
** Programming Finished **
** Verify Started **
Info : cpu0: Target halted, PC=0x400916EA, debug_reason=00000001
read 1238832 bytes from file D:/esp/remote/esp32_mp3_decoder/build/webradio.bin and flash bank 0 at offset 0x00010000 in 3.250004s (372.245 KiB/s)
contents match
** Verified OK **
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400
Error: cpu0: Failed to write breakpoint instruction (-4)!
Error: cpu0: Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
`

My initialize commands are
`mon reset halt
mon program_esp32 "D:/esp/remote/esp32_mp3_decoder/build/webradio.bin" 0x10000 verify
mon reset halt
flushregs
set remote hardware-watchpoint-limit 2`

and i have no breakpoints set

fromeijn, 2020-02-17:
Got same issue with v0.10.0-esp32-20191008 (2019-10-23-10:50) 

but this version does NOT have this issue:
v0.10.0-esp32-20190313 (2019-03-13-09:52)

you may try that version instead

axbusch, 2020-02-17:
yeah, you are right. with the old version it works

omg, this is so fu*** frustrating! you can't rely on anything. week after week new bugs, errors, subtle problems.

debugging is still unsteady as hell. sometimes breakpoints are not recognized, debugger stops completly, etc etc

igrr, 2020-02-17:
Hi @axbusch, sorry for the issue. Could you please upload the logs of OpenOCD as mentioned here? https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb

Another question, are you using the default or a custom partition table? Are there any flash-related changes in your sdkconfig?

axbusch, 2020-02-18:
hey @igrr , first: sorry for my rant yesterday!!!
not very helpful but i was just frustrated!

I tried again with Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)

Partion.csv looks like
`# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 2M,
`

My OpenOCD Log: https://gist.github.com/axbusch/2ecb6fa9cd3a9ec21414af926cc14c0c

There are two errors:
Error: No symbols for FreeRTOS
Error: Too large number of threads 4294967295!

I had some errors before because "mon program_esp32" changed to "mon program_esp" but that was my fault

And here my eclipse config
[image]

axbusch, 2020-02-18:
i do not see all threads. i think becaus of the freertos error?

here is my sdkconfig
[image]

[sdkconfig.txt](https://github.com/espressif/openocd-esp32/files/4221674/sdkconfig.txt)

gerekon, 2020-02-19:
@axbusch To get full OpenOCD log you need to run it with `-d3` option as mentioned here https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb. Could you post full log?

> i do not see all threads. i think becaus of the freertos error?

It seems that you do not resume target after programing, so it is halted after the reset and therefore you do not see any threads (app is not strated yet). You need to press `Resume` button or check `Resume` checkbox in your eclipse configuration (as described here https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse). After that app should start and stop at `app_main` where you will be able to see threads.
 
> Error: No symbols for FreeRTOS
Error: Too large number of threads 4294967295!

Sorry for inconvenience these message are not critical and can be shown when FreeRTOS has not started yet. We will change the level for them to avoid misunderstandings in future.

axbusch, 2020-02-21:
it seems like debugging works little better with debugging enabled :o: 

[log](https://gist.github.com/axbusch/df2000d295680854eb713968ab393365)

But i see not all threadnames.

[log2](https://gist.github.com/axbusch/954bd53285a43020d8950017628d6fda)
Another one where i could start debugging and then connection is lost

gerekon, 2020-02-22:
> But i see not all threadnames.

Maybe this is Due to `Error: 1955 4474 fileio.c:90 fileio_open_local(): couldn't open D:/esp/remote/mp3decoder/build/esp32_mp3_decoder.bin`  ?
```
User : 1950 4473 command.c:783 jim_echo(): ** Programming Started **
Debug: 1951 4473 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_flash write_image erase D:/esp/remote/mp3decoder/build/esp32_mp3_decoder.bin 0x10000
Debug: 1952 4473 command.c:143 script_debug(): command - ocd_flash ocd_flash write_image erase D:/esp/remote/mp3decoder/build/esp32_mp3_decoder.bin 0x10000
User : 1954 4474 command.c:544 command_print(): auto erase enabled
Error: 1955 4474 fileio.c:90 fileio_open_local(): couldn't open D:/esp/remote/mp3decoder/build/esp32_mp3_decoder.bin
Debug: 1956 4474 command.c:645 run_command(): Command 'write_image' failed with error code -1202
User : 1957 4475 command.c:708 command_run_line(): embedded:startup.tcl:480: Error: ** Programming Failed **
in procedure 'program_esp' 
in procedure 'program_error' called at file "D:\esp\toolchain\.espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 106
at file "embedded:startup.tcl", line 480
```

gerekon, 2020-02-22:
> Another one where i could start debugging and then connection is lost

The second log has the same failure. It seems that you tries to debug app using old image.

axbusch, 2020-02-23:
hey, ok you are right :/ 

i made some mistake and it was all the time using the old program from earlier (in the old debug logs i had wrote 1241088 bytes from file D:/esp/remote/esp32_mp3_decoder/build/webradio.bin in 8.140012s (148.894 KiB/s)
** Programming...)

i see the threads now!
 after lowering jtag speed to 10khz and i also use ftdi_tdo_sample_edge falling it works better.
 at the moment i have some 20cm wires from the ftdi to the board. i will try to use some shorter cables to make it more stable

tom-borcin, 2021-03-26:
Hello @axbusch, we believe this issue is resolved and because of that we are closing it.

If you think it should remain open, please let me know. Thank you for your contribution!

closed 2021-03-26 by tom-borcin

ashigupta, 2021-05-23:
Hello,

Somehow i am also getting below errors while debugging blink example code using ESP PROG on ESP32-DevKitM-1. Below is the screenshot of errors
[image]

debugging was working fine till yesterday but somehow it stopped today and stuck in this error. However direct flashing code is working fine. I have also tried re-installing the drivers for  ESP PROG but that has not helped. 

Please let me know this if any more information is needed. 

I have been using ESP now quite a  time but still debugging on ESP32 is not reliable. Debugging stops abruptly any time. Is there any otherway to debug code in more easier and faster way ?

gerekon, 2021-05-24:
Hi @ashigupta !

The reason for this issue was found. Please, see above. If you think this is another issue, please open a new issue and fill in the necessary information according to the template (including log files).