espressif/openocd-esp32 #97: openocd configuration problem (IDFGH-2708)

PDP-EFS, 2020-02-17:
Hello,
I'have a problem when I try to connect the jtag (ftdi c232hm 3.3V) to the ESP32_Devkit C_V4 I have this error message:

```
openocd -f interface/ftdi/c232hm.cfg -f target/esp32.cfg
Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
**Error: type 'esp32' is missing virt2phys
Error: An adapter speed is not selected in the init script. Insert a call to adapter_khz or jtag_rclk to proceed.**

Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of esp32.irom does not support free_driver_priv()
Warn : Flash driver of esp32.drom does not support free_driver_priv()
```

I'm under debian 10.3.0, with the ESP-IDF 4.0.
The same cable with the same board works on window 10.

Can you help me? 
Best regards,
Pietro

igrr, 2020-02-18:
```
Error: An adapter speed is not selected in the init script. Insert a call to adapter_khz or jtag_rclk to proceed.
```

`c232hm.cfg` doesn't include `adapter_khz` command. Have you tried adding this command as suggested in the error message? Alternatively you can add the command as an argument to OpenOCD.

tom-borcin, 2021-03-26:
Hello @PDP-EFS, we are closing this issue due to inactivity in an effort to keep our backlog focused and manageable. If you want to keep this issue open, please let me know.

Thank you for your contribution!

closed 2021-03-26 by tom-borcin