espressif/openocd-esp32 #99: JTAG debugging not working on ESP-WROVER-KIT V3

ls6777, 2020-02-18:
I followed all the instructions to setup JTAG debugging.  I'm trying with command line first, before moving over to eclipse debugging.  I see the following on the output:

```
C:\esp\hello_world>xtensa-esp32-elf-gdb -x gdbinit build/hello-world.elf
GNU gdb (crosstool-NG esp-2019r2) 8.1.0.20180627-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-host_w64-mingw32 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/hello-world.elf...done.
gdbinit:1: Error in sourced command file:
:3333: No connection could be made because the target machine actively refused it.
```
Flashing, without debugging, works fine through command line and eclipse.

gerekon, 2020-02-18:
@ls6777 Have you run OpenOCD before GDB start? It seems that you have not.

ls6777, 2020-02-19:
This is what happens when I try to run OpenOCD:

```
C:\esp\hello_world>openocd -f board/esp32-wrover.cfg
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: Debug adapter doesn't support any transports?
embedded:startup.tcl:24: Error:
in procedure 'script'
at file "embedded:startup.tcl", line 63
in procedure 'transport' called at file "C:\espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\share\openocd\scripts/board/esp32-wrover.cfg", line 29
in procedure 'ocd_bouncer'
at file "embedded:startup.tcl", line 24
```

The instructions say to use esp32-wrover-kit-3.3v.cfg, but I don't see that file in the board directory.

igrr, 2020-02-19:
Are you using IDF v4.0? If yes, please make sure you refer to the corresponding version of documentation. The instructions vary depending on OpenOCD version, and different IDF versions install different OpenOCD versions.
These are the instructions for IDF v4.0, and they are compatible with your version of OpenOCD.
https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/jtag-debugging/index.html#jtag-debugging-run-openocd

ls6777, 2020-02-20:
@igrr Thanks!!  There is so much documentation and it's easy to get lost between the different versions.  I followed the instructions from the link you provided and got it working.  Now I just need to get it working in eclipse.  Thanks again!!!

closed 2020-02-20 by ls6777