espressif/openocd-esp32 #100: SegFault if factory app not located at 0x10000

phatpaul, 2020-02-28:
OpenOCD crashes with "Segmentation Fault" when trying to start debugging on ESP32-WROVER (16M).  I believe I found the issue to be that my factory app was relocated to 0x20000 to make room for an additional NVS partition (for reasoning see thread: https://esp32.com/viewtopic.php?f=13&t=14129)
The problem is gone if I move the factory app back to 0x10000.

It is inconveneient to move the factory app back to 0x10000 temporarily for debugging.  I couldn't find any documentation describing this limitation or how to tell GDB/OpenOCD of the offset.

- [x] JTAG adapter type.
olimex-arm-usb-ocd-h

- [x] Release of ESP-IDF used to compile and load application that is being debugged.
ESP-IDF release 3.3.1  (Legacy GNU Make)

- [x] Details of OS used for debugging.
Windows 7 64bit.  

- [x] Is OS running natively on a PC or on a virtual machine?
Native.

- [x] Create a simple example that is representative to observed issue. Describe steps how to reproduce it. 

I just followed the guidance here: https://docs.espressif.com/projects/esp-idf/en/v3.3.1/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-command-line

Partition Table:

 Name | Type | SubType | Offset | Size | Flags
  ----|---------|----------|---------|------|-------
nvs | data | nvs | 0x9000 | 16K |  
otadata | data | ota | 0xd000 | 8K |  
phy_init | data | phy | 0xf000 | 4K |  
fdata | data | nvs | 0x10000 | 12K |  
factory | app | factory |   | 4096K |  
ota_0 | app | ota_0 |   | 4160K |  
ota_1 | app | ota_1 |   | 4160K |  
internalfs | data | fat |   | 3840K

```
cd ~/openocd-esp32
./bin/openocd.exe -d3 -s share/openocd/scripts/ -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f board/esp32-wrover.cfg 2>&1 | tee openocd.log
cd ~/my-project
xtensa-esp32-elf-gdb -ex "set remotelogfile gdb_log.txt" -x gdbinit build/esp32-ota.elf
```

- [x] Attach both openocd_log.txt and gdb_log.txt files to your issue report.
[openocd.log](https://github.com/espressif/openocd-esp32/files/4267744/openocd.log)
[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/4267746/gdb_log.txt)

gerekon, 2020-02-28:
@phatpaul Thanks for the report. I will have a look at the crash. 
To set app offset you need to add `mon esp32 appimage_offset 0x20000` into your GDB init commands or run OpenOCD with extra arguments as it is described [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#flash-mappings-vs-sw-flash-breakpoints).
Could you check it?

gerekon, 2020-02-28:
@phatpaul  BTW, where is your partition table located? What offset?

phatpaul, 2020-02-28:
Thanks @gerekon I missed that part of the documentation.

I tried setting it via telnet and via command-line but I'm still getting an error when trying to run GDB.  Please see attached logs after trying that command.
[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/4267989/gdb_log.txt)
[serial_log.txt](https://github.com/espressif/openocd-esp32/files/4267992/serial_log.txt)
[openocd.log](https://github.com/espressif/openocd-esp32/files/4268006/openocd.log)

My partition table is located at 0x8000.  It is my understanding that can't be changed as it is hardcoded into the bootloader.

gerekon, 2020-03-03:
Partition  offset can be changed, but if it is at the default one you should be able to debug w/o any extra commands.
BTW I checked debugging of blink example at 0x20000 and it works with OpenOCD `v0.10.0-esp32-20191114`.

As I see from your first log program crashes with exception cause `StoreProhibitedCause` at address `0x4000C02B` somewhere in ROM code, probably `memset` or `memcpy`. ROM code stops at SW breakpoint, OpenOCD treats this as semihosting call and tries to execute it w/o checking arguments. Invalid arguments lead to OpenOCD crash. We will add fix for this.
```
Debug: 2085 7094 xtensa_mcore.c:579 xtensa_mcore_poll(): esp32.cpu0: Target halted, pc=0x4000C02B, debug_reason=00000001, oldstate=00000001, active=true
Debug: 2086 7094 xtensa_mcore.c:585 xtensa_mcore_poll(): esp32.cpu0: Halt reason=0x00000008, exc_cause=29, dsr=0x8011cc11
Info : 2087 7094 xtensa_mcore.c:589 xtensa_mcore_poll(): Target halted. CPU0: PC=0x4000C02B (active)
Debug: 2088 7095 xtensa_mcore.c:579 xtensa_mcore_poll(): esp32.cpu1: Target halted, pc=0x40000400, debug_reason=00000001, oldstate=00000001, active=false
Debug: 2089 7095 xtensa_mcore.c:585 xtensa_mcore_poll(): esp32.cpu1: Halt reason=0x00000020, exc_cause=0, dsr=0x8190cc11
Info : 2090 7095 xtensa_mcore.c:589 xtensa_mcore_poll(): Target halted. CPU1: PC=0x40000400 
Debug: 2104 7100 esp_xtensa.c:458 esp_xtensa_do_semihosting(): Call 0x1 0x3f40922d 0xffffffff 0x0 0xdc .
Debug: 2105 7100 target.c:2343 target_read_buffer(): reading buffer of -1 byte at 0x3f40922d
```

commit d435dfab (2020-10-16): esp/semihost: Fixes invalid syscall arguments handling

Closes #100
closed 2020-10-16 by espressif-bot