espressif/openocd-esp32 #101: Why do my Eclipse GDB sessions don't work anymore ?

salocinx, 2020-03-02:
## Environment

- Development Kit:        ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROVER
- IDF version:                v4.2-dev-414-g132cc67c0
- Build System:             CMake
- Compiler version:       xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux (Ubuntu 18.04 LTS)
- Development IDE:      Eclipse CDT 2019.12
- Power Supply:            external 5V / 5A

## Problem Description

Hi. I am currently tearing my hairs off...

I followed the official installation instructions for [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-get-esp-idf), [Eclipse IDF Plugin](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md), Eclipse [JTAG](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/) and [GDB](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse) Debugging.

The GDB debugging **initially** worked quite well, but now I get some nasty errors and the debugging session won't launch anymore in Eclipse CDT. It could be that the troubles started when I tried to use GPIO 16/17 for controlling DC motors. I then realized, that these pins are **not** available on the ESP-WROVER-KIT, although they are broken out on the dev board. However, I am not using these pins in my program anymore.

I am currently trying to reanimate the GDB session/debugger with the unmodified Blink example.

I start OpenOCD with the following command: `openocd -f board/esp32-wrover-kit-3.3v.cfg`

These are the boot messages from OpenOCD:

> Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)
> Licensed under GNU GPL v2
> For bug reports, read
> 	http://openocd.org/doc/doxygen/bugs.html
> none separate
> adapter speed: 20000 kHz
> Info : Configured 2 cores
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Error: type 'esp32' is missing virt2phys
> Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
> Info : clock speed 20000 kHz
> Error: JTAG scan chain interrogation failed: all zeroes
> Error: Check JTAG interface, timings, target power, etc.
> Error: Trying to use configured scan chain anyway...
> Error: esp32.cpu0: IR capture error; saw 0x00 not 0x01
> Warn : Bypassing JTAG setup events due to errors
> Info : Listening on port 3333 for gdb connections
> Info : xtensa_mcore_poll: Target offline
> Error: xtensa_mcore_poll: Target failure
> Polling target esp32 failed, trying to reexamine
> Examination failed, GDB will be halted. Polling again in 100ms
> Error: xtensa_mcore_poll: Target failure
> Polling target esp32 failed, trying to reexamine
> Examination failed, GDB will be halted. Polling again in 300ms
> Error: xtensa_mcore_poll: Target failure
> Polling target esp32 failed, trying to reexamine
> Examination failed, GDB will be halted. Polling again in 700ms
> Error: xtensa_mcore_poll: Target failure
> Polling target esp32 failed, trying to reexamine
> Examination failed, GDB will be halted. Polling again in 1500ms
> Error: xtensa_mcore_poll: Target failure
> Polling target esp32 failed, trying to reexamine
> Examination failed, GDB will be halted. Polling again in 3100ms

As soon as I try to launch the debug session in Eclipse CDT I get these additional messages:

> Info : accepting 'gdb' connection on tcp/3333
> Error: No symbols for FreeRTOS
> Error: xtensa_mcore_poll: Target failure
> 
> Error: Target not halted
> Error: auto_probe failed
> Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
> Error: attempted 'gdb' connection rejected
> Error: xtensa_mcore_poll: Target failure
> Polling target esp32 failed, trying to reexamine

Eclipse CDT doesn't switch to the debugger view anymore. Debug session doesn't work.

### Expected Behavior

When launching the GDB launch target in debug mode in Eclipse CDT, the debugger view should show up and halting the program at app_main() function.

### Actual Behavior

Eclipse CDT doesn't start the debug session anymore. Compare to the GDB error messages above.

### Note

Please let me know if I need to provide some more logs or any configuration files or settings.

I highly appreciate any help that directs me on to the correct path to bring the debugging sessions back alive :-) ! Thank you!

salocinx, 2020-03-02:
Now it's working again, with the Blink example and my own, more complex, program. It seems that using GPIO 14, 15, 18 and 19 for MCPWM conflicted with the debug interface. Does somebody know which pins to **not** use when using JTAG debugging?

gerekon, 2020-03-02:
@salocinx You can not debug when you use JTAG pins for anything else. Please see https://github.com/espressif/esp-idf/blob/c1d0daf36d0dca81c23c226001560edfa51c30ea/components/soc/soc/esp32/include/soc/io_mux_reg.h#L198-L232

salocinx, 2020-03-02:
Thanks for clarifying! I am going to close this issue.

closed 2020-03-02 by salocinx