espressif/openocd-esp32 #103: Problem - can't debug ESP32 wrover under linux - ESP PROG, Jtag

lafar6502, 2020-03-27:
Can you please help me figure out what's wrong? Apparently i can connect to the board, but unable to start valid GDB debug session.

`
rg@espdev:~/prj/esp2test$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic
`
i'm using ESP prog jtag and debugging ESP32 Wrover board

-------------------------- GDB output ------------------
`
rg@espdev:~/prj/esp2test$ xtensa-esp32-elf-gdb 
GNU gdb (crosstool-NG esp-2019r2) 8.1.0.20180627-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target remote :3333
Remote debugging using :3333
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
(gdb) 
`

------ OPENOCD log -----------------------------------
Open On-Chip Debugger  v0.10.0-esp32-20200309 (2020-03-09-11:51)

`
adapter speed: 20000 kHz

WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future $
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the comma$
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica$
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica$
Info : Target halted. CPU0: PC=0x40090365 (active)
Info : Target halted. CPU1: PC=0x400076E0
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Error: timed out while waiting for target halted / 4 - 2
Info : cpu0: Target halted, PC=0x40090365, debug_reason=00000000
Error: xtensa_wait_algorithm: not halted 0, pc 0x40090365, ps 0x60025
Error: Failed to wait algorithm (-302)!
Error: Failed to run flasher stub (-302)!
Warn : Failed to get flash mappings (-302)!
Error: Target is already running an algorithm
Error: Failed to start algorithm (-4)!
Error: Algorithm run failed (-4)!
Error: Failed to run flasher stub (-4)!
Error: Target is already running an algorithm
Error: Failed to start algorithm (-4)!
Error: Algorithm run failed (-4)!
Error: Failed to run flasher stub (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to$
Error: attempted 'gdb' connection rejected
shutdown command invoked
Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of esp32.irom does not support free_driver_priv()
Warn : Flash driver of esp32.drom does not support free_driver_priv()

`

lafar6502, 2020-03-27:
[openocd.log](https://github.com/espressif/openocd-esp32/files/4396028/openocd.log)

gerekon, 2020-03-30:
@lafar6502 It seems that the target in some weird state and can not be halted by OpenOCD. Could you try to flash it using `esptool`? What is printed in UART?

lafar6502, 2020-04-01:
Hi, 
First of all, i got rid of the problem after some more or less relevant steps. But still dont have an explanation about what was going on
1. Switched from a virtual machine Linux to a Windows host (no VM)
2. powering the board now from its USB port, and disconnected the vcc wire from the JTag board
3. switched openocd board config script from esp32-wroom to esp32-wrover
But these scripts are almost identical, the only difference is that esp32.wrover.cfg has this line
set ESP32_FLASH_VOLTAGE 1.8
4. Also, using a different ftdi board now (generic ft 2232 instead of esp-prog)
Did all combinations of the above and finally got a working configuration, but it's very difficult to tell now what was going on. For sure the wrover board is not behaving same way as nodemcu esp32s clones.

dot4qu, 2020-11-11:
For anyone that still hasn't found a solution, I found that bumping the `adapter_khz` parameter in my custom interface .cfg file for openocd from 200 to 20000 solved my `Remote replied unexpectedly to 'vMustReplyEmpty'` issue. 

Another point of interest to note was that after doing so, I could connect to the extended-remote gdb target successfully regardless of if I had flash the board using `idf.py app-flash` or using openocd with a manual command of `openocd -f path/to/custom/interface -f target/esp32.cfg -c "program_esp path/to/.bin 0z10000 verify exit`. However, when trying to load the .elf symbols in gdb, it would time out with an error if the former was used.

So the only process that worked for me was to program through `idf.py app-flash` then start openocd with both the supplied esp32.cfg target file and my own custom interface cfg file shown below:
```
interface ftdi
ftdi_vid_pid 0x0403 0x6014    # you'll have to look these up for your ftdi ft232h board - the esp-idf openocd guide tells you how to find
ftdi_channel 0
ftdi_layout_init 0x0038 0x003b

adapter_khz 20000
```

closed 2023-01-17 by erhankur