espressif/openocd-esp32 #104: invalid command name "program_esp"

howroyd, 2020-03-31:
Using the default cfg files except I have reduced my adapter speed.

I'm getting the error `invalid command name "program_esp"` and can't figure out why.

`E:\Simon\Documents\Code\hub_firmware\hub_esp32>C:\Users\Simon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin\openocd.exe -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "program_esp build/blink.bin 0x10000 verify exit"
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 2000 kHz
ftdi samples TDO on falling edge of TCK
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Configured 2 cores
esp32 interrupt mask on
invalid command name "program_esp"
Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of irom does not support free_driver_priv()
Warn : Flash driver of drom does not support free_driver_priv()
Enter to exit`

Without the -c flag I get thr following, which looks good to me?

`E:\Simon\Documents\Code\hub_firmware\hub_esp32>C:\Users\Simon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin\openocd.exe -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 2000 kHz
ftdi samples TDO on falling edge of TCK
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Configured 2 cores
esp32 interrupt mask on
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : clock speed 2000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. PRO_CPU: PC=0x40167F02 (active)    APP_CPU: PC=0x40167F02
Info : Listening on port 3333 for gdb connections`

Any ideas?

igrr, 2020-03-31:
Hi @howroyd, which IDF version are you using?

howroyd, 2020-03-31:
I should be on ESP-IDF Release v4.0, but I am going to double check that now and make sure Windows hasn't reverted a path to an older version I still have on my harddrive

gerekon, 2020-03-31:
@howroyd Could you check your `OPENOCD_SCRIPTS` env variable? It should have value like `C:/Users/Simon/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/share/openocd/scripts`. 
Also you can try `C:\Users\Simon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin\openocd.exe -s C:/Users/Simon/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "program_esp build/blink.bin 0x10000 verify exit"`

howroyd, 2020-03-31:
`OPENOCD_SCRIPTS` had vanished so I have reinstated that and retried with the `-s` flag anyway in case windows is doing something odd.  Still getting the same error:

`E:\Simon\Documents\Code\hub_firmware\hub_esp32>C:\Users\Simon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin\openocd.exe -s "C:\Users\Simon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\share\openocd\scripts" -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "program_esp build\blink.bin 0x10000 verify exit"
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 2000 kHz
ftdi samples TDO on falling edge of TCK
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Configured 2 cores
esp32 interrupt mask on
invalid command name "program_esp"
Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of irom does not support free_driver_priv()
Warn : Flash driver of drom does not support free_driver_priv()

E:\Simon\Documents\Code\hub_firmware\hub_esp32>REM C:\Users\Simon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin\openocd.exe -s C:\Users\Simon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\share\openocd\scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg
Enter to exit`

As for my IDF version, I am actually not sure.  `idf.py --version` doesn't work and there is no hint given to how to get this on Google or `idf.py --help`

howroyd, 2020-03-31:
....checked using git.  I am on `release/v4.0`

howroyd, 2020-03-31:
Without the `-c` flag, and using GDB (through eclipse) to send the `program_esp` commands I get the following, if it is of any help?  The number of threads is interesting!

```
E:\Simon\Documents\Code\hub_firmware\hub_esp32>C:\Users\Simon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin\openocd.exe -s C:\Users\Simon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\share\openocd\scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 2000 kHz
ftdi samples TDO on falling edge of TCK
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Configured 2 cores
esp32 interrupt mask on
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : clock speed 2000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Detected debug stubs @ 3ffd2d1c on core0 of target 'esp32'
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Info : Target halted. PRO_CPU: PC=0x40167F02 (active)    APP_CPU: PC=0x40167F02
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40167F02
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 56 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 608 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40167F02
Info : Auto-detected flash size 4096 KB
Info : Using flash size 4096 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40167F02
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 56 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 608 KB
Info : Using flash size 612 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40167F02
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 56 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 608 KB
Info : Using flash size 60 KB
Warn : negative reply, retrying
Error: Too large number of threads 1073557624!
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
Info : esp32: Core 0 was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
invalid command name "program_esp"
invalid command name "program_esp"
invalid command name "program_esp"
invalid command name "program_esp"
Info : Detected debug stubs @ 3ffd2d1c on core0 of target 'esp32'
```

igrr, 2020-03-31:
Please check that you are using IDF documentation which matches the IDF version. In your case, it is https://docs.espressif.com/projects/esp-idf/en/release-v4.0/api-guides/jtag-debugging/index.html#upload-application-for-debugging. In v4.0, the command is `program_esp32`. It was renamed to `program_esp` and a more recent version. Looks like you were reading the docs for "latest" version, i.e. master branch.

howroyd, 2020-03-31:
Cannot thank you enough!  My apologies for being a complete fool and not RTFM!

Cheers

closed 2020-03-31 by howroyd