espressif/openocd-esp32 #105: Can't connect to ESP32-S2 with OpenOCD ? JTAG

csobrinho, 2020-04-20:
I tried to use the provided openocd but there doesn't seem to be any board config for the ESP32-S2. These **cfgs** seem to work:

**interface.cfg**
```
source [find interface/ftdi/esp32_devkitj_v1.cfg]
adapter_khz 26000
```

**board.cfg**
```
set ESP32_FLASH_VOLTAGE 3.3
set ESP32_RTOS freertos

source [find target/esp32s2.cfg]

# special function to program ESP32, it differs from the original 'program' that
# it verifies written image by reading flash directly, instead of reading memory mapped flash regions
proc program_esp32 {filename args} {
        program_esp $filename $args
}

add_help_text program_esp32 "write an image to flash, address is only required for binary images. verify, reset, exit are optional"
add_usage_text program_esp32 "<filename> \[address\] \[verify\] \[reset\] \[exit\]"
```

You also need a recent openocd, mine is:
```
Open On-Chip Debugger  v0.10.0-esp32-20200406 (2020-04-06-11:09)
```

igrr, 2020-04-20:
Hi @csobrinho, currently there are no board configs for ESP32-S2 because Espressif has not released any ESP32-S2 development boards yet which would have built-in JTAG adapters.

For now it is okay to use `-f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32s2.cfg`, if you are connecting an FT2232H based adapter to an ESP32-S2 dev board.

csobrinho, 2020-04-21:
hi @igrr, yes that's basically what I'm doing except the "target/esp32s2.cfg" has no `program_esp32` so maybe it can be added at the end? It's working well for me.

Only downside if that sometimes after a crash the core doesn't seem to properly reset (let's say this happens once every 1-2h) so I have to unplug the ESP32-S2 power and connect it again and the debug starts to work again.

igrr, 2020-04-21:
`program_esp32` has been replaced with non-target-specific `program_esp`, and it seems to be available for esp32s2, can you please check? `program_esp32` is still available for the ESP32, for compatibility.

Regarding the reset issue, could you try to capture the log file with debug enabled (`-d`)?

erhankur, 2023-01-17:
Closing this since not an issue anymore.

closed 2023-01-17 by erhankur