espressif/openocd-esp32 #106: Failed to connect gdb to target wrover kit

Wbaierl, 2020-05-01:
Hello, I'm new to development esp32.
I use the wrover kit 4.1 with Ubuntu18.04 on my pc.
when call idf.py gdb i get a timeout error. 
_This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
_Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
/home/wb/esp/hello_world/build/gdbinit:1: Error in sourced command file:
:3333: Die Wartezeit für die Verbindung ist abgelaufen.    <-- "wait time for the connection is exceeded"_

the same issue I also face when call idf.py openocd.
seems like there is an issue with the access to usb.
ttyUSB config is (according manuals) as follows:
crw-rw---- 1 root plugdev 188, 0 Mai  1 08:00 /dev/ttyUSB0
crw-rw---- 1 root plugdev 188, 1 Mai  1 08:12 /dev/ttyUSB1

anyone who encountered a similar problem when setting up debug for wrover kit?
thanks for your support.

igrr, 2020-05-01:
Could you please post the output you get when running `idf.py openocd`?

Wbaierl, 2020-05-02:
The folllowing output I get when running idf.py openocd:
__Executing action: openocd
Note: OpenOCD cfg not found (via env variable OPENOCD_COMMANDS nor as a --openocd-commands argument)
OpenOCD arguments default to: "-f board/esp32-wrover-kit-3.3v.cfg"
OpenOCD started as a background task 53327
Executing action: post_debug
Open On-Chip Debugger  v0.10.0-esp32-20200420 (2020-04-20-16:15)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'

openocd: src/jtag/core.c:343: jtag_checks: Assertion `jtag_trst == 0' failed.
"openocd" exited with -6
Please check JTAG connection!

igrr, 2020-05-02:
On Linux you may need to configure udev rules, please check: https://github.com/espressif/openocd-esp32/blob/1d00e3e6984868f008f3ecc2550454e67297fb86/README#L237

Wbaierl, 2020-05-03:
perfect, copy the rules file worked.

thanks for the quick support

closed 2020-05-04 by gerekon