espressif/openocd-esp32 #107: Launching debugger from eclipse - wrover 4.1 kit

Wbaierl, 2020-05-03:
After configuration of debugger according espressif docu
[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse](url)
**Start of the debug session fails with** 

_Error in final launch sequence:
Halt command not defined for device 'Generic TCP/IP'
Halt command not defined for device 'Generic TCP/IP'_

**The openocd output in the terminal shows:**
_Info : Target halted. CPU0: PC=0x400D29AE (active)
Info : Target halted. CPU1: PC=0x400E25E6 
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 22 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 75 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.flash' size 4096 KB
Info : Using flash bank 'esp32.flash' size 4096 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 22 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 75 KB
Info : Using flash bank 'esp32.irom' size 76 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 22 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 75 KB
Info : Using flash bank 'esp32.drom' size 24 KB
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : dropped 'gdb' connection
_

gerekon, 2020-05-04:
@Wbaierl Could you collect and post full OpenOCD and GDB logs as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb)?

Wbaierl, 2020-05-04:
Hello Alexej, 
following I added the logs:
1. JTAG adapter:  Wrover 4.1 Kit
2. latest release of esp-idf
3. Ubuntu 20.04
4. running native on PC

Launching openocd 
1) openocd -f board/esp32-wrover-kit-3.3v.cfg      -> openocd.log  does not contain any data
2) openocd -d 3 -f board/esp32-wrover-kit-3.3v.cfg    -> openocd.log contain the following error:
_User : 14 5 options.c:60 configuration_output_handler(): debug_level: 3
User : 15 5 options.c:60 configuration_output_handler():
Unexpected command line argument: 3_

Where/how to configure xtensa-esp32-elf-gdb -x gdbinit build/blink.elf in Eclipse to get the gdb_log?

Wolfgang

gerekon, 2020-05-04:
> openocd -d 3 -f board/esp32-wrover-kit-3.3v.cfg -> openocd.log contain the following error:

Sorry, there is a mistake in the docs.
You need to run `openocd -d3 -l openocd.log -f board/esp32-wrover-kit-3.3v.cfg`

> Where/how to configure xtensa-esp32-elf-gdb -x gdbinit build/blink.elf in Eclipse to get the gdb_log?

You need to add `set remotelogfile gdb_log.txt` as the first initialization command, just before `mon reset halt`. As described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse
) in step 8.

Wbaierl, 2020-05-04:
Hello Alexey, attached find the logfiles
[openocd.log](https://github.com/espressif/openocd-esp32/files/4576638/openocd.log)
[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/4576639/gdb_log.txt)

Wbaierl, 2020-05-05:
debugging from eclipse 20.04 with wrover kit 4.1 now works

closed 2020-05-05 by Wbaierl