espressif/openocd-esp32 #108: Debugger Wrover Kit fails -

Wbaierl, 2020-05-12:
I suddenly, after debugging worked before, a problem when I start the debugging.

Environment Ubuntu - Eclipse - Openocd

_Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections_

_Error: 384 365 target.c:588 target_halt(): Target not examined yet
Debug: 385 365 command.c:630 run_command(): Command 'halt' failed with error code -4
User : 386 365 target.c:4694 target_handle_event(): Error executing event gdb-attach on target esp32:_

[openocd.log](https://github.com/espressif/openocd-esp32/files/4613609/openocd.log)

gerekon, 2020-05-12:
@Wbaierl 

It seems like problem with accessing JTAG TAP.
Could you check the wires? Maybe you need to reconnect them.

```
Error: 327 31 core.c:1017 jtag_examine_chain_check(): JTAG scan chain interrogation failed: all ones
Error: 328 31 core.c:1018 jtag_examine_chain_check(): Check JTAG interface, timings, target power, etc.
Error: 329 31 core.c:1508 jtag_init_inner(): Trying to use configured scan chain anyway...
Debug: 330 31 core.c:1259 jtag_validate_ircapture(): IR capture validation scan
Error: 331 31 core.c:1311 jtag_validate_ircapture(): esp32.cpu0: IR capture error; saw 0x1f not 0x01
Debug: 332 31 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Warn : 333 31 core.c:1531 jtag_init_inner(): Bypassing JTAG setup events due to errors
```

Wbaierl, 2020-05-12:
I use the Wrover Kit 4.1. So there is only the USB connector. Other wiring to FTDI is on the board.

Wbaierl, 2020-05-12:
Seems the problem is related to the use of GPIO (I used GPIO 12, 14,26).
GPIO 12 and 14 are used for JTAG, that's the reason the debugger failed.

closed 2020-05-12 by Wbaierl