espressif/openocd-esp32 #109: Target not halted error

cadelasss, 2020-05-13:
Hi,
suddenly I started getting the error target not halted in openocd 2020/04/20

**Environment:**
ESP32-wroom-32D
JTAG Adapter: ESP-prog
Eclipse withe 
    
<pre>
> Executing C:\Users\Luis\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Luis\esp\esp-idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r1-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r1-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200420 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)
</pre>

Operating System: Windows 10 (OS Build 18367.778)
GDB lauch confguration on Eclipse

[image]
[image]
[image]

**Problem Description:**
I do the following steps:
Import the blink project to Eclipse
Build the program to ESP32 Thing module with Eclipse and USB Connect the ESP-prog
Run OpenOCD
Run xtensa-esp32-elf-gdb

Eclipse build console output
[Blink build.txt](https://github.com/espressif/openocd-esp32/files/4623281/Blink.build.txt)
**Last two lines**

> Generated C:/Users/Luis/eclipse-workspace/blink2/build/blink.bin
> Build complete (0 errors, 0 warnings): C:\Users\Luis\eclipse-workspace\blink2\build

I lanch the blink project in debug mode but Openocd says target not halted

<pre>
> Executing OpenOCD Debugger GDB... ["v0.10.0-esp32-20200420"]
> Open On-Chip Debugger  v0.10.0-esp32-20200420 (2020-04-20-16:15)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> adapter speed: 20000 kHz

> WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
> Info : Configured 2 cores
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
> Info : clock speed 20000 kHz
> Error: JTAG scan chain interrogation failed: all ones
> Error: Check JTAG interface, timings, target power, etc.
> Error: Trying to use configured scan chain anyway...
> Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
> Warn : Bypassing JTAG setup events due to errors
> Info : Listening on port 3333 for gdb connections
> Info : accepting 'gdb' connection on tcp/3333
> Error: No symbols for FreeRTOS
> Error: Target not examined yet
> Error executing event gdb-attach on target esp32:
> 
> Error: Target not halted
> Error: auto_probe failed
> Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
> Error: attempted 'gdb' connection rejected
</pre>

I run it now using  `openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -d3`
[Openocd -d3.txt](https://github.com/espressif/openocd-esp32/files/4623569/Openocd.-d3.txt)

After that, I launch Openocd with
`openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "gdb_memory_map disable"`

<pre>
Executing OpenOCD Debugger GDB... ["v0.10.0-esp32-20200420"]
Open On-Chip Debugger  v0.10.0-esp32-20200420 (2020-04-20-16:15)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Error: Target not examined yet
Error executing event gdb-attach on target esp32:

Error: FreeRTOS_update_threads not TARGET_HALTED!
Error: Target not examined yet
Error: Could not read FreeRTOS thread count from target
Error: Target not examined yet
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Error: esp32_soc_reset: Couldn't halt target before SoC reset

Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: esp32_soc_reset: Couldn't halt target before SoC reset
embedded:startup.tcl:449: Error: ** Unable to reset target **
in procedure 'program_esp'
in procedure 'program_error' called at file "C:/Users/Luis/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg", line 75
at file "embedded:startup.tcl", line 449
Error: Target not examined yet
Warn : target esp32 is not halted (add breakpoint)
Error: can't add breakpoint: target running</pre>

I run openocd with `openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "gdb_memory_map disable" -d3 `
[Openocd -d3 2.txt](https://github.com/espressif/openocd-esp32/files/4623535/Openocd.-d3.2.txt)

Now it connects but when I pause Eclipse to try to go step by step I get this error

[image]

gerekon, 2020-05-13:
Hi
It seems to be a problem with JTAG connection. Please, check the wrires and ensure that you do not overrdie function of JTAG pins. See https://github.com/espressif/openocd-esp32/issues/108#issuecomment-627524119

cadelasss, 2020-05-14:
ok it apears that I use one of the jtag port by mistake 
what should i do now to revert it
i use this comand but dosn't work `esptool.py --chip esp32 -p COM3 erase_flash`

gerekon, 2020-05-14:
>  i use this comand but dosn't work esptool.py --chip esp32 -p COM3 erase_flash

I talked about various GPIO pin functions which can be assigned by the code running on target. See 
https://github.com/espressif/esp-idf/blob/c1d0daf36d0dca81c23c226001560edfa51c30ea/components/soc/soc/esp32/include/soc/io_mux_reg.h#L205.

gerekon, 2020-05-14:
ESP32 Thing seems to have GPIO0 button for Download mode.

closed 2020-05-19 by cadelasss