espressif/openocd-esp32 #111: Any plan to support old ESP8266 chip?

puppywang, 2020-05-31:
There is an old openocd folk supporting ESP8266, which is based on old openocd v0.9 version, (https://github.com/sysprogs/esp8266-openocd). If we can merge that, it would be easier for enthusiast to debug RTOS on old ESP8266. Thanks.

gerekon, 2020-06-01:
Hi @puppywang. We are considering the possibility to support ESP8266 in this repo. The support for Xtensa-based chips with different HW options has been improved. See [xtensa.c](https://github.com/espressif/openocd-esp32/blob/master/src/target/xtensa.c) and [xtensa_debug_module.c](https://github.com/espressif/openocd-esp32/blob/master/src/target/xtensa_debug_module.c)
But it is not in our nearest plans and depends on how many people will actually need this.

puppywang, 2020-06-02:
Got, thanks for your information.

erhankur, 2023-01-17:
There is no update regarding ESP8266 support. Still waiting in the backlog and needs a priority.

closed 2023-01-17 by erhankur