espressif/openocd-esp32 #112: Eclipse debugging breakpoint problems

MPC-BlackBox, 2020-06-10:
Hi.
I am using the latest Ubuntu, esp-idf and eclipse with the expressif tools. Using the esp-prog jtag adapter.

I can connect to the ESP32 ok. Using the blink example I can program verify and run the program.

Inserting a break point at app_main does not work, this seems to be ignored. Also if the program is running and I insert a break point within app_main the program stops but eclipse shows the program as still running. I have to click on the suspend button which then enables the program step.

If I press step I have to click on the main thread, highlight the app_main and click suspend for eclipse to indicate the next step.

When I do suspend the debug window shows 8 threads active. Please see the sequence below.
[image]
[image]
[image]

This is the first time I have debugged a multi thread processor so it maybe my mis-understanding.

Finally I find it easier to use the external tools configuration in eclipse to start the openocd debugger. Settings I use are shown below.
[image]

Thanks,

MPC.

policeman0077, 2021-04-28:
I have similar breakpoint issues in vscode with ESP-IDF extension. When I debug the blink example, it can stop at the very first breakpoint. Then if I hit run it will never stop at any other breakpoint again. No idea if it's opened issues or the plug-in's problem though.

gerekon, 2021-04-28:
@MPC-BlackBox @policeman0077  Could you run OpenOCD with `-d3 -l oocd.log` options and post log here?

policeman0077, 2021-04-29:
@gerekon thanks Alexy for your quick reply. I am new to OpenOCD debugging and VScode. Where can I add the option in VScode?  I found in task.jason file, there is a command to start the OpenOCD should I padding those option to those command as shown in below picture:
[image]

policeman0077, 2021-04-29:
here is a screen recording which might be able to better explain my issue
[video](https://i.imgur.com/cewlEPa.mp4)

gerekon, 2021-04-29:
@policeman0077 Ok. I see. If OpenOCD halts in proper place, but VsCode does not it seems that some messages are lost at Debug Adapter layer. 
BTW, I can reproduce your problem with the latest IDF extension 1.0.3. There are several fixes are pended for Debug Adapter which solve your problem. I think we need to make new release of the extension with those fixes.

cc @brianignacio5

kolipakakondal, 2021-04-30:
Hello,

On the eclipse OpenOCD issue, please check with our recent releases since we have added **ESP-IDF GDB OpenOCD Debugging** to debug esp-idf applications, please refer to the doc here https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/OpenOCD%20Debugging.md

policeman0077, 2021-04-30:
@gerekon Thats great Alexey. Cant wait to try the new release. ：）

policeman0077, 2021-04-30:
@kolipakakondal thanks Kondal for the information. I will also try eclipse for esp32 development and debugging.

jurelak, 2021-05-07:
Hey, @policeman0077 did you managed to solve breakpoint issue ? I have the same problem and it's very anoying .

brianignacio5, 2021-05-07:
This issue should be fixed with [this PR](https://github.com/espressif/vscode-esp-idf-extension/pull/402)

jurelak, 2021-05-08:
@brianignacio5  Thank you It's working now with openocd version v0.10.0-esp32-20200709 but with latest version v0.10.0-esp32-20210401 debuging won't even start.

brianignacio5, 2021-05-10:
Thanks for testing this @jurelak ! Will debug openOCD 2021 and vscode extension together to take a deeper look at the issue. Maybe you could consider [openning an issue in vscode extension repository](https://github.com/espressif/vscode-esp-idf-extension/issues/new/choose) with any output or log from this test ?

erhankur, 2023-01-19:
Closing this since there is nothing expected from OpenOCD.

closed 2023-01-19 by erhankur