espressif/openocd-esp32 #113: JTAG scan chain interrogation failed: all ones (on previously functional setup)

v1993, 2020-06-13:
I've decided to try and debug my ESP32 chip once again, only to be welcomed by this:

```bash
v@v-home:~$ openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg
Open On-Chip Debugger  v0.10.0-esp32-20200420 (2020-04-20-16:15)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
```

I'm using ESP-32-WROOM chip on devkit and CJMCU-232H (FT232H) adapter.

I've tried disconnecting all connected devices from ESP32 and flashing hello_world example just to be sure to no avail. Moreover, some time ago I had no problems using this adapter and board with older version of openocd (sadly can't say which one, but I had to build it separately back then).

BassMatiFreecad, 2020-07-28:
I am working on an FT4232HL board and got it working on Windows 10. I stumbled across the same error. My solution was to revert the WinUSB drivers back to the FTDI drivers (using Windows Device Manager), then again back to WinUSB with Zadig. At some point I also uninstalled the FTDI drivers and reinstalled "from scratch".

Check if your JTAG interface is ok by connecting the TDO input to ground. You should get the same error, but with "all zeroes". If that is the case, the interface should be OK.

gerekon, 2020-07-29:
@v1993 The problem can be that you use wrong arguments for OpenOCD. Please, have a look at the doc https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/index.html#run-openocd

JimGaluska, 2020-12-30:
Tying GPIO pin 15 to ground during boot suppresses the boot messages sent out the serial port by the ROM bootloader ([link](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#other-pins)).

It also causes the exact same issue shown above.  

The fix is as easy as: Don't do that - Remove the jumper from GPIO 15 to ground and try it again.

JimGaluska, 2020-12-30:
Tying GPIO 15 to ground suppresses the messages sent out to the serial port by the ROM bootloader ([link](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#other-pins)).

It also causes the exact same issue shown above.  

The fix is as easy as: Don't do that - Remove the jumper from GPIO 15 to ground and try it again.

minaarmia, 2022-03-17:
Hey
I'm connecting FTDI chip between Raspberry Pi and stm32 discovery board and when I'm trying to run the cfg file I receive this
Open On-Chip Debugger 0.11.0+dev-00418-g918811529 (2021-10-12-13:34)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
ftdi layout_init data direction

(Urgently) Can anyone help

erhankur, 2023-01-17:
Closing due to inactivity.

closed 2023-01-17 by erhankur