espressif/openocd-esp32 #114: Add support to read ChipId through openocd

mikhail-khalizev, 2020-06-19:
Hi.

How I see, now only `program_esp32` procedure supported. It's good, but not enough.

For example now I need to read ChipId through JTAG. And I do not surprise if in the future someone need to read / write efuse.

Can you add support for this features to openocd?

igrr, 2020-06-19:
Hi @mikhail-khalizev, i think it should be possible to read the MAC address from eFuse using OpenOCD commands.

Read 6 bytes of the MAC address into a TCL array `mac`:
```tcl
mem2array mac 8 0x6001a004 6
```

Format `mac` array as a hexadecimal MAC address, and store result in `mac_str` variable.
```tcl
set mac_str [format %02x:%02x:%02x:%02x:%02x:%02x $mac(5) $mac(4) $mac(3) $mac(2) $mac(1) $mac(0)]
```

Depending on what you actually need to do with this information (log the MAC address into some file, embed MAC address into a binary, generate some MAC-dependent value), you may need to write a bit of TCL code to use these values.

If you simply need to display the MAC address, you can wrap the above lines into a function, save it in a `readmac.cfg` file, and then pass the cfg file to OpenOCD using an additional `-f readmac.cfg` command line argument.

```tcl
proc esp32_get_mac {} {
  mem2array mac 8 0x6001a004 6
  format %02x:%02x:%02x:%02x:%02x:%02x $mac(5) $mac(4) $mac(3) $mac(2) $mac(1) $mac(0)
}
```

Afterwards, use `esp32_get_mac` function.

I've logged a feature request to add something along these lines to `target/esp32.cfg` file.

mikhail-khalizev, 2020-06-19:
Thanks for reply. But seems `mem2array` not exists:

```
> openocd -f esp32-ft232h.cfg -c "mem2array mac 8 0x6001a004 6"
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
Info : Configured 2 cores
esp32 interrupt mask on
invalid command name "mem2array"
Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of irom does not support free_driver_priv()
Warn : Flash driver of drom does not support free_driver_priv()

> openocd -v
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
```

File `esp32-ft232h.cfg`:
```
interface ftdi
ftdi_vid_pid 0x0403 0x6014
ftdi_layout_init 0x0c08 0x0f1b

transport select jtag
adapter_khz 20000
source [find target/esp32.cfg]
```

igrr, 2020-06-19:
I think mem2array becomes available after `init` command, so you should do something like: `openocd -f board/esp32-ethernet-kit-3.3v.cfg -c 'init; mem2array mac 8 0x6001a004 6; puts [format %02x:%02x:%02x:%02x:%02x:%02x $mac(5) $mac(4) $mac(3) $mac(2) $mac(1) $mac(0)]; exit'`

mikhail-khalizev, 2020-06-19:
Yes, you right, after `init` command `mem2array` recognized successfully. But failed with error:
```
>openocd -f esp32-ft232h.cfg -c "init; mem2array mac 8 0x6001a004 6; puts [format %02x:%02x:%02x:%02x:%02x:%02x $mac(5) $mac(4) $mac(3) $mac(2) $mac(1) $mac(0)]; exit"
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
Info : Configured 2 cores
esp32 interrupt mask on
esp32_get_mac
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
Info : Listening on port 3333 for gdb connections
Error: mem2array: Read @ 0x6001a004, w=1, cnt=6, failed

Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of irom does not support free_driver_priv()
Warn : Flash driver of drom does not support free_driver_priv()

```

igrr, 2020-06-19:
This seems to be a bug in v0.10.0-esp32-20190313, which got fixed in one of the more recent versions. You can work around it by adding and additional `esp32 set_permissive 1;` command between `init` and `mem2array`.

mikhail-khalizev, 2020-06-19:
Thanks, this solution works.

closed 2020-06-19 by mikhail-khalizev