espressif/openocd-esp32 #115: openOCD - jtag debugging - Failed to write breakpoint instruction (-4)!

kifantidis, 2020-07-27:
Hello there folks !
My set-up in order to program esp32 is this:

-OS: windows 10 x64
-Eclipse
-Esspressif's plugin
-openOCD version: Tested many versions at the moment I'm using latest release (openocd-esp32-win32-0.10.0-esp32-20200709)
-Module: ESP-WROOM-32D and ESP32-WROVER (kit)
-Programmer/Debugger: ESP-PROG
-Flash Memory: 4MB -- FW installed in 0x30000 (3x OTA partitions)
-Jtag Frequency: Tested both in 20KHz and in 10KHz
-sdkconfigs are set accordingly

I have build my FW and it works great but it's time to use openOCD and xtensa debugger. When I firstly installed "ESP-IDF Eclipse plugin" in order to confirm that everything works I've compiled hello world and blink examples and everything was working fine. I've also confirmed that debugger was working.

At the moment I'm trying to use the debugger and I get this message and the programm stops:

    Info : esp32: Core 0 was reset.
    Info : esp32: Debug controller 1 was reset.
    Info : esp32: Core 1 was reset.
    Info : Target halted. CPU0: PC=0x40000400 (active)
    Info : Target halted. CPU1: PC=0x40000400
    Error: cpu0: Failed to write breakpoint instruction (-4)!
    Error: cpu0: Failed to add SW breakpoint!
    Error: can't add breakpoint: unknown reason

I'm posting my debug log files from both the devices. Also I'm attaching some screenshots.
I have confirmed this error in both esp32 chips and in both using the xtensa debugger via command line and within eclipse.

Is there anyone that had the same problem or anyone that can help me ?

Best regards, Kostas

[openocd_myLog.txt](https://github.com/espressif/openocd-esp32/files/4982279/openocd_myLog.txt)
[openocd_myLog2.txt](https://github.com/espressif/openocd-esp32/files/4982281/openocd_myLog2.txt)
[image]
[image]
[image]

gerekon, 2020-07-29:
Hi @kifantidis 

1) You need to run OpenOCD with correct arguments (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/index.html#run-openocd). This year we re-structured board and chip config files. For your board you should use `openocd -f board/esp32-wrover-kit-3.3v.cfg`.
2) I see that you are trying to debug FW at non-default flash location `0x30000`. To make this working you need to tell OpenOCD your custom app offset in flash using command `openocd -f board/esp32-wrover-kit-3.3v.cfg.cfg -c "init; halt; esp32 appimage_offset 0x30000"`. Please see details [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#flash-mappings-vs-sw-flash-breakpoints). Or you can disable flash support at all `openocd -c 'set ESP_FLASH_SIZE 0' -f board/esp32-wrover-kit-3.3v.cfg`, but in this case you will be limited to two breakpoints only.

PaulAnurag, 2022-07-29:
Hi Team

I am also having a similar issue while debugging esp32-wrover kit in eclipse.
I have attached a snippet of the error and also the log.

[image]
[openocd_ConsoleLog1.txt](https://github.com/espressif/openocd-esp32/files/9216601/openocd_ConsoleLog1.txt)

gerekon, 2022-08-01:
Hi @PaulAnurag 

I am going to close tis issue since there is no updates from the author for 2 years. The information you provided is not enough to diagnose the cause of the problem. So I'd ask you to open new issue and provide all necessary information as required by issue form.

closed 2022-08-01 by gerekon