espressif/openocd-esp32 #117: Esp-Prog ESP32 Flashing gives me error no -1401

lucaaAUT, 2020-08-02:
`Executing task in folder 200629-213850-arduino-blink: C:\Users\LucaMoser\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing esp32dev (platform: espressif32; framework: arduino; board: esp32dev)

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: 
PLATFORM: Espressif 32 1.12.4 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-openocd-esp32 1.1000.20190708 (10.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 26 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.7% (used 15348 bytes from 327680 bytes)
Flash: [==        ]  16.1% (used 211033 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esp-prog
Uploading .pio\build\esp32dev\firmware.bin
Open On-Chip Debugger  v0.10.0-esp32-20200526 (2020-05-26-09:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 20000 kHz

Warn : Transport "jtag" was already selected
User : 44 5 options.c:60 configuration_output_handler(): debug_level: 3
User : 45 5 options.c:60 configuration_output_handler():
Debug: 46 5 command.c:143 script_debug(): command - init init
Debug: 48 6 command.c:143 script_debug(): command - target target init
Debug: 50 6 command.c:143 script_debug(): command - target target names
Debug: 51 6 command.c:143 script_debug(): command - esp32 esp32 cget -event gdb-flash-erase-start
Debug: 52 6 command.c:143 script_debug(): command - esp32 esp32 configure -event gdb-flash-erase-start reset init
Debug: 53 7 command.c:143 script_debug(): command - esp32 esp32 cget -event gdb-flash-write-end
Debug: 54 7 command.c:143 script_debug(): command - esp32 esp32 configure -event gdb-flash-write-end reset halt
Debug: 55 7 command.c:143 script_debug(): command - esp32 esp32 cget -event gdb-attach
Debug: 56 7 target.c:1474 handle_target_init_command(): Initializing targets...
Debug: 57 7 xtensa_mcore.c:320 xtensa_mcore_target_init(): xtensa_mcore_target_init
Debug: 58 7 semihosting_common.c:98 semihosting_common_init():
Debug: 59 8 semihosting_common.c:98 semihosting_common_init():
Debug: 60 8 semihosting_common.c:98 semihosting_common_init():
Debug: 61 8 semihosting_common.c:98 semihosting_common_init():
Debug: 62 8 command.c:355 register_command_handler(): registering 'target_request'...
Debug: 63 8 command.c:355 register_command_handler(): registering 'trace'...
Debug: 64 8 command.c:355 register_command_handler(): registering 'trace'...
Debug: 65 8 command.c:355 register_command_handler(): registering 'fast_load_image'...
Debug: 66 9 command.c:355 register_command_handler(): registering 'fast_load'...
Debug: 67 9 command.c:355 register_command_handler(): registering 'profile'...
Debug: 68 9 command.c:355 register_command_handler(): registering 'virt2phys'...
Debug: 69 9 command.c:355 register_command_handler(): registering 'reg'...
Debug: 70 9 command.c:355 register_command_handler(): registering 'poll'...
Debug: 71 9 command.c:355 register_command_handler(): registering 'wait_halt'...
Debug: 72 9 command.c:355 register_command_handler(): registering 'halt'...
Debug: 73 9 command.c:355 register_command_handler(): registering 'resume'...
Debug: 74 9 command.c:355 register_command_handler(): registering 'reset'...
Debug: 75 10 command.c:355 register_command_handler(): registering 'soft_reset_halt'...
Debug: 76 10 command.c:355 register_command_handler(): registering 'step'...
[... 281 lines cut ...]
Debug: 365 596 xtensa.c:595 xtensa_smpbreak_set(): xtensa_smpbreak_set set smpbreak=30000, state=2
Debug: 366 597 command.c:143 script_debug(): command - esp32 esp32 invoke-event reset-init
Debug: 367 597 command.c:143 script_debug(): command - esp32 esp32 invoke-event reset-end
Debug: 368 599 command.c:143 script_debug(): command - echo echo ** Programming Started **
er : 370 601 command.c:770 jim_echo(): ** Programming Started 
Debug: 371 601 command.c:143 script_debug(): command - flash flash write_image erase .pio\build\esp32dev\firmware.bin 0x10000 verify
****Debug: 373 604 command.c:630 run_command(): Command 'flash write_image' failed with error code -1401**
**er : 374 604 command.c:695 command_run_line(): embedded:startup.tcl:449: Error: ** Programming Failed**** 
in procedure 'program_esp32' 
in procedure 'program_esp' called at file "C:/Users/LucaMoser/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp32.cfg", line 64
in procedure 'program_error' called at file "CxxUsers/LucaMoser/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 106
at file "embedded:startup.tcl", line 449
Debug: 375 605 esp_xtensa.c:213 esp_xtensa_target_deinit(): start
Debug: 376 605 xtensa.c:2157 xtensa_deinit(): start
Debug: 377 605 esp_xtensa.c:213 esp_xtensa_target_deinit(): start
Debug: 378 605 xtensa.c:2157 xtensa_deinit(): start
Debug: 379 606 target.c:2036 target_free_all_working_areas_restore(): freeing all working areas
Warn : 380 606 core.c:196 flash_free_all_banks(): Flash driver of esp32.flash does not support free_driver_priv()
Warn : 381 607 core.c:196 flash_free_all_banks(): Flash driver of esp32.irom does not support free_driver_priv()
Warn : 382 607 core.c:196 flash_free_all_banks(): Flash driver of esp32.drom does not support free_driver_priv()
*** [upload] Error 1
================================================================================================== [FAILED] Took 3.06 seconds ================================================================================================== 
The terminal process terminated with exit code: 1`

What does it mean greetz

gerekon, 2020-08-07:
@lucaaAUT 
This is [ERROR_IMAGE_TYPE_UNKNOWN](https://gitlab.espressif.cn:6688/idf/openocd-esp32/-/blob/master/src/target/image.h#L108 ) error.
The path to the binary below seems strange. Could you check it?
```
Debug: 371 601 command.c:143 script_debug(): command - flash flash write_image erase .pio\build\esp32dev\firmware.bin 0x10000 verify
```

gerekon, 2020-08-07:
Also that command looks strange. It should not have verify at the end.
Image should be [written](https://gitlab.espressif.cn:6688/idf/openocd-esp32/-/blob/master/tcl/target/esp_common.cfg#L84) with 
```
flash write_image [erase] [unlock] filename [offset] [type]
```
and then [verified](https://gitlab.espressif.cn:6688/idf/openocd-esp32/-/blob/master/tcl/target/esp_common.cfg#L89) with a separate command
```
 flash verify_bank num filename [offset]
```
Try to use 'program_esp' instead of 'program_esp32'

lucaaAUT, 2020-08-07:
Thanks for your input gerekon. 
Ill check that...- just need to work me into the Problem again.
Greets

lucaaAUT, 2020-08-08:
@gerekon

When i replace line 86 in esp_common.cfg:

`if {[catch {eval flash write_image erase $flash_args}] == 0} {`

With my hardcoded path:

`if {[catch {eval flash write_image erase C:/Users/LucaMoser/Documents/PlatformIO/Projects/200629-213850-arduino-blink/.pio/build/esp32dev/firmware.bin 0x10000}] == 0} {`

Then it works.

erhankur, 2023-01-17:
Closing due to inactivity. @lucaaAUT feel free to re-open if you have an issue with the latest OpenOCD release.

closed 2023-01-17 by erhankur