espressif/openocd-esp32 #118: ESP32-IDF v4.01: Failing to connect to open ocd in Visual Studio Code

tholtom, 2020-08-16:
I installed one of the esp32 examples in Visual Studio Code, then configured with the ESP32 add on.  I can successfully build and flash to my Espressif Vromer kit, from within the Visual Studio Code environment.

I now want to debug.  In a linux terminal window I entered:
openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg

And I connected without problems, getting the response messages indicated in Espresif's documentation.

Now when I try to OpenOCD from within Visual Studio Code, I get the response below.

Interestingly this is the response you get if you try the older (and incorrect with 4.0) openocd connect command from the command line.  

"openocd -f board/esp32-wrover.cfg" 

It's almost as if Espressif haven't modified their openocd connect script in Visual Studio Code for v4.0, and are still using the older format.  

Where would I have to modify the script so it calls the new form of openocd start?

Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:52)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
❌ Error: Debug adapter doesn't support any transports?
❌ embedded:startup.tcl:24: Error: 
in procedure 'script' 
at file "embedded:startup.tcl", line 63
in procedure 'transport' called at file "/home/jnarthur/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin/../share/openocd/scripts/board/esp32-wrover.cfg", line 29
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 24
[Stopped] : OpenOCD Server

an-dr, 2020-08-18:
Hi @tholtom!  It seems you haven't updated your openocd. For 4.0 we are using 20190708 
(https://docs.espressif.com/projects/esp-idf/en/release-v4.0/api-guides/jtag-debugging/index.html#run-openocd).
I think you could try to run the install script and the problem will gone
https://docs.espressif.com/projects/esp-idf/en/release-v4.0/get-started/index.html#id2

Please try and write some lines with results here

tholtom, 2020-08-18:
Hi an-dr.

I tried what you said.  Cloned the 4.01 (git clone -b v4.0.1 --recursive https://github.com/espressif/esp-idf.git) release and ran install and export.  When I enter: "openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg"

I get:
wroom-32.cfg
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:52)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
Info : Configured 2 cores
esp32 interrupt mask on
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Detected debug stubs @ 3ffb2958 on core0 of target 'esp32'
Info : Listening on port 3333 for gdb connections

So I'm getting a good connection to by board through oOpenOCD with version v0.10.0-esp32-20190313.

It looks like the 4.01 release of esp-idf has openocd 20190313, not the later one you are talking about. (20190708).

When I try to start OpenOCD **from VS Code**, with a project configured for esp32, I get the error:
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:52)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: Debug adapter doesn't support any transports?
embedded:startup.tcl:24: Error: 
in procedure 'script' 
at file "embedded:startup.tcl", line 63
in procedure 'transport' called at file "/home/jnarthur/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/share/openocd/scripts/board/esp32-wrover.cfg", line 29
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 24
[Stopped] : OpenOCD Server

Which is what you would get if you entered the older previous openocd command.

I guess my only options are to findout in which esp32 openocd script the command to open an openocd session is done for VS code, or to download idenpendently, openocd 20190708, and install it in the .espressif/tools folder.

Any suggestions here would be of great help...

an-dr, 2020-08-18:
Okey, just checked on my machine. It seems idf uses what it should
[image]

So, the problem could be in your environment variables. Please check either you envvars in visual studio (if you set them up) or global envvars for your machine. In the last case you can search `edit environment variables for your account` in the start menu

[image]

and check your `PATH` varibale
[image]

tholtom, 2020-08-18:
Hi an-dr,

I don't appear to have the facility to edit environment variables in my VS Code installation on Linux.

I had a look at my path with echo $PATH, and it looks like the earlier version is referenced.  This path is inserted into PATH when the ./install.sh and . ./export.sh scripts are run.

Looking at the actual installation in .espressif/tools/openocd-esp32 I have:
 v0.10.0-esp32-20190313

no other version of openOCD is resident on my machine.

I am going to try to find the archive on the espressif downloads for openOCD (v 20190708), and install that in the .expressif/tools/open-esp32 folder, then change the path and VS code environment to point to it.  That might get things going.

Thanks for the help,

Tim

an-dr, 2020-08-18:
Oh, for some reason I've decided that you are on Windows :) I'm biased a little) Well, yes, you can download it from https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20200709

Also you could check that lines in you esp-idf local directory
https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/release/v4.0/tools/tools.json#L210-213

Probably you need to pull the last commit from `release/v4.0` branch and it will fix the problem

gerekon, 2020-08-18:
@tholtom @an-dr Do I understand correctly that the problem happens because VsCode tries to run OpenOCD using new style command `openocd -f board/esp32-wrover.cfg`. If yes, this is wrong because `esp32-wrover.cfg` does not include interface file and can not be used alone. 
Instead  `board/esp32-wrover-kit-3.3v.cfg` or `board/esp32-wrover-kit-1.8v.cfg` files should be used which include all necessary stuff.

AFAIK VsCode IDF extension allows the configuration of OpenOCD config files to use in `.vscode/launch.json`. @an-dr Isn't it?

gerekon, 2020-08-18:
According to the [docs](https://github.com/espressif/vscode-esp-idf-extension/blob/a49ed5ace1fc775612eb4ee8ea59b556e054a5fa/docs/ONBOARDING.md#directly-modify-settingsjson) OpenOCD config files can be specified in `.vscode/settings.json`. 
So @tholtom you can try to set `idf.openOcdConfigs` in it like
```json
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "board/esp32-wrover.cfg"
    ],
```

gerekon, 2020-08-18:
@tholtom Starting from OpenOCD `v0.10.0-esp32-20191008` you can use new config files and have it like
```json
    "idf.openOcdConfigs": [
        "board/esp32-wrover-kit-3.3v.cfg"
    ],
```

gerekon, 2020-08-18:
@an-dr @brianignacio5 Could you check that VsCode uses proper combination OpenOCD config files. 
`openocd -f board/esp32-wrover.cfg` is invalid for any version of OpenOCD.

an-dr, 2020-08-18:
@gerekon Indeed there is one place [where used only `board/esp32-wrover.cfg`](https://github.com/espressif/vscode-esp-idf-extension/blob/master/package.json#L318) as a default value. I'll prepare a fix.

brianignacio5, 2020-08-19:
In the visual studio code extension the `idf.openOcdConfigs` has a default value of `["board/esp32-wrover.cfg"]` but when you run the command `ESP-IDF: Set Espressif device target` the value of `idf.openOcdConfigs` is ["interface/ftdi/esp32_devkitj_v1.cfg", "board/esp32-wrover.cfg"] when you select **esp32** as target and `["interface/ftdi/esp32_devkitj_v1.cfg", "target/esp32s2.cfg"]` for **esp32s2** as target.

tholtom, 2020-08-21:
Thanks guys for all your help! 

Setting idf.openOcdConfigs to use:  ["interface/ftdi/esp32_devkitj_v1.cfg", "board/esp32-wrover.cfg"]

did the trick and my VS code ide now successfully connects to OpenOCD.

BTW is there a list anywhere of all the possible idf config settings in .vscode/settings.json?  I added this one, and several more have obviously been added by the Espressif deploy process in settings.json.  Just wondered if there's a whole list of others...

Tim

raffaeler, 2022-03-01:
I still can't make it work.

The following command line works perfectly and it is pretty fast:
```
openocd -f board/esp32-wrover-kit-3.3v.cfg -c "program_esp build/blink.bin 0x10000 verify exit"
```

But when I use the following `settings.json`, the flashing procedure starts and than it hangs.
```
{
    "idf.adapterTargetName": "esp32",
    "idf.openOcdDebugLevel": 2,
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "board/esp32-wrover-kit-3.3v.cfg"
    ],
    "cmake.configureOnOpen": false,
}
```

Hints?

raffaeler, 2022-03-01:
Hoping to help someone else...
My issue was a bug of the VSCode Espressif extension (I had the 1.3.0).
I submitted a bug there and the issue has been already **solved with version 1.4.0** (published 3 hours ago).

closed 2023-01-17 by erhankur