espressif/openocd-esp32 #120: Unable to debug with external SPIRAM / JTAG

te-labs, 2020-09-28:
Hello,

I can't debug when CONFIG_ESP32_SPIRAM_SUPPORT is set. As you can see on the log below, I never reach the first breakpoint located in app_main. There is no more log after the bootloader stage.
I debug without problem when CONFIG_ESP32_SPIRAM_SUPPORT is not set.

**Setup**
- ESP-IDF v4.1
- ESP32-WROVER-IE (FLASH 16MB)
- ESP-PROG JTAG
- openocd v0.10.0-esp32-20200709

**OpenOCD**

```console
$ openocd.exe -f board/esp32-wrover-kit-1.8v.cfg
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : clock speed 5000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. CPU0: PC=0x400D755B (active)
Info : Target halted. CPU1: PC=0x00000000
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 139 KB
Info : Flash mapping 1: 0x40020 -> 0x400d0020, 698 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.flash' size 16384 KB
Info : Using flash bank 'esp32.flash' size 16384 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 139 KB
Info : Flash mapping 1: 0x40020 -> 0x400d0020, 698 KB
Info : Using flash bank 'esp32.irom' size 700 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 139 KB
Info : Flash mapping 1: 0x40020 -> 0x400d0020, 698 KB
Info : Using flash bank 'esp32.drom' size 140 KB
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400
** Programming Started **
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
** Programming Finished **
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[... 237 lines cut ...]
I (854) cpu_start: Application information:
I (858) cpu_start: Project name:     my-app
I (864) cpu_start: App version:      30eda54-dirty
I (870) cpu_start: Compile time:     Sep 28 2020 19:08:59
I (876) cpu_start: ELF file SHA256:  cac28c16f92daf46...
I (882) cpu_start: ESP-IDF:          v4.1
I (887) cpu_start: Single core mode
I (1790) spiram: SPI SRAM memory test OK
I (1790) heap_init: Initializing. RAM available for dynamic allocation:
I (1791) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1797) heap_init: At 3FFBCA68 len 00023598 (141 KiB): DRAM
I (1803) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (1810) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (1816) heap_init: At 400992C8 len 00006D38 (27 KiB): IRAM
I (1822) cpu_start: Pro cpu start user code
I (1827) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1848) spi_flash: detected chip: gd
I (1848) spi_flash: flash io: dio
I (1848) cpu_start: Starting scheduler on PRO CPU.
I (1851) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
```

Any idea?

Thanks

te-labs, 2020-09-29:
Problem solved after unset the option CONFIG_FREERTOS_UNICORE

Could someone explain why this option affect the debug?

BbIKTOP, 2021-11-05:
Any news regarding this? have the same problem