espressif/openocd-esp32 #123: Unexpected device found with the ESP32-PICO-D4

JarrettR, 2020-10-22:
I'm using a custom board with the bare ESP32-PICO-D4 on it, and the ESP-PROG for my programmer.

Results on an internet search for 0x240068cb / Dosilicon and ESP32 seem sparse, so it doesn't look like anyone else has had this issue.

```
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
Info : Configured 2 cores
Info : tcl server disabled
Info : telnet server disabled
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x240068cb (mfg: 0x465 (Dosilicon Co., Ltd.), part: 0x4006, ver: 0x2)
Warn : JTAG tap: esp32.cpu0       UNEXPECTED: 0x240068cb (mfg: 0x465 (Dosilicon Co., Ltd.), part: 0x4006, ver: 0x2)
Error: JTAG tap: esp32.cpu0  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x240068cb (mfg: 0x465 (Dosilicon Co., Ltd.), part: 0x4006, ver: 0x2)
Warn : JTAG tap: esp32.cpu1       UNEXPECTED: 0x240068cb (mfg: 0x465 (Dosilicon Co., Ltd.), part: 0x4006, ver: 0x2)
Error: JTAG tap: esp32.cpu1  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x03 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : esp32: Core 0 was reset.
Info : accepting 'gdb' connection from pipe
Error: No symbols for FreeRTOS
Info : Halt timed out, wake up GDB.
Error: timed out while waiting for target halted / 3 - 2
Error executing event gdb-attach on target esp32:

Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Error: error during select: Unknown error
.pioinit:11: Error in sourced command file:
Remote communication error.  Target disconnected.: No error.

```

Full disclosure, I'm using PlatformIO to initiate debugging, but I've also updated the built-in OpenOCD with the latest release from this repo, with no change.
I've also attempted to change the hardcoded 0x120034e5 device ID to the one that gets detected, but more /different errors happen then, implying that there are more configuration settings that need to be changed.

Serial uploading works, printing `Chip is ESP32D0WDQ5 (revision 1)`

gerekon, 2020-10-22:
Hi @JarrettR . Looks like JTAG communication problem. I suggest
1) Ensure that your program does not override functions of JTAG pins. https://github.com/espressif/esp-idf/blob/c1d0daf36d0dca81c23c226001560edfa51c30ea/components/soc/soc/esp32/include/soc/io_mux_reg.h#L198-L232
2) Try to lower adapter speed to smth like 1000. https://github.com/espressif/openocd-esp32/blob/master/tcl/interface/ftdi/esp32_devkitj_v1.cfg#L36
3) Run openocd manually `$OOCD_DISTRO/bin/openocd -s $OOCD_DISTRO/share/openocd/scripts -f board/esp32-wrover-kit-3.3v.cfg`. `OOCD_DISTRO` is the path to OpenOCD distribution.

JarrettR, 2020-10-23:
Lowering the adaptor speed helped, thanks a lot!

JTAG does some very unintuitive things with marginal electrical conditions

closed 2020-10-23 by JarrettR

mykmelez, 2023-07-28:
I experienced this problem with an ESP-PROG and an ESP32-S3 on a custom board. Lowering the adaptor speed resolved the problem, although I was also able to resolve it by following the guidance in this "info" message OpenOCD prints:

> Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"