espressif/openocd-esp32 #124: Debug on DevKitC V4 (PlatformIO) failing

Miq1, 2020-10-25:
I am trying to debug a DevKitC V4 with PlatformIO and openocd, JTAG adaptor is a genuine esp-prog. I can not get it to work. Here is an example debug output:
```
Processing az-delivery-devkit-v4 (platform: espressif32; board: az-delivery-devkit-v4; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html
PLATFORM: Espressif 32 (2.0.0) > AZ-Delivery ESP-32 Dev Kit C V4
HARDWARE: ESP32 240MHz, 520KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-openocd-esp32 1.1000.20190709 (10.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SimpleButton>
|   |-- <Wire> 1.0.1
|-- <Adafruit GFX Library> 1.10.0
|   |-- <Adafruit BusIO> 1.4.1
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Adafruit SSD1306> 2.3.1
|   |-- <Adafruit GFX Library> 1.10.0
|   |   |-- <Adafruit BusIO> 1.4.1
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <Adafruit BusIO> 1.4.1
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Ticker> 3.1.5
|-- <ArduinoOTE> 1.0
|   |-- <Ethernet> 2.0.0+sha.c9399a9
|   |   |-- <SPI> 1.0
|   |-- <Update> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <AsyncTCP> 1.1.1
|-- <Ethernet> 2.0.0+sha.c9399a9
|   |-- <SPI> 1.0
|-- <WiFi> 1.0
|-- <ModbusUnified> 0.0.1
|   |-- <AsyncTCP> 1.1.1
|   |-- <Ethernet> 2.0.0+sha.c9399a9
|   |   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
Building in debug mode
Compiling .pio\build\az-delivery-devkit-v4\src\main.cpp.o
Linking .pio\build\az-delivery-devkit-v4\firmware.elf
Retrieving maximum program size .pio\build\az-delivery-devkit-v4\firmware.elf
Checking size .pio\build\az-delivery-devkit-v4\firmware.elf
Building .pio\build\az-delivery-devkit-v4\firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   7.4% (used 39504 bytes from 532480 bytes)
Flash: [=====     ]  54.9% (used 718978 bytes from 1310720 bytes)
esptool.py v2.6
========================= [SUCCESS] Took 16.69 seconds =========================
Reading symbols from c:\Users\Micha\Documents\PlatformIO\Projects\Unibridge\.pio\build\az-delivery-devkit-v4\firmware.elf...
done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = esp-prog
[... 233 lines cut ...]
There are some output lines I am wondering about:
```
Error: Too large number of threads 1073465900!
```
and 
```
** Programming Started **
** Programming Started **
embedded:startup.tcl:449: Error: ** Programming Failed **
in procedure 'program_esp32'
in procedure 'program_esp' called at file "C:/Users/Micha/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp32.cfg", line 64
in procedure 'program_error' called at file "C:/Users/Micha/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 106
at file "embedded:startup.tcl", line 449
embedded:startup.tcl:449: Error: ** Programming Failed **
in procedure 'program_esp32'
in procedure 'program_esp' called at file "C:/Users/Micha/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp32.cfg", line 64
in procedure 'program_error' called at file "C:/Users/Micha/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 106
at file "embedded:startup.tcl", line 449
```
Why is that seemingly called twice? And why does it fail?

What I tried so far, but to no avail:
- lower the speed down to 1000 kHz
- install the latest openocd from here
- check the wiring and power supply

gerekon, 2020-10-26:
Hi @Miq1 

Could you run OpenOCD with `-d3` option to get verbose log? Unfortunately I do not know how to do this `platformio`

Miq1, 2020-10-26:
I will try today and report back if I have the verbose log.

Miq1, 2020-10-26:
FYI: the easiest way to activate the verbose log was to add ``debug_level 3`` in the ``esp-wroom-32.cfg`` file.

That was quite a lot of log now 😉 

```
Processing az-delivery-devkit-v4 (platform: espressif32; board: az-delivery-devkit-v4; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html
PLATFORM: Espressif 32 (2.0.0) > AZ-Delivery ESP-32 Dev Kit C V4
HARDWARE: ESP32 240MHz, 520KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-openocd-esp32 1.1000.20190709 (10.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SimpleButton>
|   |-- <Wire> 1.0.1
|-- <Adafruit GFX Library> 1.10.0
|   |-- <Adafruit BusIO> 1.4.1
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Adafruit SSD1306> 2.3.1
|   |-- <Adafruit GFX Library> 1.10.0
|   |   |-- <Adafruit BusIO> 1.4.1
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <Adafruit BusIO> 1.4.1
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Ticker> 3.1.5
|-- <ArduinoOTE> 1.0
|   |-- <Ethernet> 2.0.0+sha.c9399a9
|   |   |-- <SPI> 1.0
|   |-- <Update> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <AsyncTCP> 1.1.1
|-- <Ethernet> 2.0.0+sha.c9399a9
|   |-- <SPI> 1.0
|-- <WiFi> 1.0
|-- <ModbusUnified> 0.0.1
|   |-- <AsyncTCP> 1.1.1
|   |-- <Ethernet> 2.0.0+sha.c9399a9
|   |   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
Building in debug mode
Compiling .pio\build\az-delivery-devkit-v4\src\main.cpp.o
Linking .pio\build\az-delivery-devkit-v4\firmware.elf
Building .pio\build\az-delivery-devkit-v4\firmware.bin
Retrieving maximum program size .pio\build\az-delivery-devkit-v4\firmware.elf
Checking size .pio\build\az-delivery-devkit-v4\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   7.4% (used 39504 bytes from 532480 bytes)
Flash: [=====     ]  54.9% (used 718978 bytes from 1310720 bytes)
esptool.py v2.6
========================= [SUCCESS] Took 16.32 seconds =========================
Reading symbols from c:\Users\Micha\Documents\PlatformIO\Projects\Unibridge\.pio\build\az-delivery-devkit-v4\firmware.elf...
[... 582 lines cut ...]
Debug: 801 161116 xtensa.c:618 xtensa_core_status_check(): cpu0: DSR (8000C000)
Debug: 802 161116 target.c:2003 target_free_working_area_restore(): freed 124 bytes of working area at address 0x3ffc0000
Debug: 803 162110 target.c:1777 print_wa_layout():    0x3ffc0000-0x3ffc5fff (24576 bytes)
Error: 804 162111 xtensa_algorithm.c:617 xtensa_run_do(): Algorithm run failed (-4)!
Debug: 805 162111 xtensa_mcore.c:837 xtensa_mcore_run_func_image(): esp32: check/clear DSR
Debug: 806 162112 log.c:409 keep_alive(): keep_alive() was not invoked in the 1000ms timelimit (1001). This may cause trouble with GDB connections.
Debug: 808 162113 xtensa.c:618 xtensa_core_status_check(): cpu1: DSR (80800000)
Error: 809 162113 esp_xtensa.c:246 esp_xtensa_get_size(): Failed to run flasher stub (-4)!
Error: 810 163112 esp_xtensa.c:775 esp_xtensa_probe(): Failed to probe flash, size 0 KB
Error: 811 163112 core.c:263 get_flash_bank_by_num(): auto_probe failed
Error: 812 163112 gdb_server.c:1005 gdb_new_connection(): Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: 813 163113 server.c:124 add_connection(): attempted 'gdb' connection rejected
Error: 814 163113 server.c:501 server_loop(): error during select: Unknown error
Debug: 815 163113 esp_xtensa.c:213 esp_xtensa_target_deinit(): start
Info : 816 163114 esp_xtensa.c:64 esp_xtensa_dbgstubs_restore(): cpu0: Restore debug stubs address 3ffc1e78
Debug: 817 163114 log.c:409 keep_alive(): keep_alive() was not invoked in the 1000ms timelimit (1002). This may cause trouble with GDB connections.
Debug: 819 163115 xtensa.c:2157 xtensa_deinit(): start
Debug: 820 163115 esp_xtensa.c:213 esp_xtensa_target_deinit(): start
Info : 821 164114 esp_xtensa.c:64 esp_xtensa_dbgstubs_restore(): cpu1: Restore debug stubs address 3ffc1e78
Debug: 822 164115 log.c:409 keep_alive(): keep_alive() was not invoked in the 1000ms timelimit (1001). This may cause trouble with GDB connections.
Debug: 824 164116 xtensa.c:2157 xtensa_deinit(): start
Debug: 825 164117 target.c:2036 target_free_all_working_areas_restore(): freeing all working areas
.pioinit:11: Error in sourced command file:
Remote communication error.  Target disconnected.: No error.
```

gerekon, 2020-10-26:
@Miq1 It seems that now due to verbose output and low adapter freq (1000 KHz) you have errors due to timeouts. Could you increase adapter speed to 5000 KHz or 10000 KHz and ensure that log contains `** Programming Failed **` string as your initial one does?

gerekon, 2020-10-26:
@Miq1 BTW it could be the same problem described [here](https://github.com/espressif/openocd-esp32/issues/125#issuecomment-716494059)

Miq1, 2020-10-26:
Raising the adapter frequency to 20000 did not change anything in the output. I experimentally set ``debug_level 2`` and got the ``** Programming failed **`` but unfortunately no additional log output.

What else could I try?

gerekon, 2020-10-26:
Try these steps https://github.com/espressif/openocd-esp32/issues/125#issuecomment-716504583

Miq1, 2020-10-26:
I am afraid I could not transfer the command line call you mentioned in the cited thread to my system. The oocd.log is quite short:
```
WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
Error: Debug adapter doesn't support any transports?
..\share\openocd\scripts/target/esp32.cfg:2: Error: 
in procedure 'script' 
at file "embedded:startup.tcl", line 26
at file "..\share\openocd\scripts\board\esp-wroom-32.cfg", line 3
at file "..\share\openocd\scripts/target/esp32.cfg", line 2
``´
I got a WROOM-32 board, no WROVER.

Next I tried to change the call in PlatformIO main.py from ``program_esp32`` to ``program_esp`` as advised in the other thread, but it did not change the result, unfortunately.

Miq1, 2020-10-27:
This was caused by PlatformIO not correctly handling the recent versions of open-ocd. Using the latest development builds of PlatformIO core and esp32 platform solved the issue.

closed 2020-10-27 by Miq1