espressif/openocd-esp32 #125: Unable to compile on mac os 10.15.6

lienbacher, 2020-10-26:
I am trying to compile openocd-esp32 because the latest release seems incompatible with V3 silicon (I'm using a ESP32-Pico-V3). When I run the commands as per [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/building-openocd-macos.html#building-openocd-from-sources-for-macos) I get this error in make:
```
-MF .deps/libjaylink_la-discovery_tcp.Tpo -c -o libjaylink_la-discovery_tcp.lo `test -f 'discovery_tcp.c' || echo './'`discovery_tcp.c
discovery_tcp.c:305:9: error: implicit declaration of function 'select' is
      invalid in C99 [-Werror,-Wimplicit-function-declaration]
                ret = select(sock + 1, &rfds, NULL, NULL, &timeout);
                      ^
1 error generated.
make[4]: *** [libjaylink_la-discovery_tcp.lo] Error 1
make[3]: *** [all-recursive] Error 1
make[2]: *** [all] Error 2
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

I found a possible solution suggesting adding `#include <sys/select.h>` to `src/jtag/drivers/libjaylink/libjaylink/discovery_tcp.c`, which solves this error, but then I am presented with 12 new errors:
```
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl -I/usr/local/Cellar/hidapi/0.9.0/include/hidapi -I./src/jtag/drivers/libjaylink/libjaylink -I./src/jtag/drivers/libjaylink -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Werror -g -O2 -MT src/jtag/drivers/libocdjtagdrivers_la-jtag_esp_remote.lo -MD -MP -MF src/jtag/drivers/.deps/libocdjtagdrivers_la-jtag_esp_remote.Tpo -c src/jtag/drivers/jtag_esp_remote.c -o src/jtag/drivers/libocdjtagdrivers_la-jtag_esp_remote.o
src/jtag/drivers/jtag_esp_remote.c:114:42: error: redefinition of 'usb_device' as different kind of symbol
static struct jtag_libusb_device_handle *usb_device = NULL;
                                         ^
/usr/local/include/usb.h:331:20: note: previous definition is here
struct usb_device *usb_device(usb_dev_handle *dev);
                   ^
src/jtag/drivers/jtag_esp_remote.c:144:26: error: incompatible pointer types passing 'struct usb_device *(usb_dev_handle *)' (aka 'struct usb_device *(struct usb_dev_handle *)') to parameter of type 'struct usb_dev_handle *'
      [-Werror,-Wincompatible-pointer-types]
                jtag_libusb_bulk_write(usb_device,
                                       ^~~~~~~~~~
src/jtag/drivers/libusb0_common.h:68:62: note: passing argument to parameter 'dev' here
int jtag_libusb_bulk_write(struct jtag_libusb_device_handle *dev, int ep,
                                                             ^
src/jtag/drivers/jtag_esp_remote.c:168:6: error: address of function 'usb_device' will always evaluate to 'true' [-Werror,-Wpointer-bool-conversion]
        if (usb_device) {
        ~~  ^~~~~~~~~~
src/jtag/drivers/jtag_esp_remote.c:168:6: note: prefix with the address-of operator to silence this warning
        if (usb_device) {
            ^
            &
src/jtag/drivers/jtag_esp_remote.c:191:37: error: incompatible pointer types passing 'struct usb_device *(usb_dev_handle *)' (aka 'struct usb_device *(struct usb_dev_handle *)') to parameter of type 'struct usb_dev_handle *'
      [-Werror,-Wincompatible-pointer-types]
                                size_t n= jtag_libusb_bulk_read(usb_device,
                                                                ^~~~~~~~~~
src/jtag/drivers/libusb0_common.h:70:61: note: passing argument to parameter 'dev' here
int jtag_libusb_bulk_read(struct jtag_libusb_device_handle *dev, int ep,
                                                            ^
src/jtag/drivers/jtag_esp_remote.c:625:44: error: incompatible pointer types passing 'struct usb_device *(*)(usb_dev_handle *)' (aka 'struct usb_device *(*)(struct usb_dev_handle *)') to parameter of type
      'struct usb_dev_handle **' [-Werror,-Wincompatible-pointer-types]
        int r= jtag_libusb_open(vids, pids, NULL, &usb_device);
                                                  ^~~~~~~~~~~
src/jtag/drivers/libusb0_common.h:63:38: note: passing argument to parameter 'out' here
                struct jtag_libusb_device_handle **out);
                                                   ^
src/jtag/drivers/jtag_esp_remote.c:634:32: error: incompatible pointer types passing 'struct usb_device *(usb_dev_handle *)' (aka 'struct usb_device *(struct usb_dev_handle *)') to parameter of type 'usb_dev_handle *'
      (aka 'struct usb_dev_handle *') [-Werror,-Wincompatible-pointer-types]
        jtag_libusb_set_configuration(usb_device, USB_CONFIGURATION);
                                      ^~~~~~~~~~
src/jtag/drivers/libusb0_common.h:72:62: note: passing argument to parameter 'devh' here
int jtag_libusb_set_configuration(jtag_libusb_device_handle *devh,
                                                             ^
src/jtag/drivers/jtag_esp_remote.c:635:6: error: implicit declaration of function 'libusb_kernel_driver_active' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        if (libusb_kernel_driver_active(usb_device, USB_INTERFACE))
            ^
src/jtag/drivers/jtag_esp_remote.c:635:6: note: did you mean 'jtag_libusb_kernel_driver_active'?
src/jtag/drivers/libusb0_common.h:44:19: note: 'jtag_libusb_kernel_driver_active' declared here
static inline int jtag_libusb_kernel_driver_active(jtag_libusb_device_handle *devh, int iface)
                  ^
src/jtag/drivers/jtag_esp_remote.c:636:3: error: implicit declaration of function 'libusb_detach_kernel_driver' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                libusb_detach_kernel_driver(usb_device, USB_INTERFACE);
                ^
src/jtag/drivers/jtag_esp_remote.c:636:3: note: did you mean 'usb_detach_kernel_driver_np'?
/usr/local/include/usb.h:323:5: note: 'usb_detach_kernel_driver_np' declared here
int usb_detach_kernel_driver_np(usb_dev_handle *dev, int interface);
    ^
src/jtag/drivers/jtag_esp_remote.c:637:30: error: incompatible pointer types passing 'struct usb_device *(usb_dev_handle *)' (aka 'struct usb_device *(struct usb_dev_handle *)') to parameter of type 'usb_dev_handle *'
      (aka 'struct usb_dev_handle *') [-Werror,-Wincompatible-pointer-types]
        jtag_libusb_claim_interface(usb_device, USB_INTERFACE);
                                    ^~~~~~~~~~
src/jtag/drivers/libusb0_common.h:49:74: note: passing argument to parameter 'devh' here
static inline int jtag_libusb_claim_interface(jtag_libusb_device_handle *devh,
                                                                         ^
src/jtag/drivers/jtag_esp_remote.c:655:6: error: address of function 'usb_device' will always evaluate to 'true' [-Werror,-Wpointer-bool-conversion]
        if (usb_device) {
        ~~  ^~~~~~~~~~
src/jtag/drivers/jtag_esp_remote.c:655:6: note: prefix with the address-of operator to silence this warning
        if (usb_device) {
            ^
            &
src/jtag/drivers/jtag_esp_remote.c:656:33: error: incompatible pointer types passing 'struct usb_device *(usb_dev_handle *)' (aka 'struct usb_device *(struct usb_dev_handle *)') to parameter of type 'usb_dev_handle *'
      (aka 'struct usb_dev_handle *') [-Werror,-Wincompatible-pointer-types]
                jtag_libusb_release_interface(usb_device, USB_INTERFACE);
                                              ^~~~~~~~~~
src/jtag/drivers/libusb0_common.h:55:76: note: passing argument to parameter 'devh' here
static inline int jtag_libusb_release_interface(jtag_libusb_device_handle *devh,
                                                                           ^
src/jtag/drivers/jtag_esp_remote.c:657:21: error: incompatible pointer types passing 'struct usb_device *(usb_dev_handle *)' (aka 'struct usb_device *(struct usb_dev_handle *)') to parameter of type 'usb_dev_handle *'
      (aka 'struct usb_dev_handle *') [-Werror,-Wincompatible-pointer-types]
                jtag_libusb_close(usb_device);
                                  ^~~~~~~~~~
src/jtag/drivers/libusb0_common.h:64:51: note: passing argument to parameter 'dev' here
void jtag_libusb_close(jtag_libusb_device_handle *dev);
                                                  ^
12 errors generated.
make[2]: *** [src/jtag/drivers/libocdjtagdrivers_la-jtag_esp_remote.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

Any hints how I could fix this? Both ./bootstrap and ./config run fine ...

igrr, 2020-10-26:
Hi @lienbacher, support for ESP32 ECO3 has been added in release https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20191008. The latest release, https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20200709, should include that as well.

As to the compilation issues, thanks for reporting, will investigate. We cross-compile macOS releases of OpenOCD on Linux, which is why we didn't meet the same issue with `usb_device` redeclaration.

lienbacher, 2020-10-26:
Thank you @igrr! 
The reason I'm trying to compile is because I am unable to get it running using platformio: https://community.platformio.org/t/esp32-pico-v3-fails-to-upload-debug-using-esp-prog/16734

gerekon, 2020-10-26:
> Thank you @igrr!
> The reason I'm trying to compile is because I am unable to get it running using platformio: https://community.platformio.org/t/esp32-pico-v3-fails-to-upload-debug-using-esp-prog/16734
> 
> This is the error I am getting using the latest release:
> 
> ```
> WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
> ** Programming Started **
> embedded:startup.tcl:449: Error: ** Programming Failed **
> in procedure 'program_esp32' 
> in procedure 'program_esp' called at file "/Users/wolfgang/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp32.cfg", line 64
> in procedure 'program_error' called at file "/Users/wolfgang/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 106
> at file "embedded:startup.tcl", line 449
> ```

Hi @lienbacher 

Could you run OpenOCD with `-d3` option and collect  more verbose log?

lienbacher, 2020-10-26:
I would love to, however I am using the platformio plugin for vscode, and I have no idea how to add this option to the call :(

gerekon, 2020-10-26:
> I would love to, however I am using the platformio plugin for vscode, and I have no idea how to add this option to the call :(

You can check this manually. 
1) Download latest OpenOCD release.
2) Run it as `$OOCD_DISTRO_PATH/bin/openocd -s $OOCD_DISTRO_PATH/share/openocd/scripts -f board/esp32-wrover-kit-3.3v.cfg -d3 -l oocd.log -c "program_esp app.bin 0x10000 verify exit"` 
3) Post `oocd.log` here

lienbacher, 2020-10-26:
Thank you! I have manually changed the path because i have no actual openocd install apart from what comes with platformio, but it ran, here's the log:

[oocd.log](https://github.com/espressif/openocd-esp32/files/5438638/oocd.log)

gerekon, 2020-10-26:
It seems that everything worked well
```
Debug: 2354 12807 command.c:143 script_debug(): command - echo echo ** Programming Finished **
User : 2356 12807 command.c:770 jim_echo(): ** Programming Finished **
Debug: 2357 12807 command.c:143 script_debug(): command - echo echo ** Verify Started **
User : 3162 18796 command.c:770 jim_echo(): ** Verified OK **
```

lienbacher, 2020-10-26:
My day is not going very well today XD
So the problem is exclusive to platformio i guess and not related to openocd 
thank you very much!

gerekon, 2020-10-26:
You can also check that it works with legacy `program_esp32`. Because PlatformIO uses it instead of new `program_esp`

`$OOCD_DISTRO_PATH/bin/openocd -s $OOCD_DISTRO_PATH/share/openocd/scripts -f board/esp32-wrover-kit-3.3v.cfg -d3 -l oocd.log -c "program_esp32 app.bin 0x10000 verify exit"`

lienbacher, 2020-10-26:
Seems this command also worked. It did not close gracefully in the terminal though, had to Ctrl-C it, but it seems it uploaded fine. Attached is the log: 
[oocd.log](https://github.com/espressif/openocd-esp32/files/5439154/oocd.log)

I have since also managed to run openocd from within vscode and got the probler error output: https://pastebin.com/SfG1Wi6k

gerekon, 2020-10-26:
It reports error `ERROR_IMAGE_TYPE_UNKNOWN`. It seems that `write_image` is called with incorrect args
`flash write_image erase .pio/build/esp32dev/firmware.bin 0x10000 verify`
`verify` should not be passed to this command. There is known error in `program_esp32` which calls `program_esp` with improperly formed args. It should be fixed by [this PR](https://github.com/espressif/openocd-esp32/pull/122).

If it is possible use `program_esp` instead of `program_esp32`

pridgeway-rsc, 2020-10-26:
> [EXTERNAL EMAIL] DO NOT CLICK links or attachments unless you recognize the sender and know the content is safe.

Yes, you can use program_esp directly and it will work correctly (flash/verify).  We have a bunch of old build targets that use the program_esp32 that were failing without this fix when I updated to the last OpenOCD release.  Should we also add a warning statement to the program_esp32 that it is being depreciated and future calls should use program_esp directly instead?

Thanks,
Paul Ridgeway

verify should not be passed to this command. There is known error in program_esp32 which calls program_esp with improperly formed args. It should be fixed by this PR<https://github.com/espressif/openocd-esp32/pull/122>. Thanks to @pridgeway-rsc<https://github.com/pridgeway-rsc>

gerekon, 2020-10-26:
> Should we also add a warning statement to the program_esp32 that it is being depreciated and future calls should use program_esp directly instead?

@pridgeway-rsc Yes, I think this is good idea. I will add it with your PR or you can add this message to the PR yourself.

pridgeway-rsc, 2020-10-26:
I went ahead and added the warning to my pull request and tested it locally.

Thanks,
Paul Ridgeway

lienbacher, 2020-10-26:
Thank you guys for your great help!
@gerekon chaning `program_esp32` to `program_esp` indeed enabled uploading. I've posted this information in the platformio community thread as well, hopefully they will update platformio soon.
It did however not fix debugging, I can only upload using the esp-prog.

gerekon, 2020-10-26:
@lienbacher Your log says that you still fail to program the chip. So breakpoint does not trigger.

lienbacher, 2020-10-31:
The guys at platformio [have updated their project](https://community.platformio.org/t/esp32-pico-v3-fails-to-upload-debug-using-esp-prog/) to the latest open ocd and debugging V3 silicon works fine now!

The problem with compiling on mac os seems to remain though, so I'm leaving the issue open for now.

closed 2022-03-02 by lienbacher