espressif/openocd-esp32 #126: Command 'flash write_image' failed with error code -1202

RavenLite, 2020-11-10:
Hi!

I found some similar problems here in [#125](https://github.com/espressif/openocd-esp32/issues/125) and [#117](https://github.com/espressif/openocd-esp32/issues/117)

And I changed my command from
```
openocd -f interface/ftdi/olimex-arm-usb-tiny-h.cfg -f board/esp-wroom-32.cfg  -c "adapter_khz 1000" -c "program_esp32 build/partitions_singleapp.bin 0x10000 reset" -c "program_esp32 build/partitions_singleapp.bin 0x8000 verify" -c "program_esp32 build/bootloader/bootloader.bin 0x1000 verify" -c "program_esp32 build/ai-deck-jpeg-streamer-demo.bin 0x10000 verify reset exit"
```

to 

```
openocd -f interface/ftdi/olimex-arm-usb-tiny-h.cfg -f board/esp-wroom-32.cfg  -c "adapter_khz 1000" -c "program_esp build/partitions_singleapp.bin 0x10000 reset" -c "program_esp build/partitions_singleapp.bin 0x8000" -c "program_esp build/bootloader/bootloader.bin 0x1000" -c "program_esp build/ai-deck-jpeg-streamer-demo.bin 0x10000 reset exit" -d
```

according to the suggestions given in the issue (program_esp32 --> program_esp, ~~verify~~)
but it didn't still work.

<details>
<summary>Log with "-d"</summary>
<pre><code>
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 2 options.c:60 configuration_output_handler(): debug_level: 3
User : 14 2 options.c:60 configuration_output_handler(): 
Debug: 15 2 options.c:184 add_default_dirs(): bindir=/builds/idf/openocd-esp32/openocd-esp32/bin
Debug: 16 2 options.c:185 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/openocd-esp32/share/openocd
Debug: 17 2 options.c:186 add_default_dirs(): exepath=/home/bitcraze/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin
Debug: 18 2 options.c:187 add_default_dirs(): bin2data=../share/openocd
Debug: 19 2 configuration.c:42 add_script_search_dir(): adding /home/bitcraze/.openocd
Debug: 20 3 configuration.c:42 add_script_search_dir(): adding /home/bitcraze/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/../share/openocd/site
Debug: 21 3 configuration.c:42 add_script_search_dir(): adding /home/bitcraze/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/../share/openocd/scripts
Debug: 22 3 configuration.c:97 find_file(): found /home/bitcraze/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/../share/openocd/scripts/interface/ftdi/olimex-arm-usb-tiny-h.cfg
Debug: 23 3 command.c:143 script_debug(): command - interface interface ftdi
Debug: 25 3 command.c:355 register_command_handler(): registering 'ftdi_device_desc'...
Debug: 26 3 command.c:355 register_command_handler(): registering 'ftdi_serial'...
Debug: 27 3 command.c:355 register_command_handler(): registering 'ftdi_channel'...
Debug: 28 3 command.c:355 register_command_handler(): registering 'ftdi_layout_init'...
Debug: 29 3 command.c:355 register_command_handler(): registering 'ftdi_layout_signal'...
Debug: 30 3 command.c:355 register_command_handler(): registering 'ftdi_set_signal'...
Debug: 31 3 command.c:355 register_command_handler(): registering 'ftdi_get_signal'...
Debug: 32 3 command.c:355 register_command_handler(): registering 'ftdi_vid_pid'...
Debug: 33 3 command.c:355 register_command_handler(): registering 'ftdi_tdo_sample_edge'...
Debug: 34 3 command.c:143 script_debug(): command - ftdi_device_desc ftdi_device_desc Olimex OpenOCD JTAG ARM-USB-TINY-H
Debug: 36 3 command.c:143 script_debug(): command - ftdi_vid_pid ftdi_vid_pid 0x15ba 0x002a
Debug: 38 3 command.c:143 script_debug(): command - ftdi_layout_init ftdi_layout_init 0x0808 0x0a1b
Debug: 40 3 command.c:143 script_debug(): command - ftdi_layout_signal ftdi_layout_signal nSRST -oe 0x0200
Debug: 42 3 command.c:143 script_debug(): command - ftdi_layout_signal ftdi_layout_signal nTRST -data 0x0100 -oe 0x0100
Debug: 44 3 command.c:143 script_debug(): command - ftdi_layout_signal ftdi_layout_signal LED -data 0x0800
Debug: 46 3 configuration.c:97 find_file(): found /home/bitcraze/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/../share/openocd/scripts/board/esp-wroom-32.cfg
Debug: 47 3 command.c:143 script_debug(): command - echo echo WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
User : 49 3 command.c:770 jim_echo(): WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
Debug: 50 4 configuration.c:97 find_file(): found /home/bitcraze/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/../share/openocd/scripts/target/esp32.cfg
Debug: 51 4 command.c:143 script_debug(): command - transport transport select jtag
Debug: 52 4 command.c:355 register_command_handler(): registering 'jtag_flush_queue_sleep'...
Debug: 53 4 command.c:355 register_command_handler(): registering 'jtag_rclk'...
Debug: 54 4 command.c:355 register_command_handler(): registering 'jtag_ntrst_delay'...
Debug: 55 4 command.c:355 register_command_handler(): registering 'jtag_ntrst_assert_width'...
Debug: 56 4 command.c:355 register_command_handler(): registering 'scan_chain'...
Debug: 57 4 command.c:355 register_command_handler(): registering 'jtag_reset'...
Debug: 58 4 command.c:355 register_command_handler(): registering 'runtest'...
Debug: 59 4 command.c:355 register_command_handler(): registering 'irscan'...
Debug: 60 4 command.c:355 register_command_handler(): registering 'verify_ircapture'...
Debug: 61 4 command.c:355 register_command_handler(): registering 'verify_jtag'...
Debug: 62 4 command.c:355 register_command_handler(): registering 'tms_sequence'...
Debug: 63 4 command.c:355 register_command_handler(): registering 'wait_srst_deassert'...
Debug: 64 4 command.c:355 register_command_handler(): registering 'jtag'...
Debug: 65 4 command.c:355 register_command_handler(): registering 'jtag'...
[... 483 lines cut ...]
Debug: 569 787 command.c:143 script_debug(): command - esp32 esp32 cget -chain-position
Debug: 570 787 command.c:143 script_debug(): command - jtag jtag tapisenabled esp32.cpu0
Debug: 571 787 command.c:143 script_debug(): command - esp32 esp32 was_examined
Debug: 572 787 command.c:143 script_debug(): command - esp32 esp32 arp_waitstate halted 5000
Debug: 573 791 xtensa_mcore.c:442 xtensa_mcore_poll(): xtensa_mcore_poll: core_poweron_mask=3
Debug: 574 804 xtensa_mcore.c:470 xtensa_mcore_poll(): xtensa_mcore_poll: Core 1 came online, setting up DCR
Debug: 575 806 xtensa.c:595 xtensa_smpbreak_set(): xtensa_smpbreak_set set smpbreak=30000, state=2
Debug: 576 814 command.c:143 script_debug(): command - esp32 esp32 invoke-event reset-init
Debug: 577 814 command.c:143 script_debug(): command - esp32 esp32 invoke-event reset-end
Debug: 578 837 command.c:143 script_debug(): command - echo echo ** Programming Started **
User : 580 860 command.c:770 jim_echo(): ** Programming Started **
Debug: 581 860 command.c:143 script_debug(): command - flash flash write_image erase build/partitions_singleapp.bin 0x10000
Error: 583 883 fileio.c:90 fileio_open_local(): couldn't open build/partitions_singleapp.bin
Debug: 584 883 command.c:630 run_command(): Command 'flash write_image' failed with error code -1202
User : 585 884 command.c:695 command_run_line(): embedded:startup.tcl:449: Error: ** Programming Failed **
in procedure 'program_esp' 
in procedure 'program_error' called at file "/home/bitcraze/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg", line 106
at file "embedded:startup.tcl", line 449
Debug: 586 884 esp_xtensa.c:213 esp_xtensa_target_deinit(): start
Debug: 587 884 xtensa.c:2157 xtensa_deinit(): start
Debug: 588 885 esp_xtensa.c:213 esp_xtensa_target_deinit(): start
Debug: 589 885 xtensa.c:2157 xtensa_deinit(): start
Debug: 590 890 target.c:2036 target_free_all_working_areas_restore(): freeing all working areas
</code></pre>
</details>

gerekon, 2020-11-10:
Hi @RavenLite . According to the log
```
Error: 583 883 fileio.c:90 fileio_open_local(): couldn't open build/partitions_singleapp.bin
```
Could you ensure that relative path `build/partitions_singleapp.bin` is valid? Also you can try with absolute path

RavenLite, 2020-11-12:
Yes, it is!
Thanks!

closed 2020-11-12 by RavenLite