espressif/openocd-esp32 #127: Repeatable crash in xtensa-esp32s2-elf-gdb (regcache::assert_regnum)

jepler, 2020-11-12:
While debugging CircuitPython, I noticed that when the target is stopped at an "rfwu" instruction (or perhaps this specific instruction?) when the debugger connects, the debugger encounters an internal error and cannot continue.

gdb session:
```
$ xtensa-esp32s2-elf-gdb build-espressif_kaluga_1/firmware.elf
GNU gdb (crosstool-NG esp-2020r3) 8.1.0.20180627-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32s2-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build-espressif_kaluga_1/firmware.elf...done.
(gdb) target remote :2331
Remote debugging using :2331
mp_hal_stdin_rx_chr () at ../../supervisor/shared/micropython.c:49
49	        mp_handle_pending();
(gdb) b ../../esp-idf/components/freertos/xtensa/xtensa_vectors.S:1855
Breakpoint 1 at 0x400280db: file ../../esp-idf/components/freertos/xtensa/xtensa_vectors.S, line 1855.
(gdb) c
Continuing.
esp32s2: Target halted, PC=0x400280DB, debug_reason=00000001
[New Thread 1073604992]
[New Thread 1073346540]
[New Thread 1073607640]
[New Thread 1073346136]
[Switching to Thread 1073573040]

Thread 1 hit Breakpoint 1, _WindowUnderflow8 () at ../../esp-idf/components/freertos/xtensa/xtensa_vectors.S:1855
1855	    rfwu
(gdb) target remote :2331
A program is being debugged already.  Kill it? (y or n) y
Remote debugging using :2331
_WindowUnderflow8 () at ../../esp-idf/components/freertos/xtensa/xtensa_vectors.S:1855
1855	    rfwu
/builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/gdb/gdb/regcache.c:395: internal-error: void regcache::assert_regnum(int) const: Assertion `regnum >= 0 && regnum < gdbarch_num_regs (arch ())' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) y
```

openocd session:
```
openocd -f board/esp32s2-kaluga-1.cfg -c gdb_port 2331
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s2: Debug controller 0 was reset.
Info : esp32s2: Core 0 was reset.
Info : Listening on port 2331 for gdb connections
Info : accepting 'gdb' connection on tcp/2331
Error: No symbols for FreeRTOS
Info : esp32s2: Target halted, PC=0x400A06B2, debug_reason=00000000
Info : Detected ESP32-S2 chip
Info : esp32s2: Target halted, PC=0x400314BA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 300 KB
Info : Flash mapping 1: 0x60020 -> 0x40080020, 943 KB
Info : esp32s2: Target halted, PC=0x400314BA, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s2.flash' size 4096 KB
Info : Using flash bank 'esp32s2.flash' size 4096 KB
Info : esp32s2: Target halted, PC=0x400314BA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 300 KB
Info : Flash mapping 1: 0x60020 -> 0x40080020, 943 KB
Info : Using flash bank 'esp32s2.irom' size 944 KB
Info : esp32s2: Target halted, PC=0x400314BA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 300 KB
Info : Flash mapping 1: 0x60020 -> 0x40080020, 943 KB
Info : Using flash bank 'esp32s2.drom' size 304 KB
Info : esp32s2: Target halted, PC=0x400280DB, debug_reason=00000001
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/2331
Info : dropped 'gdb' connection
```

igrr, 2020-11-12:
@jepler Could you describe, what is your process for stopping the program on `rfwu`? OpenOCD contains some logic to prevent that from occurring, maybe it isn't working correctly.

One other thing, could you please try setting `export ESP_XTENSA_GDB_PRIV_REGS_FIX=1` in the environment before running GDB? It may help in this situation.

erhankur, 2023-01-17:
Closed due to inactivity.

closed 2023-01-17 by erhankur