espressif/openocd-esp32 #128: JTAG flashing program_esp verify is failing

pwmb, 2020-12-09:
> Till then you can flash individual bins using `openocd` from terminal/cmd
> 
> `$ openocd -f board/esp32-wrover-kit-3.3v.cfg -c "program_esp <app_name>.bin 0x10000 verify exit" `
> 
> More info can be [obtained here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/index.html#upload-application-for-debugging)

@pwmb: I tried this on the [Adafruit HUZZAH32](https://www.adafruit.com/product/3405) without success.

I can flash this board using the serial cable but I cannot using JTAG.
I can debug this board with my JTAG connection - without problem. (The same command as below - without the `-c` option)

Here is the output: 
```
C:\somepath>openocd -f ./esp-huzzah-32.cfg  -c "program_esp build/myapp.elf 0x10000 verify reset exit"
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info: esp-huzzah-32.cfg is not an official configuration file.
Info : Configured 2 cores
Info : J-Link V10 compiled Jul 23 2019 13:46:10
Info : Hardware version: 10.10
Info : VTarget = 3.321 V
Info : clock speed 14000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset.
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Target halted, PC=0x40149F66, debug_reason=00000000
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400
** Programming Started **
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 120 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 492 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.flash' size 4096 KB
Info : Using flash bank 'esp32.flash' size 4096 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 120 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 492 KB
Info : Using flash bank 'esp32.irom' size 496 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 120 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 492 KB
Info : Using flash bank 'esp32.drom' size 124 KB
Warn : Adding extra erase range, 0x3f410000 .. 0x3f41001f
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : no flash bank found for address 0x3f41f000
Warn : no flash bank found for address 0x3ffcdb60
Warn : no flash bank found for address 0x40090000
Warn : Adding extra erase range, 0x400e0000 .. 0x400e001f
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : no flash bank found for address 0x4014c000
** Programming Finished **
** Verify Started **
Info : File content exceeds flash bank size. Only comparing the first 4128768 bytes of the file
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
** Verify Failed **
shutdown command invoked

C:\somepath>
``` 

How to work around this problem?
Let me know if this should become a separate issue.

_Originally posted by @TobiasBoeschMiele in https://github.com/espressif/vscode-esp-idf-extension/issues/161#issuecomment-741739097_

pwmb, 2020-12-09:
@gerekon seems like this is open ocd issue and the user is unable to flash bin (failing at verify stage).

This issue is created via https://github.com/espressif/vscode-esp-idf-extension/issues/161#issuecomment-741739097

TobiasBoeschMiele, 2020-12-10:
Can someone please help me with this? It would be very helpful to get this solved soon.

gerekon, 2020-12-10:
Hi
@TobiasBoeschMiele Could you collect and post logs as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb)?

TobiasBoeschMiele, 2020-12-11:
Instructions taken from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb

1. In issue report provide details of your configuration:

   a. JTAG adapter type, and the chip/module being debugged.
        * First it was a SEGGER J-Link (with libusb driver / Zadig)
        * Now the ESP-Prog Board is used (with libusb driver / Zadig)
        * Yes, both show the same behavior

   b. Release of ESP-IDF used to compile and load application that is being debugged.
        * ESP-IDF v4.3-dev-1901-g178b122c1-dirty (Commit [178b122c145c19e94ac896197a3a4a9d379cd618](https://github.com/espressif/esp-idf/tree/178b122c145c19e94ac896197a3a4a9d379cd618))

   c. Details of OS used for debugging.
        * Windows 10 (1909)

   d. Is OS running natively on a PC or on a virtual machine?
        * Native

2. Create a simple example that is representative to observed issue. Describe steps how to reproduce it. In such an example debugging should not be affected by non-deterministic behaviour introduced by the Wi-Fi stack, so problems will likely be easier to reproduce, if encountered once.
    * Take the `Hello-world` example and debug it with a standard debug configuration in eclipse (with espressif plugin):
[image]
    * Append ` -c "program_esp build/hello-world.elf 0x10000 verify exit"` to the `Config options` and start it again

3. Prepare logs from debugging session by adding additional parameters to start up commands.
    OpenOCD:
        * `openocd -l openocd_log.txt -d3 -f board/esp32-wrover-kit-3.3v.cfg`
        * [openocd_log.txt](https://github.com/espressif/openocd-esp32/files/5678698/openocd_log.txt)
    Debugger:
    xtensa-esp32-elf-gdb -ex "set remotelogfile gdb_log.txt" <all other options>
        * Don't know how to do that - which other options? I use the eclipse plugin. There might be other options, but I don't have a handy command.

gerekon, 2020-12-11:
@TobiasBoeschMiele Please, collect openocd.log for your problem mentioned in start post.
I think you can do this by invoking  `openocd -l openocd_log.txt -d3 -f ./esp-huzzah-32.cfg  -c "program_esp build/myapp.elf 0x10000 verify reset exit"`

gerekon, 2020-12-11:
Also it would be good to have your `esp-huzzah-32.cfg` file.

gerekon, 2020-12-11:
BTW I just notices that you are trying to flash ELF file. That's wrong you need to flash `build/myapp.bin`.
So, please, try the following command `penocd -l openocd_log.txt -d3 -f ./esp-huzzah-32.cfg -c "program_esp build/myapp.bin 0x10000 verify reset exit`

TobiasBoeschMiele, 2020-12-14:
> BTW I just noticed that you are trying to flash ELF file. That's wrong, you need to flash `build/myapp.bin`.
> So, please, try the following command `openocd -l openocd_log.txt -d3 -f ./esp-huzzah-32.cfg -c "program_esp build/myapp.bin 0x10000 verify reset exit`

Using the `bin` file instead of the `elf`file makes flashing and verifying work. Thanks for looking at the details!
The [eclipse plugin](https://github.com/espressif/idf-eclipse-plugin) automatically selects the elf file for flashing. That should be changed.

closed 2020-12-14 by gerekon