espressif/openocd-esp32 #131: JTAG scan chain interrogation failed: all ones

MAbjadpour, 2020-12-18:
I've been trying to program the _ESP32-WROOM-32E_ and _ESP-DevKitC-v4_ (based on 32E) modules over JTAG for the past two days and every single try has failed following the same error.

I've used both **SEGGER JLink** and **ESP-Prog** during my tests using the standard JTAG pins, i.e. TMS, TDO, TDI, TCK, GND and Vref. After running this command:
```
openocd -f board/esp32-wrover-kit-3.3v.cfg
```
I get:
```
Open On-Chip Debugger  v0.10.0-esp32-20201202 (2020-12-02-17:38)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : FreeRTOS creation
Info : FreeRTOS creation
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link V10 compiled Apr 23 2020 16:49:14
Info : Hardware version: 10.10
Info : VTarget = 3.339 V
Info : clock speed 1000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
```
As stated in the ESP-IDF JTAG Debugging page, such an error is most probably referring to a problem in pin connections.
However, I have checked them multiple times and I have no clue where this error originates from. I have gone through similar issues, but none has really been helpful for my case.

An interesting thing is, when I hold the **EN** **button** on the DevKit and run the command above, the error changes from `Error: JTAG scan chain interrogation failed: all ones` to `Error: JTAG scan chain interrogation failed: all zeros`. This also happens when I ground the TDO.

I get the exact same behavior no matter which debugger, JLink or ESP-Prog, I use. I'd be grateful if anyone has any advice.

Thanks.

gerekon, 2020-12-21:
Hi @MAbjadpour 
Could you post pinout of you JTAG wrires? How did you connect ESP-Prog to the chip?

MAbjadpour, 2020-12-22:
> Hi @MAbjadpour
> Could you post pinout of you JTAG wires? How did you connect ESP-Prog to the chip? I mean what pin numbers on ESP-DevKitC-v4 did you connect to.
> What length of wires do you use?

Hi @gerekon
Thanks for your reply. I actually figured out that the issue I've been facing originates from the original ESP module's firmware. Apparently the modules come pre-programmed with the default AT command firmware which interferes with JTAG pins and makes it impossible to use JTAG.

Is there anyway I can erase the stock firmware via JTAG, and not via the serial bootloader?

Wires are about 15 cm. Here's the pinout, using the [2x5 JTAG connector](https://github.com/espressif/esp-iot-solution/blob/master/documents/_static/esp-prog/JTAG_pin.png):
1 --> Module's 3v3
2 --> IO 14
3 --> Module's GND
4 --> IO 13
6 --> IO 15
8 --> IO 12

igrr, 2020-12-23:
@MAbjadpour You can re-program flash on these pre-programmed modules over JTAG if you start the module in download mode. To do this, make sure GPIO0 is grounded at power-on.

MAbjadpour, 2020-12-23:
> @MAbjadpour You can re-program flash on these pre-programmed modules over JTAG if you start the module in download mode. To do this, make sure GPIO0 is grounded at power-on.

Thanks a lot @igrr. That solved the problem!

closed 2020-12-23 by MAbjadpour

shanfei-huang, 2021-02-03:
你好我遇到你和你一样的问题，不知道怎么解决，可以说的具体或者是详细一点吗，可以邮箱联系1493207731@qq.com

minaarmia, 2022-03-17:
Hey 
I'm connecting FTDI chip between Raspberry Pi and stm32 discovery board and when I'm trying to run the cfg file I receive this 
Open On-Chip Debugger 0.11.0+dev-00418-g918811529 (2021-10-12-13:34)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
  ftdi layout_init data direction

(Urgently) Can anyone help