espressif/openocd-esp32 #132: JTAG Debugging Error (IDFGH-4502)

kaibin1111, 2020-12-28:
C:\Users\make\esp\esp-idf>openocd -f interface/jlink.cfg -f board/esp-wroom-32.cfg
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 15000 kHz

WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Info : J-Link V9 compiled Apr 20 2018 16:47:26
Info : Hardware version: 9.40
Info : VTarget = 3.301 V
Info : clock speed 15000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
shutdown command invoked

igrr, 2020-12-28:
@kaibin1111 The "IR capture error; saw 0x1f not 0x01" message indicates that the debug probe is not able to communicate with the target. Please check the wiring and ensure that the code running on the target doesn't reconfigure JTAG pins for other purposes (such as SPI).

kaibin1111, 2020-12-28:
The wiring should be all right，Now the output is as follows

C:\Users\make\esp\esp-idf>openocd -f interface/jlink.cfg -f board/esp-wroom-32.cfg
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 15000 kHz

WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Info : J-Link V9 compiled Apr 20 2018 16:47:26
Info : Hardware version: 9.40
Info : VTarget = 3.306 V
Info : clock speed 15000 kHz
Info : TAP esp32.cpu0 does not have valid IDCODE (idcode=0xffffe000)
Info : TAP esp32.cpu1 does not have valid IDCODE (idcode=0xfffff000)
Info : TAP auto0.tap does not have valid IDCODE (idcode=0xfffff800)
Info : TAP auto1.tap does not have valid IDCODE (idcode=0xfffffc00)
Info : TAP auto2.tap does not have valid IDCODE (idcode=0xfffffe00)
Info : TAP auto3.tap does not have valid IDCODE (idcode=0xffffff00)
Info : TAP auto4.tap does not have valid IDCODE (idcode=0xffffff80)
Info : TAP auto5.tap does not have valid IDCODE (idcode=0xffffffc0)
Info : TAP auto6.tap does not have valid IDCODE (idcode=0xffffffe0)
Info : TAP auto7.tap does not have valid IDCODE (idcode=0xfffffff0)
Info : TAP auto8.tap does not have valid IDCODE (idcode=0xfffffff8)
Info : TAP auto9.tap does not have valid IDCODE (idcode=0xfffffffc)
Info : TAP auto10.tap does not have valid IDCODE (idcode=0xfffffffe)
Error: esp32.cpu0: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
Info : cpu0: Detected debug stubs @ 69d60000
shutdown command invoked
Info : cpu0: Restore debug stubs address 69d60000
Info : cpu1: Restore debug stubs address 69d60000

gerekon, 2020-12-28:
@kaibin1111 
I recommend to lower the adapter clock to 1000-3000 KHz
```
Info : clock speed 15000 kHz
```

kaibin1111, 2020-12-29:
The board example is bink. Now the output is as follows.

C:\Users\make\esp\esp-idf>openocd -f interface/jlink.cfg -f board/esp-wroom-32.cfg
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz

WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Info : J-Link V9 compiled Apr 20 2018 16:47:26
Info : Hardware version: 9.40
Info : VTarget = 3.327 V
Info : clock speed 1000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
shutdown command invoked

gerekon, 2020-12-29:
> The board example is bink. Now the output is as follows.

Is it flashed (already in flash)? Just to be sure that JTAG pins functions are not overriden by app. Or you can put board in to UART download mode by pressing 'Boot' button (or connecting GPIO0 to GND) during reset.

Could you post how you connect JTAG pins to JLINK?

kaibin1111, 2020-12-30:
Board :ESP-WROOM-32. 
Leds can flash( LED-GPIO2)

Connect
ESP32     J-LINK
TDO        TDO
TDI          TDI
TCK         TCLK
TMS        TMS
GND        GND

gerekon, 2020-12-30:
> Connect
ESP32 J-LINK
TDO TDO
TDI TDI
TCK TCLK
TMS TMS
GND GND

Looks correct ;-), but I asked about ESP32 pin numbers you connected to. 
BTW, What type of ESP32 board do you use?

kaibin1111, 2020-12-30:
Board ： NodeMCU-32S

gerekon, 2020-12-30:
It seems that you should connect JLINK to:
JTAG >  NodeMCU-32S (ESP32)
TDI -> Pin18 (GPIO12)
TCK -> Pin20 (GPIO13)
TMS -> Pin17 (GPIO14)
TDO -> Pin21 (GPIO15)

Could you check this?

kaibin1111, 2020-12-31:
I checked it. There are no mistakes

kaibin1111, 2020-12-31:
Thank you very much for your answer. I've solved it

closed 2020-12-31 by kaibin1111