espressif/openocd-esp32 #133: Enable remote_bitbang by default on next release?

brainstorm, 2021-01-02:
Hello Espressif,

I'm connecting to the ESP32S2 using the newly released "Glasgow" explorer board:

https://www.crowdsupply.com/1bitsquared/glasgow

Which supports precisely that probe interfacing, via commands:

```shell
$ glasgow run jtag-openocd --pin-tck 0 --pin-tms 3 --pin-tdi 2 --pin-tdo 1 -V 3.3 tcp:localhost:2222
I: g.device.hardware: building bitstream ID b97acf8ca32a08beb46d6713ecdb8b98
Warning: Wire top.$flatten\i2c_target.\bus.$verilog_initial_trigger has an unprocessed 'init' attribute.
Warning: Wire top.$flatten\multiplexer.\U$$0.\U$$2.$verilog_initial_trigger has an unprocessed 'init' attribute.
Warning: Wire top.$flatten\multiplexer.\U$$0.\U$$2.\bus.$verilog_initial_trigger has an unprocessed 'init' attribute.
Warning: Wire top.$verilog_initial_trigger has an unprocessed 'init' attribute.
I: g.cli: running handler for applet 'jtag-openocd'
I: g.applet.interface.jtag_openocd: port(s) A, B voltage set to 3.3 V
I: g.applet.interface.jtag_openocd: socket: listening at tcp:localhost:2222
```

Unfortunately, to my surprise, I found that `remote_bitbang` is disabled by default on my build?

```
$ openocd --version
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2

$ openocd -c 'interface remote_bitbang; remote_bitbang_port 2222'
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (remote_bitbang)
The following debug interfaces are available:
1: ftdi
2: usb_blaster
3: ft232r
4: jlink
5: vsllink
6: ulink
7: hla
8: osbdm
9: opendous
10: aice
11: xds110
```

Could you please enable it back on future builds? Thanks in advance!

brainstorm, 2021-01-02:
OTOH, enabling `BUILD_REMOTE_BITBANG  1` from a `master` clone still fails:

```
$ ./openocd -c 'interface remote_bitbang; remote_bitbang_port 2222'
Open On-Chip Debugger  v0.10.0-esp32-20201202 (2021-01-02-21:22)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (remote_bitbang)
The following debug interfaces are available:
1: ftdi
2: usb_blaster
3: esp_usb_jtag
4: esp_usb_jtag
5: ft232r
6: presto
7: usbprog
8: openjtag
9: jlink
10: vsllink
11: rlink
12: ulink
13: arm-jtag-ew
14: hla
15: osbdm
16: opendous
17: aice
18: cmsis-dap
19: kitprog
20: xds110
21: jtag_esp_remote
```

But there's the last `jtag_esp_remote` which seems to [have been implemented just a few months ago](https://github.com/espressif/openocd-esp32/commit/1b3455a5ed3c79e36806a8fb42be2a22fa111630)... are there any official docs on how to use that new interface? What was the reasoning behind not having the generic `remote_bitbang`?

OffTopic: Two `esp_usb_jtag`? What is the difference between them?

brainstorm, 2021-01-02:
Navigating the `jtag_esp_remote` commit referred above and its sample `.cfg` file...

```
$ ./src/openocd -c 'interface jtag_esp_remote; set protocol tcp; jtag_esp_remote_set_port tcp:2222; jtag_esp_remote_set_address "127.0.0.1"'
Open On-Chip Debugger  v0.10.0-esp32-20201202 (2021-01-02-21:22)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Error: TCP protocol must be set up for jtag_esp_remote_set_port
```

brainstorm, 2021-01-02:
Finally:

```
$ ./src/openocd -c 'interface jtag_esp_remote; jtag_esp_remote_protocol tcp; jtag_esp_remote_set_port 2222; jtag_esp_remote_set_address "127.0.0.1"'
Open On-Chip Debugger  v0.10.0-esp32-20201202 (2021-01-02-21:22)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : TCP protocol set for esp remote
Info : Set server port to 2222
Info : Set server address to 127.0.0.1
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Connection to 127.0.0.1 : 2222 succeed
Info : This adapter doesn't support configurable speed
Warn : There are no enabled taps.  AUTO PROBING MIGHT NOT WORK!!
```

igrr, 2021-01-04:
@brainstorm `jtag_esp_remote` is a different protocol from `remote_bitbang`, so as you have probably already realized, it won't work with tools that implement `remote_bitbang`. So far, we don't have devices which natively support `jtag_esp_remote` protocol released publicly, it is added as a provision for future devices.
Can definitely enable `remote_bitbang` protocol in the next release.

brainstorm, 2021-01-04:
Yes, figured, thanks Igor, I'll fix up the PR tonight ;)

closed 2023-01-17 by erhankur

brainstorm, 2023-01-18:
@erhankur ... so, was it solved elsewhere or will the PR be merged?

erhankur, 2023-01-18:
@brainstorm Sorry, I missed adding a closing comment. Yes, we built with `--enable-remote-bitbang` option in the latest release.