espressif/openocd-esp32 #136: esp prog (FT2232H) debug using openocd is extremely slow

summer20100514, 2021-01-12:
1. JTAG adapter type, and the chip/module being debugged.
**[JTAG]**
it is espressif official esp-prog jtag, which is based on FT2232H. 
[ESP-Prog_guide_en](https://github.com/espressif/esp-iot-solution/blob/master/documents/evaluation_boards/ESP-Prog_guide_en.md)
**[CHIP]**
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7104
ho 0 tail 12 room 4
load:0x40078000,len:14052
load:0x40080400,len:4380
entry 0x400806f4
I (75) boot: Chip Revision: 1
I (75) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF fd34784-dirty 2nd stage bootloader
I (41) boot: compile time 10:33:04
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Flash RID  : 0x684018
I (51) boot: SPI Flash  MF  : 0x68
I (55) boot: SPI Flash  ID  : 0x4018
I (59) boot: SPI Speed      : 40MHz
I (64) boot: SPI Mode       : DIO
I (68) boot: SPI Flash Size : 2MB
I (72) boot: Partition Table:
I (75) boot: ## Label            Usage          Type ST Offset   Length                                            
I (83) boot:  0 nvs              WiFi data        01 02 00009000 00006000                                          
I (90) boot:  1 phy_init         RF data          01 01 0000f000 00001000                                          
I (97) boot:  2 factory          factory app      00 00 00010000 00100000                                          
I (105) boot: End of partition table
I (109) boot_comm: chip revision: 1, min. application chip revision: 0
I (116) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x04e84 ( 20100) map
I (133) esp_image: segment 1: paddr=0x00014eac vaddr=0x3ffb0000 size=0x017b4 (  6068) load
I (137) esp_image: segment 2: paddr=0x00016668 vaddr=0x40080000 size=0x00400 (  1024) load
I (144) esp_image: segment 3: paddr=0x00016a70 vaddr=0x40080400 size=0x08c00 ( 35840) load
I (167) esp_image: segment 4: paddr=0x0001f678 vaddr=0x00000000 size=0x00998 (  2456) 
I (168) esp_image: segment 5: paddr=0x00020018 vaddr=0x400d0018 size=0x12988 ( 76168) map
I (206) boot: Loaded app from partition at offset 0x10000
I (207) boot: Disabling RNG early entropy source...
I (207) cpu_start: Pro cpu up.
I (211) cpu_start: Application information:
I (215) cpu_start: Project name:     hello-world
I (221) cpu_start: App version:      fd34784-dirty
I (226) cpu_start: Compile time:     Jan 12 2021 10:33:00
I (232) cpu_start: ELF file SHA256:  05ca891aadcbd6a7...
I (238) cpu_start: ESP-IDF:          fd34784-dirty
I (244) cpu_start: Single core mode
I (248) heap_init: Initializing. RAM available for dynamic allocation:
I (255) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (261) heap_init: At 3FFB26E0 len 0002D920 (182 KiB): DRAM
I (267) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (274) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (280) heap_init: At 40089000 len 00017000 (92 KiB): IRAM
I (286) cpu_start: Pro cpu start user code
I (304) spi_flash: detected chip: generic
I (304) spi_flash: flash io: dio
W (304) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (315) cpu_start: Starting scheduler on PRO CPU.
Hello world!
This is ESP32 chip with 1 CPU cores, WiFi/BT/BLE, silicon revision 1, 2MB external flash

2. Release of ESP-IDF used to compile and load application that is being debugged.
ESP-IDF: fd34784-dirty
3. Details of OS used for debugging.
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
4. Is OS running natively on a PC or on a virtual machine?
VirtualBox
version 6.1.16 r140961 (Qt5.6.2)

I am debugging the hello world example resides in the esp idf. I have setup the environment successfully according to the official guide, but each time I type 'n' in gdb, it will take the debugger several seconds to finish. I have lowered the clock speed and execute command "ftdi_tdo_sample_edge falling", but nothing changed. I have attached the openocd log below. Thank you.
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/5799549/openocd_log.txt)

gerekon, 2021-01-14:
Hi @summer20100514 
There is some optimization work planned for FreeRTOS related code. Currently every time target is stopped it goes through the whole tasks list and read them out.
If you do not need RTOS support you can try to run OpenOCD with the following command line:
`openocd -c 'set ESP_RTOS "none"' -f board/esp32-wrover-kit-3.3v.cfg`
or 
`openocd -c 'set ESP_RTOS "hwthread"' -f board/esp32-wrover-kit-3.3v.cfg`

gerekon, 2021-01-14:
Also, please disable OpenOCD verbose log out put `-d2` and default adapter clock 20 MHz

summer20100514, 2021-01-15:
@gerekon the debug speed improved a lot after I disabled FreeRTOS support, thank you. but still not as quick as possible with noticeable delay. what should I do next?

summer20100514, 2021-01-15:
@gerekon seems better on win10 instead of virtualbox ubuntu18.04 even if with ESP_RTOS support enabled. so is there any compatibility issue with virtualbox openocd?

gerekon, 2021-01-15:
>  so is there any compatibility issue with virtualbox openocd?

There should be no compatibility issue. But when you run SW in virtual machine it is normal that it works slower. In case of OpenOCD USB data have to go through two USB stacks: Windows + Ubuntu and this introduces additional delays. 
To get the higher speed it is recommended to run OpenOCD on host system and connect JTAG adapter directly to host USB port (w/o additional USB-hubs).

gerekon, 2021-01-15:
@summer20100514 
Can we close this issue?

summer20100514, 2021-01-18:
@gerekon ok, thank you.

closed 2021-01-18 by summer20100514