espressif/openocd-esp32 #137: debugging with JTAG no longer works

ChrisIdema, 2021-01-19:
1. my board: esp32doit-devkit-v1
2. debugger: jtag-lock-pick_tiny_2 (based on FTDI FT232H)
3. OS: Ubuntu 20
4. ide: vscode with platformio
5. Project: https://github.com/ChrisIdema/esp32_debug_test

Debugging with JTAG no longer works. I've tried multiple devkits. Replaced wiring. Tried different versions of openocd-esp32 and of platform espressif32. Erased flash. Reinstalled everything. Added offset_address. It just doesn't seem to work anymore. It worked fine about half a year ago. I've just ordered an ESP-prog debugger in the hopes that one works, but I suspect that won't make a difference.

I've reduced the adapter speed to 1000 kHz as I suspect the default 20000 kHz could cause problems, although it was pretty stable when it still worked.

```
Processing esp32 (platform: espressif32; framework: espidf; board: esp32doit-devkit-v1)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.1.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (custom) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40100.200827 (4.1.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-ninja 1.7.1
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Retrieving maximum program size .pio/build/esp32/firmware.elf
Checking size .pio/build/esp32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.2% (used 10616 bytes from 327680 bytes)
Flash: [=         ]  13.9% (used 145242 bytes from 1048576 bytes)
========================= [SUCCESS] Took 6.37 seconds =========================
Reading symbols from /home/e2/e2/lib6/esp32_debug_test/.pio/build/esp32/firmware.elf...
done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = custom
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.10.0-esp32-20201202 (2020-12-02-17:38)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : FreeRTOS creation
adapter speed: 1000 kHz

Info : tcl server disabled
Info : telnet server disabled
Info : clock speed 1000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : accepting 'gdb' connection from pipe
Warn : No symbols for FreeRTOS!
Info : esp32.cpu0: Target halted, PC=0x40007BA1, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Error: Target not examined yet
Error: Failed to wait algorithm (-4)!
Error: Target not examined yet
Error: failed to restore 28 bytes of working area at address 0x3ffc0590
Error: Target not examined yet
Error: failed to restore 28 bytes of working area at address 0x40091840
Error: Target not examined yet
Error: failed to restore 1300 bytes of working area at address 0x3ffc007c
Error: Target not examined yet
Error: failed to restore 6208 bytes of working area at address 0x40090000
Error: Target not examined yet
Error: failed to restore 124 bytes of working area at address 0x3ffc0000
Error: Algorithm run failed (-4)!
Warn : Empty flash mapping!
Warn : Failed to get flash mappings (0)!
Error: Failed to find HALTED core!
Error: Failed to run flasher stub (-4)!
Error: Failed to find HALTED core!
Error: Failed to run flasher stub (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response

```

gerekon, 2021-01-19:
Hi @ChrisIdema 
Could you collect verbose logs as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb)?

ChrisIdema, 2021-01-19:
@gerekon
I've updated my post with the log.

ChrisIdema, 2021-01-19:
Switch back to the old wiring and now it works. Must have been a bad contact in combination with incorrect settings. Now everything works.

kkuez, 2021-03-30:
Hi @ChrisIdema,
may you tell about the wiring?
Kind regards