espressif/openocd-esp32 #139: Error: JTAG scan chain interrogation failed: all ones

sylents-dev-e, 2021-01-27:
Hi @all,

i know this problem has been discussed before but none of the fixes works for my setup. I cannot start the debugger properly.
+ ESP32 Wrover Kit v4.1
+ OpenOCD v0.10.0-esp32-20200709 
+ ESP-IDF v4.2

I checked if i have occupied a pin that is reserved for JTAG. Pin 14 was actually selected for usage but i removed it however, it is still not working.
Any idea how to fix it?

**Full message:**
$ openocd -f board/esp32-wrover-kit-3.3v.cfg
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections

Greets

gerekon, 2021-02-02:
@sylents-dev-e  Can you try with `hello_world` IDF example? Also it would be good to have verbose OpenOCD log `openocd -f board/esp32-wrover-kit-3.3v.cfg -d3 -l oocd.log`

sylents-dev-e, 2021-02-02:
No, i tried with my own projekt, but i will try "hello_world" it now. Thanks in advance!
Trying the OpenOCD logging gives me the follwing message:

$ openocd -f board/esp32-wrover-kit-3.3v.cfg -d3 -l oocd.log
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 13 4 options.c:60 configuration_output_handler(): debug_level: 3
User : 14 8 options.c:60 configuration_output_handler():
Debug: 15 11 command.c:143 script_debug(): command - log_output log_output oocd.log
Assertion failed!

Program: PATH/openocd.exe
File: src/jtag/core.c, Line 343

Expression: jtag_trst == 0

Greets

gerekon, 2022-10-25:
Closing as no updates for a long time

closed 2022-10-25 by gerekon