espressif/openocd-esp32 #141: Flash ESP32s2 With ESP prog and OpenOCD error

alexiskyriakou, 2021-02-12:


alexiskyriakou, 2021-02-12:
after cleaning the project an setting the target to esp32s2 and downloaded the latest OpenOCD version the i build and tried to flash but i got the following error now:

>Open On-Chip Debugger  v0.10.0-esp32-20201202 (2020-12-02-17:38)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
>
>Info : FreeRTOS creation
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
C:/Users/USER/.espressif/tools/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg:13❌ 0: Error: c:UsersUSERespesp-rainmakerexamplesswitchuild/flasher_args.json: Invalid argument
in procedure 'program_esp_bins' 
at file "C:/Users/USER/.espressif/tools/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg", line 130

gerekon, 2021-02-12:
Hi @alexanderkurz 
It seems to be VsCode extension problem. What extension version do you use?

@pwmb  @brianignacio5  Can you have a look at it?
1) Before using `program_esp_bins` extension should check OpenOCD version and report error in case of missmatch
2) It seems that there is problem with passing path to OpenOCD on Windows.

alexiskyriakou, 2021-02-12:
> Hi @alexanderkurz
> It seems to be VsCode extension problem. What extension version do you use?
>
> @pwmb @brianignacio5 Can you have a look at it?
>
>     1. Before using `program_esp_bins` extension [...]

the ESP extension is as follow:

[image]

after installing the extension for ESP i try to flash but i got an error for OpenOCD version say that is old and i must use newer version and i download the latest, extract it to users\user\.espressif\tools\ and replace the old openOCD with the latest. after that i getting that error.

pwmb, 2021-02-18:
> Hi @alexanderkurz
> It seems to be VsCode extension problem. What extension version do you use?
>
> @pwmb @brianignacio5 Can you have a look at it?
>
> 1. Before using `program_esp_bins` extension should [...]

The OpenOCD Version Precheck code is working fine in windows, do you have a specific version of open-ocd and windows combination where its not working ?

alexiskyriakou, 2021-02-18:
the OpenOCD version is v0.10.0-esp32-20200709 in the .espressif directory and windows 10

pwmb, 2021-02-18:
You need `v0.10.0-esp32-20201125` above to use this feature

alexiskyriakou, 2021-02-18:
Ok... How can i install an newer version of openocd?

pwmb, 2021-02-19:
Its as simple as downloading the latest version of open-ocd from [release tab](https://github.com/espressif/openocd-esp32/releases) and then replace the downloaded folder with the installed open-ocd in your system, which might be inside `$user/.espressif`

brianignacio5, 2021-02-19:
Here is a comment about how to manually replace openOCD on extension configuration settings manually

https://github.com/espressif/vscode-esp-idf-extension/issues/289#issuecomment-758372220

alexiskyriakou, 2021-02-19:
thanks guys i will try your suggestion and i will let you know

alexiskyriakou, 2021-02-19:
Hi @brianignacio5, @pwmb 

i did your suggestion and now i am getting the following error

>Open On-Chip Debugger  v0.10.0-esp32-20201202 (2020-12-02-17:38)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

>Info : FreeRTOS creation
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
C:\Users\USER\.espressif\tools\openocd-esp32\v0.10.0-esp32-20201202/openocd-esp32/share/openocd/scripts/target/esp_c❌ ommon.cfg:130: Error: c:UsersUSERespesp-rainmakerexamplesgpiouild/flasher_args.json: Invalid argument
in procedure 'program_esp_bins' 
at file "C:\Users\USER\.espressif\tools\openocd-esp32\v0.10.0-esp32-20201202/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 130

meltdown03, 2021-02-20:
@alexiskyriakou, see https://github.com/espressif/vscode-esp-idf-extension/pull/301 and https://github.com/espressif/vscode-esp-idf-extension/issues/297. Do you have the most recent IDF extension installed? 1.0.0 was just released. This exact issue was already reported and should be fixed.

alexiskyriakou, 2021-02-21:
Yes, extention version is 1.0.0

alexiskyriakou, 2021-02-22:
thank you @meltdown03 for the solution... its working now

gerekon, 2021-02-24:
@alexiskyriakou Can we close the issue?

closed 2021-02-24 by alexiskyriakou