espressif/openocd-esp32 #143: Inline debugging ESP32 beyond deep sleep ?

nullptrdesaster, 2021-02-18:
I try to debug beyond deep sleep. I have one breakpoint set at  esp_deep_sleep(1000000 * 10). It is hit and I can step until esp_deep_sleep_start() is called. The MCU is regularly active, and I can monitor its output which is sent over telnetstream. Albeit obviously running, the debugger never stops again, seemingly ignorig  breakpoints set wherever in my code (and eaqually not in void RTC_IRAM_ATTR esp_wake_deep_sleep(void) at  esp_default_wake_deep_sleep(). 
I whouldn't expect debugging to survive deep sleep, but I see on the debug console these lines:

Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
esp32.cpu0: Debug controller was reset.
esp32.cpu0: Core was reset.
Info : esp32.cpu0: Detected debug stubs @ 3ffc24b0
esp32.cpu0: Detected debug stubs @ 3ffc24b0

looping in 10 sec inteval, suggesting that some debugging stubs may have survived and may be accessible for further inline debugging.
Can someone please offer hints to achieve this? Thanks for your time.

env: 
debian 10.8, VSC 1.53.2, PIO Core 5.1.0, Home 3.3.3, ESP-Prog, 
platformio.ini with 
platform = https://github.com/platformio/platform-espressif32
board = featheresp32
framework = arduino
upload_protocol = esp-prog
upload_port = /dev/ttyUSB_prog
monitor_port = /dev/ttyUSB_jtag

alecks68, 2021-05-05:
I am stuck with the same issue. 

My first idea was to break at ESP.deepSleep and just restart the debugger for the next "round". There are 2 problems with this.
a) the content of RTC memory is lost
b) while starting the debugger, the system under test (ESP) is already running and might go into sleep and wake-up, disturbing or resetting the JTAG connection. Maybe someone knows how to halt the core until debugging actually starts?

Now I am using an emulation like below which is close to real behaviour, although variables are not reset and of course everything the is in the WLAN part is not re-initialized. Maybe someone has an idea how to reset/re-initilazie all (own) static and global variables without creating a function manually?

I am also using rtc_wake_stub functions in RTC IRAM for periodic measurements without full reboot into setup()/loop(). I tried to do similar emulation without success - I did not manage to call the wake_stub() from within loop().
Of course the best solution would be to either have the debugger survive deep sleep as this is a fundamental feature of the chip, or to have a kind of soft_restart function to go through reboot without resetting the debugger.

// in header file; works with ESP.deepSleep() calls directly in setup() or in loop().
#ifdef DEBUG_FTDI
bool emul_sleep = false;
#define ESP MyESPEmul
static class
{
public:
  void deepSleep(long long unsigned s) {emul_sleep = true; Serial.printf("emulating %ds deep sleep in 1s; put a breakpoint at setup() to stop longer\n", (int)(s/1000000)); delay(1000);}
} ESP;
#define deepSleep(t) deepSleep(t);return
#endif

// first thing in loop()
void loop()
{
#ifdef DEBUG_FTDI
  while(emul_sleep) {
    emul_sleep = false;
    setup();
  }
#endif
...

nullptrdesaster, 2021-05-07:
To let own variables survive deep sleep could be solved.
You mentioned, that RTC memory does not survive deep sleep, but I think that they do:  
"RTC memory regions hold their contents during deep sleep." https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/deep-sleep-stub.html
In fact, this works and variables can be accessed after deep sleep for ULP programming, as shown in  example progs (e.g ulp_adc_example_main.c )
So, as long as needed for debugging, variables could be placed in the RTC memory.

alecks68, 2021-05-08:
Maybe my description was misleading. 
My setup is the following: I use ESP.deepSleep for deep sleep, RTC memory to retain data between sleep periods and wake_stub for fast checking of measurements ot rexternal events (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/deep-sleep-stub.html#). The wake_stub is located in RTC IRAM.
In a non debug environment all memory is reset at power on. During setup() I configure my wake_stub and start using local, global and static variables. Data that shall survive sleep is stored in RTC memory. 
After deep sleep the wake stub is called where I decide to sleep again or continue starting the system into setup(). All memory apart from RTC memory is reset at every wakeup. 
Now I want to debug. Just restarting the debug session each time ESP.deepSleep is called does not work. The RTC memory is reset if you restart the debugger and every wake-up is exactly the same, i.e. like first power on. I also see that while I start the debugger, the existing code is already up and running and might go through sleep/wake-up giving me JTAG connection errors.
Next idea is to avoid calling ESP.deepSleep and just do a short delay and then call setup(). This works quite ok, but of course in this scenario global and static variables are not reset and also the WLAN and RTOS part is just kept where it was. For this you could avoid static and re-initialize global variables manually in setup(). In any case it is only an approximation of the real behaviour.
Now I also want to test the wake stub. So instead of ESP.deepSleep I would like to run a short delay, call wake_stub (also disable deep sleep here) and then call setup. However, I did not manage to call the wake_stub function from loop().
Any ideas to get such emulation closer to reality are appreciated. Of course, if there were a possibility to restart the ESP without loosing the debugger, this would be best.

nullptrdesaster, 2021-05-12:
Yes, I see we agree.  For now, I don't see how to get JTAG debugging survive deep (or not so deep) sleep. Maybe, one of the espressif developers can shed a light on this problem ?

igrr, 2021-05-12:
Unfortunately I don't see a way to have the breakpoints re-created early enough after chip reset to stop in the deep sleep wake stub. My recommendation is to use a planted breakpoint in your deep sleep stub code, for example:
```c
if (need_break()) {
    while (!cpu_ll_is_debugger_attached()) {
        // wait for the debugger to connect
        // might need to feed the watchdogs here, not sure
    }
    __asm__ __volatile__("break 1, 1");
}
```
where need_break would be checking e.g. a value in some variable, `RTC_DATA_ATTR g_need_break;`.

Before entering deep sleep, stop the program and change the `g_need_break` value in the debugger, then resume. After the deep sleep reset g_need_break will be kept, and the program will wait until the debugger re-connects to the CPU, and then will hit the "break" instruction, stopping. Then you can step over the breakpoint (or do `set $pc=$pc+3` in GDB) and continue debugging the stub.

nullptrdesaster, 2021-05-15:
Hi Ivan, 
thanks for your input. Embedding GDB breakpoints is a great idea. I've no experience with that so far. 
Unfortunately, I had no success to break execution after wake up by putting your code in void RTC_IRAM_ATTR esp_wake_deep_sleep(void), just after   esp_default_wake_deep_sleep();
   
To clarify that for me,  I've put 
this code:

 if ( cpu_ll_is_debugger_attached())
        {
        __asm__ __volatile__("break 1, 1");
        }
  
 in the initializing part of my code, i.e. before deep sleep. With "debug_init_break = tbreak app_main" in my platformio.ini file, the debugger waits in the first line of app_main(), as expected. After hitting <F5>, the program runs, but doesn't stop at the embedded asm line. Execution is stopped at a later breakpoint.

In contrast, stepping into the code somehow stops execution, gdb issuing "esp32.cpu0: Stepping doesn't seem to change PC! dsr=0x8080cc11". Although knowing that the 'stepping into'  scenario whould not match the intended behaviour of a breakpoint in the free running process, I tried to get over/out of this break. That didn't work either. I entered  'set $pc=$pc+3', as you proposed, and got 'Selected thread is running. (from interpreter-exec console "set $pc=$pc+3")'  an PC remained the same. 
I also tried  set $epc2 = $pc + 3 (according to https://visualgdb.com/documentation/startupdebug ) with the same effect, programm counter did not proceed.

So, to sum up, the inline asm break command doesn't work for me, even before deep sleep.
I know, that it should work, as the command is decribed as a part of the Xtensa instruction set ( https://0x04.net/~mwk/doc/xtensa.pdf, page 596)

alecks68, 2021-05-16:
Hi Ivan,

thanks for your hints and thanks also to nullptrdesaster to test it. 
I am using Platformio on Windows with 'framework = arduino'. In this context I cannot link cpu_ll_is_debugger_attached().
Nevertheless I tried __asm__ __volatile__("break 1, 1"); in a simple code without sleep and it does not stop for me either.

Maybe we can go one step back before we address wake_stub behaviour. 
The initial question was how we can make the debugger survive deep_sleep (=reset) or to reconnect it in a proper state afterwards.
Alternatively, what would be the best solution to have a close to real life emulation of deep sleep without reset.

erhankur, 2024-01-15:
Closed due to inactivity.

closed 2024-01-15 by erhankur