espressif/openocd-esp32 #144: ESP32-C3 JTAG -- incorrect documentation -- unable to debug (IDFGH-4760)

pjsg, 2021-02-20:
I'm trying to port my app to the early esp32-c3 devkit that you mailed me (thanks!). I realize that the JTAG is not available from the USB, but hopefully is available from the pins using an external JTAG device. However, the only documentation that I can find on which pins are which -- https://docs.espressif.com/projects/esp-idf/en/v4.3-beta1/esp32c3/api-guides/jtag-debugging/configure-other-jtag.html -- unfortunately has a table copied from the esp32-s2 page and talks about GPIOs that are way out of range for the ESP32-C3-DevKitM-1 V1.0

Is there a better table somewhere of the pin mapping?

I'm a bit stuck not being able to do any debugging to see what is going wrong...

Thanks.

igrr, 2021-02-20:
Could you please try running a debugging session from the command line to narrow down the cause of the issue?

1. Pull latest master branch of openocd-esp32 and build it. We have merged support for resetting the target yesterday, it might help.
2. Start OpenOCD: `src/openocd -s tcl -c interface/jlink.cfg -c target/esp32c3.cfg` (assuming you are using JLink now)
3. Create gdbinit file with the following content:
   ```
   target remote :3333
   mon reset halt
   thb app_main
   continue
   ```
3. In another terminal, run `idf.py flash monitor` in your project directory
3. In yet another terminal, run `riscv32-esp-elf-gdb build/app-name.elf -x gdbinit`

If that still doesn't work, please add `-d3` argument to OpenOCD command line, and post the logs you get (OpenOCD, serial monitor, GDB).

Thanks.

pjsg, 2021-02-20:
This works much better thank you. I now get:

```
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
```
when stepping through the code. It doesn't seem to interfere with normal use, so I'm happy to ignore it.....

Thank you.

igrr, 2021-02-20:
This warning was added recently in https://github.com/espressif/openocd-esp32/commit/9c56cb14952b411a2b62d9ffe5c8897d96505ca6, possibly there is some bug that it gets printed. @erhankur could you help troubleshoot this part?

pjsg, 2021-02-20:
I tried to reproduce with debugging turned on, and it didn't reproduce. It doesn't happen in every debugging session. So far I've made it happen twice (with maybe 5 attempts) when not running openocd -d, and never (with maybe 10 attempts) with openocd -d! 

I'm driving from eclipse if that makes a difference.

erhankur, 2021-02-21:
Hi @pjsg,

I will try to reproduce the error from the eclipse. In the meantime could you please try again from command line without -d option. Once you see the error you can change openocd debug level from gdb screen with `mon debug_level 3` 
Then please run some commands (step, continue, break etc..) and post the logs you collected.

Thanks

pjsg, 2021-02-23:
I attach the log from the openocd instance -- it actually reproduced on the first attempt.

[openocd.txt](https://github.com/espressif/openocd-esp32/files/6026206/openocd.txt)

pjsg, 2021-02-23:
And this was the gdb log:
```
~/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gdb build/nodemcu.elf
GNU gdb (crosstool-NG 1.24.0.123_64eb9ff) 8.3.1.20191211-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/nodemcu.elf...
(gdb) target remote :3333
Remote debugging using :3333
0x4208b4c8 in esp_pm_impl_waiti ()
    at /home/philip/eclipse-workspace/nodemcu/sdk/esp32-esp-idf/components/hal/esp32c3/include/hal/cpu_ll.h:156
156	    asm volatile ("wfi\n");
(gdb) set remote hardware-watchpoint-limit 2
(gdb) mon reset halt
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
(gdb) flushregs
Register cache flushed.
(gdb) thb app_main
Hardware assisted breakpoint 1 at 0x420082be: file /home/philip/eclipse-workspace/nodemcu/components/base_nodemcu/user_main.c, line 149.
(gdb) c
Continuing.
[New Thread 1070227208]
[New Thread 1070195484]
[New Thread 1070221580]
[New Thread 1070216412]
[New Thread 1070200532]
[New Thread 1070178176]
[Switching to Thread 1070198136]

Thread 1 hit Temporary breakpoint 1, app_main ()
    at /home/philip/eclipse-workspace/nodemcu/components/base_nodemcu/user_main.c:149
149	  task_init();
(gdb) step
FreeRTOS uxTaskNumber seems to be corrupted!
FreeRTOS uxTaskNumber seems to be corrupted!
FreeRTOS uxTaskNumber seems to be corrupted!
FreeRTOS uxTaskNumber seems to be corrupted!
task_init () at /home/philip/eclipse-workspace/nodemcu/components/task/task.c:105
105	  pending = xSemaphoreCreateBinary ();
(gdb) mon debug_level 3
debug_level: 3

(gdb) step
FreeRTOS uxTaskNumber seems to be corrupted!
FreeRTOS uxTaskNumber seems to be corrupted!
FreeRTOS uxTaskNumber seems to be corrupted!
FreeRTOS uxTaskNumber seems to be corrupted!
FreeRTOS uxTaskNumber seems to be corrupted!
FreeRTOS uxTaskNumber seems to be corrupted!
FreeRTOS uxTaskNumber seems to be corrupted!
FreeRTOS uxTaskNumber seems to be corrupted!
xQueueGenericCreate (uxQueueLength=uxQueueLength@entry=1, uxItemSize=uxItemSize@entry=0, 
    ucQueueType=ucQueueType@entry=3 '\003')
    at /home/philip/eclipse-workspace/nodemcu/sdk/esp32-esp-idf/components/freertos/queue.c:384
384			configASSERT( uxQueueLength > ( UBaseType_t ) 0 );
(gdb) quit
A debugging session is active.

	Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) y
Detaching from program: /home/philip/eclipse-workspace/nodemcu/build/nodemcu.elf, Remote target
Ending remote debugging.
[Inferior 1 (Remote target) detached]
```

erhankur, 2021-02-25:
Hi @pjsg,

Good news, I have re-produced the issue. Looks like ESP32-C3 rtos integration missing one necessary function call after reset the target. In each reset FreeRTOS `uxTaskNumber` variable set to zero but openocd does not cleanup rtos related counters. This breaks the synchronization.

Luckily, this is not blocker for debugging. Fix will be ready in the next release.

Thanks for your support.

dau42, 2021-04-26:
I have a similar issue getting JTAG to work with ESP32C3 using a flyswatter2.  The setup works as expected with an ESP32-DevKit-V4.    My "board" is an ESP32C3 Wroom with the following connections:
   Pin 1  3.3V with some caps to GND for decoupling
   Pin 2  Reset 10k pull-up with 1uF to GND
   Pin 3  TMS
   Pin 4  TDI
   Pin 5  TCK
   Pin 6  TDO
   Pin 9   GND 

My eclipse output is...

Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 6000 kHz

force hard breakpoints
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 6000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
semihosting is enabled

Error: Too large number of threads 4058895555!
Error: Too large number of threads 4058895555!
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
===== RISC-V Registers
(0) zero (/32)
(1) ra (/32)
(2) sp (/32)
(3) gp (/32)
(4) tp (/32)
(5) t0 (/32)
(6) t1 (/32)
(7) t2 (/32)
(8) fp (/32)
(9) s1 (/32)
(10) a0 (/32)
(11) a1 (/32)
(12) a2 (/32)
(13) a3 (/32)
(14) a4 (/32)
(15) a5 (/32)
(16) a6 (/32)
(17) a7 (/32)
(18) s2 (/32)
(19) s3 (/32)
(20) s4 (/32)
(21) s5 (/32)
(22) s6 (/32)
(23) s7 (/32)
(24) s8 (/32)
(25) s9 (/32)
(26) s10 (/32)
(27) s11 (/32)
(28) t3 (/32)
(29) t4 (/32)
(30) t5 (/32)
(31) t6 (/32)
(32) pc (/32)
[... 202 lines cut ...]

If I run the following on the command line to bypass Eclipse I get something similar...

% openocd -f $OPENOCD_SCRIPTS/interface/ftdi/flyswatter2.cfg -f $OPENOCD_SCRIPTS/target/esp32c3.cfg
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 6000 kHz

force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 6000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset

gerekon, 2021-04-27:
> Pin 3 TMS
Pin 4 TDI
Pin 5 TCK
Pin 6 TDO

@dau42 You seem to have another issue. Check you JTAG pins connection
The proper connection is 
GPIO4 -> TMS
GPIO5 -> TDI
GPIO6 -> TCK
GPIO7 -> TDO

gerekon, 2021-04-27:
@dau42 Note that before https://github.com/espressif/esp-idf/commit/cccdb3e4aaabbe6946d3ab798c8d1fa7b23edef3. Blink example used GPIO5 to control LED. GPIO5 is TDI on ESP32C3

dau42, 2021-04-27:
I have switched to ESP32-C3 Dev Kit V2.  Connections for JTAG are the same.  
  GPIO4 -> TMS 
  GPIO5 -> TDI 
  GPIO6 -> TCK 
  GPIO7 -> TDO
I have blink GPIO set to GPIO10.  I can successfully flash using the USB connection but get the same results when using Command Line JTAG

```
% openocd -f $OPENOCD_SCRIPTS/interface/ftdi/flyswatter2.cfg -f $OPENOCD_SCRIPTS/target/esp32c3.cfg
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 6000 kHz

force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 6000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
```
I have set aside Eclipse for now to concentrate on getting command line flash via JTAG working.

Also, do you know where I can get the example code that comes on a virgin ESP32-C3 Dev Kit?  It looks like it simple lights up the SK68XX RGB LED with cyan color.

dau42, 2021-04-28:
The "Hart 0 unexpectedly reset" messages are caused by a mismatch of the ESP32c3 revision.  In config there is a "Component config\ESP32c3 Specific\Minimum Supported ESP32C3 Revision" setting that defaults to Rev 3.  My ESP32c3's are Revision 2 parts.  This causes the boot loader to continually reset the part.  If you monitor the TX line the boot loader sends out a message to warn about this condition.
```
ESP-IDF v4.4-dev-1183-g9d34a1cd42-dirty 2nd stage bootloader[0m
[0;32mI (35) boot: compile time 14:26:18[0m
[0;32mI (35) boot: chip revision: 2[0m
[0;31mE (38) boot_comm: This chip is revision 2 but the application is configured for minimum revision 3. Can’t run.[0m
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d11b0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1658
load:0x403ce000,len:0x8b4
load:0x403d0000,len:0x2ba4
entry 0x403ce000
[0;32mI (34) boot: ESP-IDF v4.4-dev-1183-g9d34a1cd42-dirty 2nd stage bootloader[0m
[0;32mI (35) boot: compile time 14:26:18[0m
[0;32mI (35) boot: chip revision: 2[0m
[0;31mE (38) boot_comm: This chip is revision 2 but the application is configured for minimum revision 3. Can’t run.[0m
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
```

dau42, 2021-04-28:
I have ESP32c3 Debug up and running on the Command Line and in Eclipse...

For the command line I use:
```
openocd -f $OPENOCD_SCRIPTS/interface/ftdi/flyswatter2.cfg  -f $OPENOCD_SCRIPTS/target/esp32c3.cfg 
```
to start up OpenOCD.  I then open a second terminal session and use:
```
~/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gdb 
```
to start the correct version of gdb.  Within gdb I use these commands:
```
set arch risk:rv32
target remote tcp::3333
```
to make the connection and start debugging.  At this point everything works.

For Eclipse I setup a Debug Configuration with the OpenOCD config options as:
```
-s ${openocd_path}/share/openocd/scripts  -f interface/ftdi/flyswatter2.cfg -f target/esp32c3.cfg
```
and the correct risk version of gdb with the commands:
```
set arch riscv:rv32
set mem inaccessible-by-default off
```
At this point debugging on the ESP32C3 works fine on Eclipse.

dau42, 2021-04-28:
I still cannot program the part via JTAG.  When I use 
```
% openocd -f $OPENOCD_SCRIPTS/interface/ftdi/flyswatter2.cfg  -f $OPENOCD_SCRIPTS/target/esp32c3.cfg -c "program_esp build/blink.bin 0x10000 verify exit"
```
I get...
```
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 6000 kHz

force hard breakpoints
Info : clock speed 6000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:100: Error: invalid command name "esp"
in procedure 'program_esp' 
at file "~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 100
```

Something is still amiss with programming my ESP32c3 via JTAG

igrr, 2021-04-29:
@dau42 Flash programming over JTAG and flash breakpoints aren't yet implemented for ESP32-C3. We are working on it and this will come in the next release.

closed 2021-06-03 by espressif-bot

chegewara, 2021-06-13:
I believe i have the latest openocd installed but i am having the same issue as OP when i am trying to use buildin JTAG:
```
Open On-Chip Debugger v0.10.0-esp32-20210401 (2021-04-01-15:45)
Licensed under GNU GPL v2
For bug reports, read
 http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info : hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : dropped 'gdb' connection
```
GDB logs:
```
GNU gdb (crosstool-NG 1.24.0.123_64eb9ff) 8.3.1.20191211-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
gdbinit:1: Error in sourced command file:
bfd requires xlen 8, but target has xlen 4
```
@igrr

erhankur, 2021-06-13:
Hi @chegewara, 
Looks like gdb could not recognize the correct target len. It thinks as 64 bit instead of 32. If you give your elf file, it will get correct XLEN from there.

`riscv32-esp-elf-gdb build/hello-world.elf -x gdbinit`

Or you can explicitly force gdb to 32-bit using `set arch riscv:rv32`

chegewara, 2021-06-13:
@erhankur Thanks, adding elf file in command line helped.

giacomo-zanichelli, 2021-08-03:
> dau42 Flash programming over JTAG and flash breakpoints aren't yet implemented for ESP32-C3. We are working on it and this will come in the next release.

Hi, is there an update on when programming over JTAG may be available? Thanks

gerekon, 2021-08-03:
@giacomo-zanichelli [It was released](https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20210721). But recommended openocd version is not updated in IDF yet