espressif/openocd-esp32 #145: Trying to use a ST-Link/v2 adapter for jtag debugging

yz70s, 2021-03-05:
Hello, i am trying to use a stlink  adapter and i have managed to find the commands (shown below) to initialize it, however it gives an error at the "target create" command.
To test you don't need an actual stlink plugged-in, just create a "my.cfg" file with the following content

```
interface hla
hla_layout stlink
hla_device_desc "ST-LINK"
hla_vid_pid 0x0483 0x3744 0x0483 0x3748 0x0483 0x374b 0x0483 0x374d 0x0483 0x374e 0x0483 0x374f 0x0483 0x3752 0x0483 0x3753

adapter_khz 100

transport select hla_jtag

hla newtap esp32 cpu0 -irlen 5 -expected-id 0x120034e5
hla newtap esp32 cpu1 -irlen 5 -disable -expected-id 0x120034e5

target create esp32 esp32 -chain-position esp32.cpu0
```
and try it with
```
openocd -f "my.cfg" -d4
```
at the end you will see the following messages
```
Debug: 69 258 command.c:143 script_debug(): command - target target create esp32 esp32 -chain-position esp32.cpu0
Error: 70 264 target.c:5550 target_create(): The selected transport doesn't support this target
User : 71 268 command.c:695 command_run_line(): invalid command name "esp32"

UweBonnes, 2021-03-05:
Stlink firmware denies JTAG access to non ST devices. StlinkV3 even denies SWD access to non-ST devices and often fail to access STM32 devices that STLINKV2 can accesss

yz70s, 2021-03-05:
Thanks, moreover (by looking at file stlink_usb.c) it seems that it cannot be used as a generic jtag tool since there are only high-level debug commands.

Closing.

closed 2021-03-05 by yz70s

UweBonnes, 2021-03-05:
You can use stlink as generic SWJ tool with other firmware, e.g. blackmagic. However there is no openocd blackmagic backend yet and blackmagic does not support ESP32 yet (OCD-314)