espressif/openocd-esp32 #146: Tensilica On Chip Debugging Guide

UweBonnes, 2021-03-05:
Traces on the web indicate that a document describing  Tesilica On-Chip Debug / onchip_debug_guide.pdf  exists.  However I did not find that document available. I would like to have a look at that document it access implementing  ESP32 debugging in the blackmagic probe.

Can anybody help, with that document or in other ways?

Thanks

igrr, 2021-03-05:
Greetings @UweBonnes, unfortunately Espressif is unable to distribute the document publicly. However I'm sure we can come up with a solution to give you enough information about the OCD interface. Could you please email me at ivan at company name com?

nullptrdesaster, 2021-05-15:
may be this from 2010 helps?  https://0x04.net/~mwk/doc/xtensa.pdf

UweBonnes, 2021-05-15:
Nice document, thaks for the pointer. But not the on-chip debug document I am searching...

closed 2022-10-25 by gerekon

fariouche, 2023-07-26:
Not sure if this document can help
https://www2.lauterbach.com/pdf/debugger_xtensa.pdf

UweBonnes, 2023-07-27:
This is not the low level JTAG protocol that is needed for reimplementing.