espressif/openocd-esp32 #147: esp32s with FT232H: cpu id confusion

olegst90, 2021-03-11:
Hi everybody. I'm debugging esp32s (AI Thinker ESP32-CAM board, to be more specific) using FT232H-based JTAG debugger.
I successfully launched openocd-esp32 and it connected to the target, connected from xtensa gdb (from sdk) and set a breakpoint. Target halted when one of the cores reached the address. However, gdb shows me registers of another core, not one which reached the breakpoint. Here is the output:

openocd:

```
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu0: Detected debug stubs @ 3ffc39e0
Info : esp32.cpu0: Target halted, PC=0x4015EE8A, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x400D1312, debug_reason=00000001
Warn : No symbols for FreeRTOS!
```

gdb:
```

(gdb) info break
Num     Type           Disp Enb Address    What
4       hw breakpoint  keep y   0x400d1312 
(gdb) frame
#0  0x4015ee8a in ?? ()
(gdb) 
```

as you can see, although it did halt on 0x400d1312, the active frame in gdb is one of the other core.

What did I do wrong?

gerekon, 2021-03-12:
@olegst90 Hi

It looks like FreeRTOS is not running yet when your breakpoint hit. Can you confirm this? 
Could you post full OpenOCD log (with `-d3` option)?

gerekon, 2021-03-12:
To debug w/o RTOS I would suggest to run OpenOCD as
`openocd  -c 'ESP_RTOS none' -f board/esp32-wrover-kit-1.8v.cfg` or `openocd  -c 'ESP_RTOS hwthread' -f board/esp32-wrover-kit-1.8v.cfg`.

olegst90, 2021-03-12:
> To debug w/o RTOS I would suggest to run OpenOCD as
> `openocd -c 'ESP_RTOS none' -f board/esp32-wrover-kit-1.8v.cfg` or `openocd -c 'ESP_RTOS hwthread' -f board/esp32-wrover-kit-1.8v.cfg`.

It worked for me when I run it as:
`openocd -c 'set ESP_RTOS hwthread' -f board/esp32-wrover-kit-1.8v.cfg`

Thank you for your help.

closed 2023-01-17 by erhankur