espressif/openocd-esp32 #149: Consecutive apptrace statements not output to the JTAG console without stepping through (IDFGH-4906)

TobiasBoeschMiele, 2021-03-16:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v4.4-dev-4-g73db14240
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [].
- Using an IDE?: [Yes eclipse-plugin]
- Power Supply:         [USB]

## Problem Description

From two consecutive apptrace instruction only the last one gets output to the JTAG console unless one steps through each instruction manually.
I also tried to [set the openocd command for tracing to blocking ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html#openocd-application-level-tracing-commands)without success.

### Expected Behavior

Running the program puts out all trace data each after another without halting the program.

### Actual Behavior

Running the program only puts out the trace data of the last apptrace instruction

### Steps to reproduce

1. Setup a Hello World example
2. Insert the code shown below in the hello_world_main.c
3. Setup a JTAG flashing debug configuration (see attachement)
4. Setup a JTAG debug configuration with openocd tracing (see attachement)
5. Enable application level tracing in sdkconfig
6. Build
7. Run by stepping through
8. Reset
9. Run without stepping
10. Compare console output of step 7 and 9

### Code to reproduce this issue

```cpp
/* Hello World Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "esp_app_trace.h"
#include "esp_log.h"

void app_main(void)
{
    char *ptr = NULL;

    ptr = (char *)esp_apptrace_buffer_get(ESP_APPTRACE_DEST_TRAX, 32, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    if (ptr == NULL) {
        ESP_LOGE("---", "Failed to get buffer!");
    }
    sprintf(ptr, "__Hello World!\n");
    esp_err_t res = esp_apptrace_buffer_put(ESP_APPTRACE_DEST_TRAX, (uint8_t*) ptr, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    res += esp_apptrace_flush(ESP_APPTRACE_DEST_TRAX, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    if (res != ESP_OK) {
        /* in case of error host tracing tool (e.g. OpenOCD) will report incomplete user buffer */
        ESP_LOGE("---", "Failed to put buffer!");
    }

    ptr = (char *)esp_apptrace_buffer_get(ESP_APPTRACE_DEST_TRAX, 32, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    if (ptr == NULL) {
        ESP_LOGE("---", "Failed to get buffer!");
    }
    sprintf(ptr, "__Hello Again!\n");
    res = esp_apptrace_buffer_put(ESP_APPTRACE_DEST_TRAX, (uint8_t*) ptr, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    res += esp_apptrace_flush(ESP_APPTRACE_DEST_TRAX, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    if (res != ESP_OK) {
        /* in case of error host tracing tool (e.g. OpenOCD) will report incomplete user buffer */
        ESP_LOGE("---", "Failed to put buffer!");
    }
}
```

## Debug Logs
Stepping through:
```
[...]
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : cpu0: No free slots to add HW breakpoint!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
__Hello World!
Info : Target halted. CPU0: PC=0x400D4024 (active)
Info : Target halted. CPU1: PC=0x400E2B1E 
Warn : cpu0: HW breakpoint not found!
[...]
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : cpu0: No free slots to add HW breakpoint!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
__Hello Again!
Info : Target halted. CPU0: PC=0x400D407C (active)
Info : Target halted. CPU1: PC=0x400E2B1E 
[...]
```

Running:
```
Warn : cpu0: No free slots to add HW breakpoint!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Target halted. CPU0: PC=0x400D3FE0 (active)
Info : Target halted. CPU1: PC=0x400E2B1E 
Warn : cpu0: HW breakpoint not found!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
__Hello Again!
```

## Other items if possible
[hello_world flash Configuration.launch.txt](https://github.com/espressif/esp-idf/files/6130161/hello_world.flash.Configuration.launch.txt)
[hello_world debug Configuration.launch.txt](https://github.com/espressif/esp-idf/files/6130162/hello_world.debug.Configuration.launch.txt)
[esp-huzzah-32.cfg.txt](https://github.com/espressif/esp-idf/files/6130168/esp-huzzah-32.cfg.txt) (if interesting - nothing special here I think)

gerekon, 2021-03-16:
It seems to be OpenOCD issue. So moved it here.

TobiasBoeschMiele, 2021-03-18:
> It seems to be OpenOCD issue. So moved it here.

Thanks.

>```
> tb app_main
> commands
> mon esp32 apptrace start con: 0
> c
> end
>```

I've added that to the gdb init commands in eclipse, but I cannot test it since the debug session does not fully start anymore. Do you know why? Logs of the starting and not starting debug session below:

Log not using gdb commands but `-c "init; esp32 apptrace start con: 0"` when starting openocd which ends in a working debugging session:
```
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info: esp-huzzah-32.cfg is not an official configuration file.
Info : Configured 2 cores
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. CPU0: PC=0x400D3FE0 (active)
Info : Target halted. CPU1: PC=0x400E2B1E 
Info : Listening on port 3333 for gdb connections
Info : Total trace memory: 16384 bytes
App trace params: from 2 cores, size 4294967295 bytes, stop_tmo -1 s, poll period 0 ms, wait_rst 0, skip 0 bytes
Connect targets...
Info : Targets connected.
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 23 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 76 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.flash' size 4096 KB
Info : Using flash bank 'esp32.flash' size 4096 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 23 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 76 KB
Info : Using flash bank 'esp32.irom' size 80 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 23 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 76 KB
Info : Using flash bank 'esp32.drom' size 24 KB
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400 
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
[... 227 lines cut ...]
Info : Using flash bank 'esp32.drom' size 24 KB
undefined debug reason 8 - target needs reset
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400 
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400 
```

The entry `undefined debug reason 8 - target needs reset` seems new in the NOT working log.

SimonPVS, 2021-04-26:
Any updates on this?

gerekon, 2021-04-26:
@SimonPVS @TobiasBoeschMiele  Sorry, not yet. I asked a guy to have a look at it.

erhankur, 2021-05-20:
@SimonPVS @TobiasBoeschMiele you can try with the latest commit. Please make sure to wait host connection before sending data. 

```c
    while (!esp_apptrace_host_is_connected(ESP_APPTRACE_DEST_TRAX))       
          vTaskDelay(1);  
```

TobiasBoeschMiele, 2021-05-25:
> Please make sure to wait host connection before sending data.

Waiting for something here is not what I would like to do. Can I also ask if a connection is persistent and then "print" the data?

I'll try that and report back.

erhankur, 2021-05-25:
Yes. Once you connect before sending the first data, it should stay connected until the stop condition is met.

TobiasBoeschMiele, 2021-06-14:
I was able to test this with commit 1d7068e4be430edd92bb63f2d922036dcf5c3cc1.
I added 

> ```c
> while (!esp_apptrace_host_is_connected(ESP_APPTRACE_DEST_TRAX))       
>           vTaskDelay(1);  
> ```

in the hello-world example code mentioned above in this issue as the first statement:

> ```c++
> void app_main(void)
> {
>     char *ptr = NULL;
>
>     // Added:
>     while (!esp_apptrace_host_is_connected(ESP_APPTRACE_DEST_TRAX))       
>           vTaskDelay(1);  
> 
>     ptr = (char *)esp_apptrace_buffer_get(ESP_APPTRACE_DEST_TRAX, 32, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
>     if (ptr == NULL) {
>         ESP_LOGE("---", "Failed to get buffer!");
>     }
> [...]
> ```

That makes the task wait forever.

Without the waiting statement the example code does what it did before updating to 1d7068e4be430edd92bb63f2d922036dcf5c3cc1: It only 

> puts out the trace data of the last apptrace instruction

erhankur, 2021-06-14:
Hi @TobiasBoeschMiele ,

Did you update your opencod? 

I have re-tested with the latest openocd [commit](https://github.com/espressif/openocd-esp32/commit/fd267384f3e8c5ebd94f3c4ff45952d5c325ac9a) and with the [idf](https://github.com/espressif/esp-idf/commit/1d7068e4be430edd92bb63f2d922036dcf5c3cc1) you mentioned. It is working as expected. 

```
:apptrace-test $ ocd32ethkit -c "init; esp32 apptrace start con: 0"
Open On-Chip Debugger  v0.10.0-esp32-20210401-39-gfd267384 (2021-06-14-21:36)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : Listening on port 3333 for gdb connections
Info : Total trace memory: 16384 bytes
App trace params: from 2 cores, size 4294967295 bytes, stop_tmo -1 s, poll period 0 ms, wait_rst 0, skip 0 bytes
Connect targets...
Info : esp32.cpu0: Target halted, PC=0x400E3A8E, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x400E3A8E, debug_reason=00000000
Info : Targets connected.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
__Hello World!
__Hello Again!
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Warn : esp32.cpu0 apptrace connection is lost. Re-connect.
Warn : esp32.cpu1 apptrace connection is lost. Re-connect.
Warn : esp32.cpu1 apptrace connection is lost. Re-connect.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Warn : esp32.cpu1 apptrace connection is lost. Re-connect.
Warn : esp32.cpu0 apptrace connection is lost. Re-connect.
Warn : esp32.cpu1 apptrace connection is lost. Re-connect.
__Hello World!
__Hello Again!
```

````
I (185) cpu_start: App cpu up.
I (213) cpu_start: Pro cpu start user code
I (214) cpu_start: cpu freq: 160000000
I (214) cpu_start: Application information:
I (218) cpu_start: Project name:     apptrace
I (223) cpu_start: App version:      1
I (227) cpu_start: Compile time:     Jun 14 2021 21:15:41
I (234) cpu_start: ELF file SHA256:  9efd50f28d94e67b...
I (240) cpu_start: ESP-IDF:          v4.4-dev-1594-g1d7068e4be
I (246) heap_init: Initializing. RAM available for dynamic allocation:
I (253) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (259) heap_init: At 3FFB30B0 len 0002CF50 (179 KiB): DRAM
I (266) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (272) heap_init: At 3FFE4350 len 00013CB0 (79 KiB): D/IRAM
I (278) heap_init: At 4008B19C len 00014E64 (83 KiB): IRAM
I (285) esp_apptrace: Initialized TRAX on CPU0
I (290) spi_flash: detected chip: generic
I (294) spi_flash: flash io: dio
W (298) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (313) cpu_start: Starting scheduler on PRO CPU.
I (203) esp_apptrace: Initialized TRAX on CPU1
I (0) cpu_start: Starting scheduler on APP CPU.
E (1997) ---: Heartbeat!
E (2747) ---: Heartbeat!
E (3497) ---: Heartbeat!
E (4247) ---: Heartbeat!
````

TobiasBoeschMiele, 2021-06-15:
> Did you update your openocd?

Not intentionally. I updated esp-idf to the mentioned commit (reset hard with all submodules) and then started the tools install again. (I use the eclipse plugin and used the GUI to start the tools install)
That should also update openocd I think.
I'll test it again and check my version of openocd. With which openocd version did you test it?

I also use a different command for debugging, too I think. In the eclipse GUI I have a debug configuration with this command:

[openocdExecutable] `-s ${openocd_path}/share/openocd/scripts -f ${project_loc}/esp-huzzah-32.cfg -c "init; esp32 apptrace start con:"`

The huzzah configuration just lists the JTAG interface and adapter speed.

gerekon, 2021-06-15:
@TobiasBoeschMiele 

> Not intentionally. I updated esp-idf to the mentioned commit (reset hard with all submodules) and then started the tools install again. (I use the eclipse plugin and used the GUI to start the tools install)

That fix in OpenOCD is not released yet. So you need to build OpenOCD from `master`. Will be released in the nearest future. We are testing new release.

@erhankur Am I right?

gerekon, 2021-06-15:
@TobiasBoeschMiele 

If you are not able to build OpenOCD your self we can upload binaries for your OS for pre-release check

erhankur, 2021-06-15:
@gerekon  you are right. Build from master branch is needed to validate. 

@TobiasBoeschMiele Regarding openocd configuration before -c part can be different according to your board.  For apptrace commands you can read details from  https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html#openocd-application-level-tracing-commands

TobiasBoeschMiele, 2021-06-15:
> @TobiasBoeschMiele
> 
> If you are not able to build OpenOCD yourself we can upload binaries for your OS for pre-release check

That would be helpful. I don't kmow if I need special tools for building, but I don't have administrative rights on my machine. So it might take a while.

TobiasBoeschMiele, 2021-06-15:
> @gerekon you are right. Build from master branch is needed to validate.
>
> @TobiasBoeschMiele Regarding openocd configuration before -c part can be different according to your board. For apptrace [...]

We come back to check that if there is the issue persists after using the correct openocd version. Thanks.

erhankur, 2021-06-15:
You can try with this windows build.

[openocd-esp32-win32-0.10.0-esp32-20210401-39-gfd267384f.zip](https://github.com/espressif/openocd-esp32/files/6655299/openocd-esp32-win32-0.10.0-esp32-20210401-39-gfd267384f.zip)

TobiasBoeschMiele, 2021-06-15:
> You can try with this windows build.
> 
> [openocd-esp32-win32-0.10.0-esp32-20210401-39-gfd267384f.zip](https://github.com/espressif/openocd-esp32/files/6655299/openocd-esp32-win32-0.10.0-esp32-20210401-39-gfd267384f.zip)

Using this build makes the issue dissapear.
I don't need to add the following statement to wait for the connection to be established, but that might be just a lucky case where the connection is already present.

> @SimonPVS @TobiasBoeschMiele you can try with the latest commit. Please make sure to wait host connection before sending data.
> 
> ```c
>     while (!esp_apptrace_host_is_connected(ESP_APPTRACE_DEST_TRAX))       
>           vTaskDelay(1);  
> ```

closed 2021-06-15 by TobiasBoeschMiele

erhankur, 2021-06-15:
Yes, it will not be stable without waiting the connection. That was the lucky case.

TobiasBoeschMiele, 2021-06-16:
Thanks for your help.