espressif/openocd-esp32 #150: LIBUSB_ERROR_TIMEOUT when programming ESP32-S2-WROOM with J-Link EDU Mini

luizribeiro, 2021-03-28:
I'm trying to program an ESP32-S2-WROOM with a J-Link EDU Mini and running into issues with programming.

This is the command I'm running:

```
/home/luiz/openocd-esp32/src/openocd -d3 -s /home/luiz/openocd-esp32/tcl -f interface/jlink.cfg -f target/esp32s2.cfg -c "adapter_khz 1000" -c "program_esp {{/home/luiz/hw-sandbox/esp32-s2/.pio/build/esp32doit-devkit-v1/firmware.bin}} 0x10000 verify" -c "reset run; shutdown"
```

This is the output with debugging enabled:
https://gist.github.com/luizribeiro/e69ba8896c5514e4d87fa530bf07a307

Note that the timeouts happen just after `libjaylink` adjusts the buffer size:

```
Debug: 1767 305 target.c:2236 target_write_buffer(): writing buffer of 512 byte at 0x40030000
Debug: 1770 306 jlink.c:534 jaylink_log_handler(): Adjusted buffer size to 4096 bytes.
Warn : 1782 1327 jlink.c:534 jaylink_log_handler(): Failed to send data to device: LIBUSB_ERROR_TIMEOUT.
Warn : 1784 2327 jlink.c:534 jaylink_log_handler(): Failed to send data to device: LIBUSB_ERROR_TIMEOUT.
Error: 1786 2327 jlink.c:534 jaylink_log_handler(): Sending data to device timed out.
```

I made a very naive change to `libjaylink` by changing `CHUNK_SIZE` on `libjaylink/transport_usb.c` to `32768` and after that the programming was successful.

Any idea why this happens? Maybe it's an incompatibility between `openocd-esp32` and the J-Link EDU Mini?

igrr, 2021-03-28:
Hi @luizribeiro, it is quite possible that `jlink` driver in OpenOCD doesn't fully support J-Link Edu Mini. Is it only flashing the code that is failing for you, or is debugging failing as well? Similar issues have been reported with this adapter in the past (e.g. https://sysprogs.com/w/forums/topic/jtag-debugging-esp32-with-jlink-edu-mini/), and we had assumed it was because the adapter is restricted to Cortex-M cores by Segger.  It's mentioned in the [product announcement](https://www.segger.com/fileadmin/documents/Press_Releases/PR_170606_SEGGER_JLinkEDUMini.pdf), although it isn't clear if the limitation applies to J-Link tools only, or is enforced in the adapter firmware somehow.

luizribeiro, 2021-03-29:
@igrr, thanks for the quick response!

To be clear, after updating `CHUNK_SIZE` on the driver flashing worked. Debugging wasn't working before that change (since the connection failed really early in the process), but I'm not sure if it was working after.

Indeed, Segger seems to be fairly unclear about which devices are supported by the J-Link Edu Mini. I believe the limitation is to SWD, since it only contains the SWD pins? Perhaps that's why they only mention Cortex-M on the announcement?

I'm still running into issues with the flashed ESP32-S2 though: making pin 13 simply oscillate between low/high with a simple program didn't work after flashing with the J-Link Edu Mini (even though `openocd` tells me the flashing was successful). That might be my hardware's fault or just a software problem on my `platformio` setup.

I'll try flashing the ESP32 with a J-Link Edu (non-Mini) later this week and see if I get different results.

igrr, 2021-03-29:
> I'm still running into issues with the flashed ESP32-S2 though: making pin 13 simply oscillate between low/high with a simple program didn't work after flashing

Just to check, if this is the first time you are flashing this module, did you flash the bootloader and the partition table binaries as well? The command you have posted only contains the application binary.

luizribeiro, 2021-03-29:
@igrr, I have not! Is it possible to do this through JTAG?

igrr, 2021-03-29:
Yes, same way as you flash the application, you can flash the bootloader and the partition table to their respective offsets. The bootloader should be at 0x1000 and the partition table is usually at 0x8000, although the latter can be adjusted in menuconfig.
IDF build system produces the list of files and offsets in build/flasher_args.json, under "flash_files" dictionary. I don't know if platformio does that as well.

luizribeiro, 2021-03-31:
I was able to successfully flash my ESP32! I had a few problems:

1. There is indeed some sort of bug with the J-Link driver when using the J-Link EDU Mini. I can only flash if I change `CHUNK_SIZE` to `32768` (I didn't try other values).
2. I had to flash the bootloader and the partition table along with the firmware. This command worked: `openocd -s /home/luiz/ocd/tcl -f interface/jlink.cfg -f target/esp32s2.cfg -c "adapter_khz 4000" -c "program_esp .pio/build/esp32dev/bootloader.bin 0x1000 verify" -c "program_esp .pio/build/esp32dev/partitions.bin 0x8000 verify" -c "program_esp .pio/build/esp32dev/firmware.bin 0x10000" -c "reset run; shutdown"
3. My PCB is using GPIO0, which was preventing the ESP32 from booting as it is a strapping pin. I guess I'll have to use another GPIO pin on my next PCB version :)

StarWitch, 2021-04-01:
Commenting to say I also had this problem with the ESP32-S2-WROVER. I'll try your workaround, @luizribeiro. 

Were you also able to get debugging working?

emileb, 2021-07-29:
Anyone get the J-Link EDU working to debug the ESP32 yet? Getting this as well:

`Info : tcl server disabled
Info : telnet server disabled
Info : J-Link EDU Mini V1 compiled Jun  9 2020 13:36:46
Info : Hardware version: 1.00
Info : VTarget = 3.272 V
2
Info : Reduced speed from 5000 kHz to 4000 kHz (maximum).
Info : clock speed 5000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : accepting 'gdb' connection from pipe
Warn : No symbols for FreeRTOS!
Info : esp32.cpu0: Target halted, PC=0x400E29B3, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x400E2BBE, debug_reason=00000000
2
Warn : Failed to send data to device: LIBUSB_ERROR_TIMEOUT.
Error: Sending data to device timed out.
Error: transport_write() failed: timeout occurred.
Error: jaylink_jtag_io() failed: timeout occurred.
Warn : Last read operation left 2048 bytes.
2
Warn : Failed to send data to device: LIBUSB_ERROR_TIMEOUT.
Error: Sending data to device timed out.
Error: transport_write() failed: timeout occurred.
Error: jaylink_jtag_io() failed: timeout occurred.
Warn : Last read operation left 2048 bytes.
2
Warn : Failed to send data to device: LIBUSB_ERROR_TIMEOUT.`

luizribeiro, 2021-07-30:
@emileb, did you try the patch to `libjaylink` I mentioned above?

emileb, 2021-07-30:
> @emileb, did you try the patch to `libjaylink` I mentioned above?

Thanks for the reply. No not yet, I am on Windows so recompiling the library is a bit harder. Willing to give it a try if there is a good chance this will fix debugging (as well as flashing). Certainly looks like the same issue you had.

erhankur, 2023-01-17:
Closing this. If there is still an issue with our latest release, feel free to re-open.