espressif/openocd-esp32 #151: Cannot run the latest v0.10.0-esp32-20210329 on Windows

valeros, 2021-03-30:
Hello, it seems that the latest release is broken on Windows for some reason:
[image]

edbek, 2021-03-30:
Yes, I confirm, I have the same

gerekon, 2021-03-30:
Hi @valeros ! 
Sorry for inconvenience. Since we added compressed uploads to flash openocd needs `zlib` now. Unfortunately we missed it in Windows release distro. We will fix it soon. As a temporary workaround if you have mingw32 installed you can run openocd from its shell or copy `libgcc_s_dw2-1.dll ` and `zlib1.dll ` to openocd bin dir.

igrr, 2021-04-09:
Should be fixed by https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20210401.

closed 2021-04-09 by igrr