espressif/openocd-esp32 #152: Error: Unsupported DTM version: 14

MabezDev, 2021-04-20:
Trying to interrogate the riscv core in the esp32c3 through a segger j-link, unfortunately I am running into the output in the issue title.

Full output:
```bash
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
force hard breakpoints
none
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link V10 compiled Jul 23 2019 13:46:10
Info : Hardware version: 10.10
Info : VTarget = 3.270 V
Info : clock speed 1000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Error: Unsupported DTM version: 14
Info : Listening on port 3333 for gdb connections
```

Output with `-d3`:
```bash
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 13 1 options.c:57 configuration_output_handler(): debug_level: 3
User : 14 1 options.c:57 configuration_output_handler(): 
Debug: 15 1 options.c:181 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 16 1 options.c:182 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 17 1 options.c:183 add_default_dirs(): exepath=/home/mabez/programs/openocd-esp32/bin
Debug: 18 1 options.c:184 add_default_dirs(): bin2data=../share/openocd
Debug: 19 1 configuration.c:42 add_script_search_dir(): adding /home/mabez/.openocd
Debug: 20 1 configuration.c:42 add_script_search_dir(): adding /home/mabez/programs/openocd-esp32/bin/../share/openocd/site
Debug: 21 1 configuration.c:42 add_script_search_dir(): adding /home/mabez/programs/openocd-esp32/bin/../share/openocd/scripts
Debug: 22 1 configuration.c:97 find_file(): found openocd.cfg
Debug: 23 1 command.c:143 script_debug(): command - adapter_khz adapter_khz 1000
Debug: 25 1 core.c:1704 jtag_config_khz(): handle jtag khz
Debug: 26 1 core.c:1667 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 27 1 core.c:1667 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 28 1 configuration.c:97 find_file(): found /home/mabez/programs/openocd-esp32/bin/../share/openocd/scripts/interface/jlink.cfg
Debug: 29 1 command.c:143 script_debug(): command - interface interface jlink
Debug: 31 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 32 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 33 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 34 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 35 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 36 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 37 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 38 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 39 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 40 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 41 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 42 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 43 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 44 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 45 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 46 1 command.c:355 register_command_handler(): registering 'jlink'...
Debug: 47 1 configuration.c:97 find_file(): found /home/mabez/programs/openocd-esp32/bin/../share/openocd/scripts/target/esp32c3.cfg
Debug: 48 1 command.c:143 script_debug(): command - transport transport select jtag
Debug: 49 1 command.c:355 register_command_handler(): registering 'jtag_flush_queue_sleep'...
Debug: 50 1 command.c:355 register_command_handler(): registering 'jtag_rclk'...
Debug: 51 1 command.c:355 register_command_handler(): registering 'jtag_ntrst_delay'...
Debug: 52 1 command.c:355 register_command_handler(): registering 'jtag_ntrst_assert_width'...
Debug: 53 1 command.c:355 register_command_handler(): registering 'scan_chain'...
Debug: 54 1 command.c:355 register_command_handler(): registering 'jtag_reset'...
Debug: 55 1 command.c:355 register_command_handler(): registering 'runtest'...
Debug: 56 1 command.c:355 register_command_handler(): registering 'irscan'...
Debug: 57 1 command.c:355 register_command_handler(): registering 'verify_ircapture'...
[... 221 lines cut ...]
Debug: 402 15 tcl.c:498 handle_nand_init_command(): Initializing NAND devices...
Debug: 403 15 command.c:143 script_debug(): command - pld pld init
Debug: 405 15 pld.c:206 handle_pld_init_command(): Initializing PLDs...
Debug: 406 15 gdb_server.c:3584 gdb_target_start(): starting gdb server for esp32c3 on 3333
Info : 407 15 server.c:311 add_service(): Listening on port 3333 for gdb connections
```
## Setup

Using the latest release, `2021-04-01`, my openocd file contains the following:

```bash

adapter_khz 1000

# jlink
source [find interface/jlink.cfg]

# Source the ESP32-C3 configuration file
source [find target/esp32c3.cfg]

set ESP32_RTOS none
```
I believe the wiring is correct as the JTAG tap is correctly connected to and printed to stdout before the core examine fails. 

Let me know if any other information is required.

igrr, 2021-04-20:
What application have you flashed into the ESP32-C3? The "blink" example used to toggle GPIO5, until this was changed recently. GPIO5 is one of the JTAG pins on ESP32-C3, and overriding its function from the app may cause such an issue.

Please try flashing the hello_world example and see if that changes the behavior.

MabezDev, 2021-04-20:
Ah, that's almost definitely it. Just so happened I chose the blinky example to verify my jtag setup. Will verify tomorrow and close :)

MabezDev, 2021-04-21:
That was exactly it, thanks!

closed 2021-04-21 by MabezDev