espressif/openocd-esp32 #153: ESP32C3 WDT's not being disabled by openocd

MabezDev, 2021-04-29:
Openocd config:

```tcl

adapter_khz 10000

# jlink
source [find interface/jlink.cfg]

# Source the ESP32-C3 configuration file
source [find target/esp32c3.cfg]
```

Openocd version:

```bash
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
```

I believe this behavior isn't seen in the IDF because of https://github.com/espressif/esp-idf/blob/1f54d17503ff17d830de75eed4eb47bc6a3aafc3/components/esp_system/port/soc/esp32c3/system_internal.c#L47-L56.

When I add the wdt disabling into [my program (sorry in advance, it's Rust :D)](https://github.com/MabezDev/esp32c3-experiments/blob/master/src/main.rs) I no longer get the RTC & TIMG wdt resets.

With that said, I am still getting a `rst:0x12 (SUPER_WDT_RST)` after these changes, this is probably not the place to ask, but do you know whats causing that? I looked in the [prelim trm](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf), section 1.14 and there are no comments next to Super watchdog reset. Any ideas?

igrr, 2021-05-07:
Hi @MabezDev, esp32c3 config files should [disable the watchdogs](https://github.com/espressif/openocd-esp32/blob/5b67ad1b15938f524f133afb8ef652c990f570eb/tcl/target/esp32c3.cfg#L35-L42) on [halt](https://github.com/espressif/openocd-esp32/blob/5b67ad1b15938f524f133afb8ef652c990f570eb/tcl/target/esp32c3.cfg#L78) or [reset](https://github.com/espressif/openocd-esp32/blob/5b67ad1b15938f524f133afb8ef652c990f570eb/tcl/target/esp32c3.cfg#L65). Could you describe the scenario, i.e. what steps in OpenOCD/gdb do you perform to observe this issue?

Regarding SUPER_WDT_RST, that's a good point. It's another watchdog timer that observes the "RTC" (always-on power domain) hardware itself. For example, if the RTC clock stops for some reason, then the SUPER_WDT will trigger and reset the chip. The RTC block has an "auto-feed" feature, where it can keep feeding the SUPER_WDT, as long as the RTC works normally. This can be enabled as follows: https://github.com/espressif/esp-idf/blob/639e7ad494d93fc82159f9fe854041bc43a96d5f/components/bootloader_support/src/esp32c3/bootloader_esp32c3.c#L259-L264. I think we need to add this step to the esp32c3.cfg.

While these watchdogs are disabled by OpenOCD, you likely still need to disable them in your Rust app, if the app can be executed without OpenOCD, for example from flash. Only if the app is always executed by loading it into IRAM (e.g. with gdb `load` command) can it rely on OpenOCD to disable the watchdogs.

MabezDev, 2021-05-10:
Hi @igrr, thanks for the tip about the SUPER_WDT_RST, when I add the code to set that to autofeed, all is well. I believe what I was experiencing was the following:

* Connect via openocd, halt issued (other wdt's disabled)
* Initial wdt reset was cause is super watchdog
* Chip resets, at which point openocd does not try and connect again
* The other watchdogs are not disabled hence my confusion when seeing them as the reset reason

I think this can be closed now, unless you want to keep it open for tracking disabling the SUPER_WDT.

igrr, 2021-05-11:
Thanks for confirming. Let's keep this open until the fix to disable the SWD is merged.

closed 2021-05-12 by espressif-bot