espressif/openocd-esp32 #156: OpenOCD Debugging not working on ESP32

ashigupta, 2021-05-24:
### Environment
- Development Kit: ESP32-DevKitM-1 
- Module or chip used: ESP32-Mini-1
- Debug Adapter: ESP PROG
- OpenOCD version: openocd-esp32-win32-0.10.0-esp32-20191114
- Operating System: Windows
- Using an IDE?: Eclipse 2020-06 (4.16.0)
- OpenOCD command line: openocd -l opocdlog.txt -d3 -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-Mini-1-kit-3.3v.cfg 
- JTAG Clock: 20KHz
- ESP-IDF version: v4.2

### Problem Description
Debugging on ESP32 is highly un-reliable. Sometime debugging works and sometime it start giving issues even without any code change. Here i am able to debug code once but then it is just not going in debug mode. I have tried many things but still nothing works. Not sure the exact problem here. 
I am using a custom partition table and here is how it looks like
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,3M,

For reference i have also attached sdkconfig file.

Steps to reproduce the behavior:
1. Build ESP project on Eclipse
2. Flash code (without debugger)
3. Attach ESP PROG with ESP32-DevKitM-1
4. Open ESP-IDF terminal and hit command "openocd -l opocdlog.txt -d3 -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-Mini-1-kit-3.3v.cfg"
5. Debug using Eclipse

### Debug Logs
1. Attached openOCD debug logs 
2. Eclipse Screenshot
[image]
3. Attached sdkconfig file

### Expected behavior
Ideally debugging should work and execution should halt at app_main as configuration done in Eclipse as mentioned in this link
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse

[opocdlog.txt](https://github.com/espressif/openocd-esp32/files/6533319/opocdlog.txt)

erhankur, 2021-05-24:
Hi @ashigupta,

Could you please test with the latest openocd release? [20210401](https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20210401)

You can set your board config as `-f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32-solo-1.cfg`

Then please share the openocd log again.

Thanks.

gerekon, 2021-05-25:
> Hi @ashigupta,
>
> Could you please test with the latest openocd release? [20210401](https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20210401)
>
> You can set your board config as [...]

Hi @ashigupta !

Also, please, do not forget to update `OPENOCD_SCRIPTS` env var or use `-s` option properly to point to the scripts directory of `esp32-20210401` release. According to your logs you use `esp32-20200709` scripts dir while you reported that version of OpenOCD is  `esp32-20191114`
```
Debug: 19 3 options.c:186 add_default_dirs(): exepath=C:/Users/E0532110/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin
Debug: 20 3 options.c:187 add_default_dirs(): bin2data=../share/openocd
Debug: 21 3 configuration.c:42 add_script_search_dir(): adding C:\Users\E0532110\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\share\openocd\scripts
Debug: 22 3 configuration.c:42 add_script_search_dir(): adding C:\Users\E0532110\AppData\Roaming/OpenOCD
Debug: 23 3 configuration.c:42 add_script_search_dir(): adding C:/Users/E0532110/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/../share/openocd/site
Debug: 24 3 configuration.c:42 add_script_search_dir(): adding C:/Users/E0532110/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/../share/openocd/scripts
```

Also if you want t use your custom `board/esp32-Mini-1-kit-3.3v.cfg`, please post its contents here.

Thanks

ashigupta, 2021-05-25:
Hello @erhankur @gerekon [openSSL_log_1.txt](https://github.com/espressif/openocd-esp32/files/6539149/openSSL_log_1.txt)
As suggested by Erhan, i have used the shared openOCD version. Please see attached openOCD logs.
I am able to start debugging with this version but it is very unreliable. Sometime it hits on app_main and after doing some step wise debugging execution moves directly to some RTOS files and sometime it just show "UserExceptionVector" and halts. Attached screenshot of the same
[openSSL_log_1.txt](https://github.com/espressif/openocd-esp32/files/6539149/openSSL_log_1.txt)[image]

One more observation, i am providing power for ESP-DevKitM-1 from JTAG on ESP PROG. However to make debugging work, i need to give a separate USB supply. Is this expected behavior ?

Also let me know know how i can make debugging more reliable. There are many uncertainties when it comes to debug on ESP32.

gerekon, 2021-05-25:
@ashigupta I think 20000 Khz speed is too fast for Jlink. From My experiments I was able to debug ESP32 with Jlink reliably with adapter speed at 3000 Khz.

Also, please note that it seems you still use OpenOCD `v0.10.0-esp32-20200709` (not one mentioned by @erhankur)
From your log
```
Debug: 16 3 options.c:184 add_default_dirs(): bindir=/builds/idf/openocd-esp32/openocd-esp32/bin
Debug: 17 3 options.c:185 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/openocd-esp32/share/openocd
Debug: 18 3 options.c:186 add_default_dirs(): exepath=C:/Users/E0532110/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin
Debug: 19 3 options.c:187 add_default_dirs(): bin2data=../share/openocd
Debug: 20 3 configuration.c:42 add_script_search_dir(): adding C:\Users\E0532110\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\share\openocd\scripts
Debug: 21 4 configuration.c:42 add_script_search_dir(): adding C:\Users\E0532110\AppData\Roaming/OpenOCD
Debug: 22 4 configuration.c:42 add_script_search_dir(): adding C:/Users/E0532110/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/../share/openocd/site
Debug: 23 4 configuration.c:42 add_script_search_dir(): adding C:/Users/E0532110/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/../share/openocd/scripts
Debug: 24 10 configuration.c:97 find_file(): found C:\Users\E0532110\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\share\openocd\scripts/interface/ftdi/esp32_devkitj_v1.cfg
```

gerekon, 2021-05-25:
@ashigupta Regarding your debug session... What I learned from the log...
1. At the moment you start OpenOCD your program is stopped due to an exception `LoadProhibitedCause`. I think you should see corresponding output in UART (smth like `Guru mediation error...`). 
```
Debug: 351 107 xtensa_mcore.c:614 xtensa_mcore_poll(): esp32.cpu0: Target halted, pc=0x400D444E, debug_reason=00000001, target_state_old=00000001, active=true
Debug: 352 107 xtensa_mcore.c:620 xtensa_mcore_poll(): esp32.cpu0: Halt reason=0x00000008, exc_cause=28, dsr=0x8000cc11
```
2. After that you start debug session in Eclipse:
    - Connect GDB to OpenOCD.
    - Reset target and halt
    - Set breakpoints
    - Resume execution
3. Execution is stopped at the place of crash mention in step 1. At address 0x400D444E.
```
Debug: 2657 18148 xtensa_mcore.c:614 xtensa_mcore_poll(): esp32.cpu0: Target halted, pc=0x400D444E, debug_reason=00000001, target_state_old=00000001, active=true
Debug: 2658 18148 xtensa_mcore.c:620 xtensa_mcore_poll(): esp32.cpu0: Halt reason=0x00000002, exc_cause=28, dsr=0x8000cc11
```
4. You try to continue execution
```
Debug: 4123 38016 gdb_server.c:3265 gdb_input_inner(): received packet: 'vCont;s:0;c'
```
5. And get into exception vector because instruction your program was stopped at in step 3 re-generates execption.
```
Debug: 4151 38024 xtensa.c:1238 xtensa_do_step(): Stepped from 400D444E to 40080343
Debug: 4152 38024 xtensa.c:1241 xtensa_do_step(): Done stepping, PC=40080343
```

So from my point of view, OpenOCD behaves as expected. But it seems that we need to print a special warning message when the program is stopped due to an exception to make users aware of this condition.

gerekon, 2021-05-26:
@ashigupta 

> However to make debugging work, i need to give a separate USB supply. Is this expected behavior ?

I think ESP-PROG is not suitable to power itself and the board connected to it because in this case 500mA from USB is shared among all consumers. What prevents you from powering your board using separate USB cable?

> @gerekon can you provide command that i can use with -s option.

Just to be sure that OpenOCD (20210401) uses proper scripts dir use absolute path to it or set `OPENOCD_SCRIPTS` env var.

gerekon, 2021-05-26:
@ashigupta Any progress with the latest OpenOCD 20210401?

ashigupta, 2021-05-26:
Hello @gerekon,

We were debugging earlier after powering ESP32-DEveKitM-1 board via JTAG on ESP PROG and somehow that stopped working hence had this question. Noting stops to power board from another USB cable and that's what i am using now.

With the new version of OpenOCD 20210401, i was able to enable debugging and execution is reaching till app_main() which was not happening earlier. I will continue to debug more and see if this is working fine and will update here.

As you recommended, it will also be great to add a message with details of the exception that has occurred.

Is this the recommended version of OpenOCD to use for debugging with ESP32 Mini 1 and ESP32 C3 hardware?

gerekon, 2021-05-26:
@ashigupta 

> Is this the recommended version of OpenOCD to use for debugging with ESP32 Mini 1 and ESP32 C3 hardware?

Yes, we are updating recommended OpenOCD version in IDF branches now to this one.

gerekon, 2021-05-27:
> I was powering ESP32 board using JTAG on ESP PROG and that was also working but suddenly now it has stopped working hence had this question.

I agree that it is possible to power boards from ESP-PROG, but need to take into account power-sharing fact and that poer regulator in this case resides on ESP-PROG, and wires length can have an impact too.

gerekon, 2021-05-27:
> As you recommended, it will also be great to add a message with details of the exception that has occurred.

Will keep it open until it is implemented

erhankur, 2023-01-17:
Closing this, Since exception details are added into [v0.11.0-esp32-20221026](https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20221026)

closed 2023-01-17 by erhankur