espressif/openocd-esp32 #157: Cannot debug with esp-wroom-32 and ft232r

bibble235, 2021-05-26:
Trying to debug with the following
*esp-wroom-32
*ft232r
*unbuntu 21.04

```
interface ft232r
# ft232r_restore_serial 0x15

ft232r_tdi_num TXD
ft232r_tck_num DTR
ft232r_tdo_num RXD
ft232r_tms_num CTS
# trst/srst are not used but must have different values than above
ft232r_trst_num DCD
ft232r_srst_num RI

reset_config none
Using the ft232r.cfg

```
Here is the log
```
WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 3000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : esp32.cpu0: Target halted, PC=0x400E4C76, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
```
On the gdb side of the deal
```
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/blink.elf...done.
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
gdbinit2:1: Error in sourced command file:
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
(gdb) 
```
I am not a very knowledgeable in this field but suspect it is trying to upload via JTAG (which I am told you cannot do on a esp-wroom-32 with CP210x 10c4:ea60) but I of course have no real idea. Happy to provide any info.

gerekon, 2021-05-26:
Hi @bibble235 ,

Please attach verbose debug logs for OpenOCD and GDB.  See [item 3 here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb) for how to generate verbose logs.

Also it would be good to have your `gdbinit` file.

bibble235, 2021-05-26:
[ft232r.log](https://github.com/espressif/openocd-esp32/files/6547434/ft232r.log)

bibble235, 2021-05-26:
Let me know if this is OK.

erhankur, 2021-05-26:
Hi @bibble235, 

Your jtag adapter speed looks like 500 khz which is pretty slow. This might cause timeout issue on the gdb side.
You can increase this value at ft232r.cfg 
```
interface ft232r
adapter_khz 5000
```

If you see a problem with the high jtag speeds, instead you can increase GDB timeout.  ie: set timeout to 100 seconds
```
set remotetimeout 100
target remote :3333
set remote hardware-watchpoint-limit 20
mon reset halt
flushregs
thb app_main
c
```

bibble235, 2021-05-26:
I have tried high and low numbers with no difference.Sent from Android device

bibble235, 2021-05-26:
[ft232r_withtimeout.log](https://github.com/espressif/openocd-esp32/files/6550148/ft232r_withtimeout.log)

This with the timeout  and adapter changes

erhankur, 2021-05-27:
Hi @bibble235,

From the logs I am not sure what is happening. Lets go one by one. 

Please leave adapter speed as before. Which is 500 khz. Then just increase the gdb remote timeout as 100sec. Then share your gdbinit and openocd log files again.

bibble235, 2021-05-27:
Before that, can I just say what I have discovered and bear in mind I am not expert at all.

I am reasonable good a googling and have not found evidence of anyone else doing what I am attempting which is using FT232R device as a JTAG interface with an ESP32 ESP-WROOM-32 device.

Here is what mine looks like https://www.amazon.com/sunfounder-arduino-program-download-mega2560/dp/b00gd6ge6o

I have found people using Jlink devices and have found an FTDI  using a MPSSE device but remember FT232R uses BIT-BANG and not MPSSE

Also note that the flashing via JTAG is not possible from my understanding because my board only has one port. I don't understand this but was told on the esp32 forum and may well have misunderstood. I can upload using the UART on the board though of course

I have distilled the script for testing purposes (I use the standard ones for logs) down to this in case it helps. Attatched.

bibble235, 2021-05-27:
[esp32_iain.cfg.txt](https://github.com/espressif/openocd-esp32/files/6554126/esp32_iain.cfg.txt)

As above, what I understand to be what is executed with 
./openocd -d  -s ../share/openocd/scripts  -f ../share/openocd/scripts/interface/ft232r.cfg -f ../share/openocd/scripts/board/esp-wroom-32.cfg

bibble235, 2021-05-27:
[iain_500.log](https://github.com/espressif/openocd-esp32/files/6554240/iain_500.log)
The ocd log

bibble235, 2021-05-27:
gdbinit
```
set remotetimeout 100
target remote :3333
set remote hardware-watchpoint-limit 20
mon reset halt
flushregs
thb app_main
c

```

erhankur, 2021-05-28:
@bibble235 there is no error in the `iain_500.log` file. Did you see any issue so far?

bibble235, 2021-05-28:
Left it for 10 minutes and the gdb side did not get to the prompt. Never seen the openocd work so cannot comment. Not at a PC but was expecting (gdb) prompt

bibble235, 2021-05-28:
Thanks btw

erhankur, 2021-05-28:
Can we close the issue?

bibble235, 2021-05-28:
No, I cannot debug. Whilst you may see no errors in the log there is no ability to debug because the GDB session does not come back but hangs the same as the ocd session

igrr, 2021-05-29:
Can you try interrupting the program in GDB? You can do this by pressing Ctrl-C. Does that result in the program being halted?

If yes, then the most likely cause is that the breakpoint at app_main doesn't trigger. This is usually caused by mismatch between the program actually present in flash and the ELF file passed to the debugger.

If GDB prints some errors instead of interrupting the program, then the cause is indeed on the OpenOCD side.

LixinGuoX, 2022-01-18:
I am using GDB STUB which is supported by Zephyr, and I have this problem too. It seems that there is something wrong with the hardware config.

bibble235, 2022-01-18:
I was using incorrect hardware with the FT232R which did not support MPSSE. You can read my sorry tale at https://www.bibble.co.nz/mediawiki/index.php?title=ESP32 but in the end I bought a ESP-PROG. It was cheap and did the job and I don't mind giving the ESP32 people the money. I

LixinGuoX, 2022-01-18:
I am trying GDB Stub for debugging over serial line. This allows to use GDB with resource-limited target systems without a dedicated debug interface via JTAG or some other means. And I got the same error log with yours.
I am confused and can not find the root cause.
```
(gdb) target remote /dev/ttyUSB0
Remote debugging using /dev/ttyUSB0
Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df...Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df...Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df...Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df...Timed out.
Timed out.
Timed out.
Ignoring packet error, continuing...
Packet qSupported (supported-packets) is supported
warning: unrecognized item "timeout" in "qSupported" response
Sending packet: $vMustReplyEmpty#3a...Sending packet: $vMustReplyEmpty#3a...Sending packet: $vMustReplyEmpty#3a...Sending packet: $vMustReplyEmpty#3a...Timed out.
Timed out.
Timed out.
Ignoring packet error, continuing...
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
```

bibble235, 2022-01-18:
Right so catching up a bit. Zephyr is an opensource alternative to FreeRTOS? You could try providing a debug log and confirm the hardware, OS and versions. When I ran openocd I needed parameters e.g. 
`./openocd  -d3   -f ../share/openocd/scripts/interface/ft232r.cfg -f ../share/openocd/scripts/board/esp-wroom-32.cfg`
so these might be useful. Generally most people want to help, it is in their DNA. Hopefully this is  helpful. I am now interested in putting Zephyr on my ESP32 just cos I can. Good luck.

LixinGuoX, 2022-01-18:
Yes, Zephyr supports many boards. I build and flash samples/helloworld with CONFIG_GDBSTUB=y, CONFIG_PRINTK=n, CONFIG_LOG=n, then start a gdb client and have a connection failure.
If you are interested in zephyr, you can follow those docs:
https://docs.zephyrproject.org/latest/getting_started/index.html
https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html
https://docs.zephyrproject.org/latest/guides/debug_tools/gdbstub.html

gerekon, 2022-01-27:
@LixinGuoX your problem belongs to GDB stub in IDF. Could you create a ticket in esp-idf repo?

I am closing this ticket since the topic starter solved his problem.

closed 2022-01-27 by gerekon