espressif/openocd-esp32 #158: Programming before debug  ESP32-C3 with built in USB JTAG fails

BrianAtDocumentedDesigns, 2021-05-26:
### Environment
- Development Kit: ESP32-C3 mini devkit
- Module or chip used: ESP32-C3 mini
- Debug Adapter: Built in
- OpenOCD version: v0.10.0-esp32-20210401
- Operating System: Windows 10
- Using an IDE?: ESP-IDF Eclipse
- OpenOCD command line: -s ${openocd_path}/share/openocd/scripts -f board/esp32c3-builtin.cfg -c "program_esp build/blink.bin 0x10000 verify exit"
- 
- JTAG Clock: speed 40000 kHz
- ESP-IDF version: Don't know: most recent

### Problem Description
Steps to reproduce the behavior: 
In Eclipse
1. Compile blink example
2. Set up debug profile. (Note you have to manually replace actual executable with path to riscv32-esp-elf\bin\riscv32-esp-elf-gdb.exe) and command line to -s ${openocd_path}/share/openocd/scripts -f board/esp32c3-builtin.cfg -c "program_esp build/blink.bin 0x10000 verify exit"
3. Click Debug

Script fails !
### Debug Logs

Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
force hard breakpoints
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
C:\Users\bjpic\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:101: Error: invalid command name "esp"
in procedure 'program_esp' 
at file "C:\Users\bjpic\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 101

### Expected behavior
As per numerous examples on the espressif website, use of -c "program_esp build/blink.bin 0x10000 verify exit" should program the device. Manual programming through the "run" configuration works fine. It is necessary to manually program, change configuration, then run OCD without the automatic programming

### Screenshots
If applicable, add screenshots to help explain your problem.

gerekon, 2021-05-26:
Hi @BrianAtDocumentedDesigns,

OpenOCD has no flash support yet. It is [disabled by default](https://github.com/espressif/openocd-esp32/blob/master/tcl/target/esp32c3.cfg#L4). It is planned for the next release.
So for now you need to use `idf.py` or `esptool.py` directly to program esp32c3.

BrianAtDocumentedDesigns, 2021-05-26:
It is a pity that the documentation says otherwise. I hours trying to solve this, made several posts on the forum, etc..

see https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/index.html#setup-of-openocd

gerekon, 2021-05-26:
Sorry for the inconvenience.

gerekon, 2021-05-27:
Will keep it open until ESP32-C3 flash support is merged

closed 2021-06-16 by espressif-bot