espressif/openocd-esp32 #159: VSCode+PlatformIO+ESP32-WemosMiniD1: Error: unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'

juergs, 2021-06-09:
### Environment
- Development Kit: 		wemos_d1_mini32
- Module or chip used: 	ESP32-WROOM-32
- Debug Adapter: 		ESP-PROG
- OpenOCD version: 		v0.10.0-esp32-20201202 (2020-12-02-17:38)
- Operating System: 	Windows 10 Prof.
- Using an IDE?: 		VScode with PlatformIO
- OpenOCD command line: openocd.exe -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32.cfg
- JTAG Clock: 			adapter speed: 5000 kHz	
- ESP-IDF version: 	toolchain-xtensa32@2.50200.97

### Problem Description
Steps to reproduce the behavior:
1. See error starting  debug-session:		  
 _"unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'"_
2. Message: Assertion failed!
3. Using FTDI **FT2232H**, channel 0 = WinUSB-driver / channel 1 usb cdc-driver = COM30 (Zadig configured!)
4. ftdi device with vid 0403, pid 6010 present
5. board/interface has <u>no</u> jtag_trst !
5. Some days left, it worked sucessfully!

Debug-session output:
===============

_Reading symbols from C:\Users\js\Documents\PlatformIO\Projects\ESP32_TEST_DEBUG\.pio\build\wemos_d1_mini32\firmware.elf...
done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = esp-prog
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.10.0-esp32-20201202 (2020-12-02-17:38)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
added infos by me:
*** In esp-wroom-32.cfg: C:Usersjs.platformiopackages	ool-openocd-esp32shareopenocdscriptsoardesp-wroom-32.cfg
*** In esp32.cfg:  C:Usersjs.platformiopackages	ool-openocd-esp32shareopenocdscripts	argetesp32.cfg
*** In esp_common.cfg:  C:Usersjs.platformiopackages	ool-openocd-esp32shareopenocdscripts	argetesp_common.cfg
2

Info : FreeRTOS creation
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Error: failed to reset FTDI device: LIBUSB_ERROR_PIPE
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Error: no device found
[openocd.log.txt](https://github.com/espressif/openocd-esp32/files/6625270/openocd.log.txt)

!!!!!!!  **Assertion failed!**    <======   ./src/jtag/core.c, Line 343

Program: C:\Users\js\.platformio\packages\tool-openocd-esp32\bin\openocd.exe
File: ../src/jtag/core.c, Line 343

Expression: jtag_trst == 0
.pioinit:11: Error in sourced command file:
Remote communication error.  Target disconnected.: No error._

===============================================
Manually invoked:
_C:\Users\js\Downloads\openocd-esp32-win32-0.10.0-esp32-20210401\openocd-esp32\bin>openocd.exe -l ./openocd.log.txt -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -d_
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Assertion failed!

Program: C:\Users\js\Downloads\openocd-esp32-win32-0.10.0-esp32-20210401\openocd-esp32\bin\openocd.exe
File: ../src/jtag/core.c, Line 343

Expression: jtag_trst == 0
================================================

**No more idea what to resolve this issue.**

regards
Juergen

juergs, 2021-06-10:
Trying to connect manually with some other configfiles seems to be successful:
'''
C:\Users\js\Downloads\openocd-esp32-win32-0.10.0-esp32-20210401\openocd-esp32\bin>openocd -d  -s C:\Users\js\Downloads\openocd-esp32-win32-0.10.0-esp32-20210401\openocd-esp32\bin  -f "C:\Users\js\.platformio\packages\tool-openocd-esp32\share\openocd\scripts\interface\ftdi\esp32_devkitj_v1.cfg" -f  "C:\Users\js\Downloads\openocd-esp32-win32-0.10.0-esp32-20210401\openocd-esp32\share\openocd\scripts\board\esp32-wrover.cfg" > openocd.log.txt 2>&1
'''
Therefore i will try to use this version "openocd-esp32-win32-0.10.0-esp32-20210401" with VSCode in conjunction with their their configfiles.

[openocd.log.txt](https://github.com/espressif/openocd-esp32/files/6630690/openocd.log.txt)

closed 2021-06-10 by juergs

wilian-wagner, 2023-01-19:
oh Thanks, But how do i solve it? i mean, i have to create a file with that lines and put in the project ? i Don't understand...