espressif/openocd-esp32 #162: Can't Build on Windows (for real this time)

BrianAtDocumentedDesigns, 2021-06-20:
### Environment
Attempting to build on Windows 10/MSYS2

- Development Kit:  ESP32-C3 mini Devkit
- Module or chip used: ESP32-C3 mini 
- Debug Adapter: Built it
- OpenOCD version: Current master branch
- Operating System: Windows
- Using an IDE?: MSYS2
- OpenOCD command line: na
- ESP-IDF version: current master branch

### Problem Description

I posted this to the espressif forum but did not get any suggestions. I had previously posted a similar issue but I was not following the most recent instructions.

I have been trying to build a current version of openocd on Windows using the instructions here https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/index.html#jtag-debugging-configuring-target

I have tried it multiple times but it fails with

configure: error: libusb-1.x is required for the MPSSE mode of FTDI based devices

even though I downloaded and unzipped libusb-1.x as per the instructions and when I ls I see it is were I think it should be

$ echo $CPPFLAGS
-I/c/Users/bjpic/esp/libusb/include/libusb-1.0 -D__USE_MINGW_ANSI_STDIO=1 -Wno-error

bjpic@LAPTOP-70Q5CT1Q MSYS ~/esp
$ echo $LDFLAGS
-L/c/Users/bjpic/esp/libusb/MinGW32/.libs/dll

bjpic@LAPTOP-70Q5CT1Q MSYS ~/esp
$ ls
buildsetup.bat config.bat examples install.bat libusb-1.0.22.7z openocd-esp32 sdkconfig.ci
CMakeLists.txt example_test.py export.bat libusb Makefile README.md sdkconfig.defaults

Because configure fails I cannot run make. If I run configure without the options (i.e. --disable-doxygen-pdf --enable-ftdi --enable-jlink --enable-ulink --build=i686-w64-mingw32 --host=i686-w64-mingw32) It give me errors because libusb is not found.

I have attached a fill trace of the problem.

I am trying to build the most recent version of openocd because when I try to run it as per https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/index.html#jtag-debugging-configuring-target

I get the following

C:\Users\bjpic\Desktop\esp-idf>openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.10.0-esp32-20210401 (2021-04-01-15:46)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info : hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections

Note the **Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED**

This occurs even though I have tried all the permutations of drivers for the USB port. When I try actual debugging in Eclipse I get a register dump and numerous lines like 

Info : [0] Found 8 triggers
Error: esp_usb_jtag: usb sent only 0 out of 17 bytes.
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
...
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: esp_usb_jtag: usb sent only 0 out of 2601 bytes.

It seemed that since ESP32-C3 is new it makes more sense to use the most recent version of Openocd before trying to figure out the GDB issue.

### Debug Logs
See attached for full dump of build output

### Expected behavior
I would expect to be able to build openocd following the provided instructions.

### Screenshots
na
[openocdfails.txt](https://github.com/espressif/openocd-esp32/files/6682310/openocdfails.txt)

gerekon, 2021-06-24:
@BrianAtDocumentedDesigns Can you tell the revision of eps32c3 you have? The very first revisions had problems with builtin USB_JTAG and need to disable it for debugging with external debug adapter.
You can see revision printed to UART during boot proces. Smth like
```
I (31) boot: chip revision: 0
```
or get it using esptool
```
hello_world(master) ✗: esptool.py -p /dev/ttyUSB0 chip_id
esptool.py v3.2-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 0)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:32:21:d0
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-C3 has no Chip ID. Reading MAC instead.
MAC: 7c:df:a1:32:21:d0
Hard resetting via RTS pin...
```

BrianAtDocumentedDesigns, 2021-06-25:
gerekon
My problem is building OpenOCD on Windows. This does not require an ESP32 of any type. Either I am not following the instructions correctly or the instructions are wrong. Given the state of ESP32 related in general, the documentation is likely out of date. 

The binary included in ESP-IDF is a dated one from April. That is why I want to build the binary from sources. I subsequently discovered that the Eclipse ESP32 plug in is not compatible with Eclipse versions after March 2021, but that does not mean OpenOCD is up to date. I know, for example that there has been at least one recent bug fix.

For the record, the ESP32-C3 I am using was supplied to me by Espressif in June for evaluation of the JTAG function. 

If the instruction 

This is the dump I get when I flash the device
[2/3] cmd.exe /C "cd /D C:\Users\bjpic\Desktop\esp-idf\components\esptool_py && C:\Users\bjpic\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="C:/Users/bjpic/Desktop/esp-idf" -D SERIAL_TOOL="python C:/Users/bjpic/Desktop/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32c3" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="C:/Users/bjpic/eclipse-windows/SoftAP1/build" -P C:/Users/bjpic/Desktop/esp-idf/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32c3 -p COM6 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 wifi_softAP.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.1-dev
Serial port COM6
Connecting...
Chip is unknown ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:66:a6:3c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.

Thanks
Brian

BrianAtDocumentedDesigns, 2021-06-25:
I should have added that because of a problem with Eclipse (see https://www.esp32.com/viewtopic.php?f=13&t=21654#p78920) because I can't download or build the version of OpenOCD binary which fixes this bug https://github.com/espressif/openocd-esp32/issues/158 I have to close and reopen Eclipse when switching configurations. So for every change in source/debug session:

1) Switch to Run config
2) Modify code, compile, flash
3) Switch to Debug config
4) close Eclipse
5) Open Eclipse and start debug session.

The version of OpenOCD which is in the ESP-IDF is v0.10.0-esp32-20210401

gerekon, 2021-06-28:
@BrianAtDocumentedDesigns 
```
Chip is unknown ESP32-C3 (revision 3)
```
Revision 3 should have no problems with builtin USB_JTAG.

Regarding your build problem... I think you need to add path to your libusb pkgconfig file (`libusb.pc`) to `PKG_CONFIG_PATH`. See `configure --help` output.

Meanwhile I attach Windows distro build for the latest OpenOCD `master`.
[openocd-esp32-win32-0.10.0-esp32-20210401-51-g84ee1302b.zip](https://github.com/espressif/openocd-esp32/files/6729282/openocd-esp32-win32-0.10.0-esp32-20210401-51-g84ee1302b.zip)

Let us know the results.

erhankur, 2021-06-30:
Hi @BrianAtDocumentedDesigns ,

sample `pkgconfig/libusb.pc` can be like below. You just need to change prefix path. 

```
prefix=/home/eloha/esp/openocd_libs/libusb
exec_prefix=${prefix}
libdir=${exec_prefix}/MinGW32/dll
includedir=${prefix}/include/libusb-1.0

Name: libusb-1.0
Description: C API for USB device access from Linux, Mac OS X, Windows, OpenBSD/NetBSD and Solaris userspace
Version: 1.0.22
Libs: -L${libdir} -lusb-1.0
Libs.private: 
Cflags: -I${includedir}/libusb-1.0
```

then you can export the path which has `pkgconfig` directory.

`export PKG_CONFIG_PATH=$PWD/pkgconfig`

gerekon, 2021-07-22:
@BrianAtDocumentedDesigns  Can we close this issue? 

Updated ESP32-C3 support is released. Please see [v0.10.0-esp32-20210721](https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20210721).
As far as I understand your current problems are connected with Eclipse, not with OpenOCD.

BrianAtDocumentedDesigns, 2021-07-22:
I haven't had time to test it but if you want to close it, fine with me.

To be clear I have found significant issues with esp-idf, Eclipse, OpenOCD, and the documentation for all these components with respect to the ESP32-C3. I am more than happy that the list is reduced to only esp-idf, Eclipse, and the documentation.

closed 2021-07-22 by BrianAtDocumentedDesigns

TravisRo, 2021-07-22:
I just fought this problem all morning until I finally found this thread.  Using this newer version v0.10.0-esp32-20210721 definitely fixes it!  They need to get this in their esp-idf-tools installer and Visual Studio Code.  I was beginning to think my C3 dev board was to blame.