espressif/openocd-esp32 #163: Debugging ESP32-C3 fails

frederictobiasc, 2021-06-25:
### Environment
- Development Kit: ESP32-C3-DevKitC-02
- Debug Adapter: built-in JTAG Interface (via USB port)
- OpenOCD version: v0.10.0-esp32-20210401
- Operating System: [e.g. Linux]
- Using an IDE?: CLI
- OpenOCD command line: openocd -f board/esp32c3-builtin.cfg
- JTAG Clock: default
- ESP-IDF version: v4.4-dev-1594-g1d7068e4b

### Problem Description
Steps to reproduce the behavior:
1. Install dev environment as described in the [getting started guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/get-started/index.html)
2. Follow the JTAG debugging guide until [Run OpenOCD](https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32c3/api-guides/jtag-debugging/index.html#run-openocd)
4. See error

### Debug Logs
```
% openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!
```
### Expected behavior
A starting OpenOCD server

### Futher description
Everything else works normal, eg. Flashing applications, monitoring output via `idf.py -p /dev/ttyUSB0 monitor`

igrr, 2021-06-28:
@frederictobiasc Could you please check which revision of ESP32-C3 you have? It will be shown if you run esptool.py, for example:
```
$ esptool.py --port /dev/ttyUSB0 read_mac
esptool.py v3.2-dev
Serial port /dev/ttyUSB0
Connecting...
Failed to get PID of a device on /dev/ttyUSB0, using standard reset sequence.
....._
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:86:d8:bc
Uploading stub...
Running stub...
Stub running...
MAC: 7c:df:a1:86:d8:bc
Hard resetting via RTS pin...
```
Here the chip is of revision 3.

Could you also confirm that the device was enumerated by running `lsusb`?

frederictobiasc, 2021-06-28:
Hi @igrr and thanks for your response.
`esptool.py` says it's a revision 3:
```
% esptool.py --port /dev/ttyUSB0 read_mac
esptool.py v3.1-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is unknown ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:<removed>
Uploading stub...
Running stub...
Stub running...
MAC: 7c:df:a1:<removed>
Hard resetting via RTS pin...
```
`lsusb` shows a `Silicon Labs CP210x UART Bridge` when connected:

```
% lsusb
Bus 002 Device 007: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 002 Device 006: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 005: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 04f2:b604 Chicony Electronics Co., Ltd Integrated Camera (1280x720@30)
Bus 001 Device 005: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 015: ID 04f2:b612 Chicony Electronics Co., Ltd USB2.0 FHD UVC WebCam
Bus 001 Device 014: ID 0bda:48a5 Realtek Semiconductor Corp. USB Audio
Bus 001 Device 017: ID 046a:0021 Cherry GmbH CyMotion Expert Combo
Bus 001 Device 016: ID 046d:c069 Logitech, Inc. M-U0007 [Corded Mouse M500]
Bus 001 Device 013: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 012: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 003: ID 058f:9540 Alcor Micro Corp. AU9540 Smartcard Reader
Bus 001 Device 002: ID 046d:0a45 Logitech, Inc. 960 Headset
Bus 001 Device 019: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

igrr, 2021-06-28:
> lsusb shows a Silicon Labs CP210x UART Bridge when connected:

This means that the development board is connected via the CP2102 USB-UART bridge, not via the built-in USB peripheral. This explains why OpenOCD can not find the device with required VID and PID.

If you have some dupont wires, a spare USB cable, and a soldering iron, can try to wire up a USB cable to the D+ and D- GPIOs of ESP32-C3: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html#hardware-requirements. Or find an existing one like this: https://www.adafruit.com/product/4448.

igrr, 2021-06-28:
Here is some more information about this development board: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitc-02.html and schematics: https://dl.espressif.com/dl/schematics/SCH_ESP32-C3-DEVKITC-02_V1_1_20210126A.pdf. It is possible to replace R1/R4 jumpers to R2/R3 footprints, thereby connecting the USB socket to the ESP32-C3 instead of CP2012. The downside is that these resistors are very tiny. Making a USB breakout cable might be easier.

frederictobiasc, 2021-06-28:
Hi, I now connected the development via the D+ and D- GPIOs. `lsusb` now shows `Espressif USB JTAG/serial debug unit`.

Thanks a lot for your support @igrr . Now, OpenOCD starts. 
But only when executing with root privileges.
```
% sudo openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
```

Without sudo:

```
% openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: could not find or open device!
```

closed 2021-06-28 by frederictobiasc