espressif/openocd-esp32 #164: OpenOCD/GDB connection is dropped on startup

readmodifywrite, 2021-06-29:
### Environment
ESP-WROVER-KIT V4.1
Linux Mint 20.04
openocd -f board/esp32-wrover-kit-3.3v.cfg
Eclipse debug configured from getting started guide
ESP32 SDK version:v3.3.5-17-gbc357a320
openocd v0.10.0-esp32-20200709 (2020-07-09-08:54)

OpenOCD console output:
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset.
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : cpu0: Detected debug stubs @ 3ffbda04
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Info : Target halted. CPU0: PC=0x4014A6E6 (active)
Info : Target halted. CPU1: PC=0x400843B3 
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 92 KB
Info : Flash mapping 1: 0x30018 -> 0x400d0018, 492 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.flash' size 4096 KB
Info : Using flash bank 'esp32.flash' size 4096 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 92 KB
Info : Flash mapping 1: 0x30018 -> 0x400d0018, 492 KB
Info : Using flash bank 'esp32.irom' size 496 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 92 KB
Info : Flash mapping 1: 0x30018 -> 0x400d0018, 492 KB
Info : Using flash bank 'esp32.drom' size 96 KB
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400 
Info : Target halted. CPU0: PC=0x400E5E60 (active)
Info : Target halted. CPU1: PC=0x40089DC3 
Info : cpu0: Detected debug stubs @ 3ffbda04
Info : dropped 'gdb' connection

GDB console output (from Eclipse):

Temporary breakpoint 2, app_main () at user_main.c:28
28	{

/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]

This is a bug, please report it.  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.

/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]

igrr, 2021-06-29:
@readmodifywrite could you please check if `CONFIG_FREERTOS_TASK_FUNCTION_WRAPPER` is enabled in sdkconfig?

readmodifywrite, 2021-06-29:
Looks like it is not, and I cannot find an option in the menuconfig to enable it.  Editing the sdkconfig file also didn't work, the changes are overwritten by the build.

igrr, 2021-06-29:
This option depends on CONFIG_COMPILER_OPTIMIZATION_DEFAULT (https://github.com/espressif/esp-idf/blob/b65a37795f3c04d36f0a504a74e7e56950e3e3d9/components/freertos/Kconfig#L413), perhaps you have CONFIG_OPTIMIZATION_LEVEL_RELEASE enabled?

readmodifywrite, 2021-06-29:
Ok, that does it, thanks!  Could we get an update on the docs?  They are otherwise extremely thorough and well put together, the rest of the setup was great.

igrr, 2021-06-30:
Thanks for the feedback. Will keep the issue open to track the documentation update.